weliem/blessed-android #2: lower minSdkVersion to 19

oliexdev, 2019-05-25:
First of all great library. Currently I am trying out your library with my project openScale to replace the RxAndroidBle library but your minSdkVersion is 24 which is not really necessary, isn't it?
I lowered your library to 19 and I had no problems? Is it possible that lower your minSdkVersion to 19, so that I can use your library without increasing my minSdkVersion? :wink:

weliem, 2019-05-25:
Hi Olie, 

I don't have any Android phones at my disposal that have a lower version than 23. So I can't test below that. The library also uses the 'new' scanner that was introduced in 21. So I don't think it will work on versions below that...did you try on it on 4.4? I'd be surprised if it works...

So I guess I could put it to 21....does that help?

oliexdev, 2019-05-25:
yes it works also with minSdkVersion 19. You can emulate Android 4.4 and compile your example app. It will work.

weliem, 2019-05-25:
I'll play around to see if it works. Although I don't really understand why....perhaps the appcompat library makes it work?  But that is only used in the demo app....

oliexdev, 2019-05-26:
I am not 100% why its works but BLE support was introduced in API 18.
I had also implemented with the native Android Ble the openScale backend and I didn't see any problems or received problems reports with Android 4.4.

weliem, 2019-05-26:
I tried running the demo app on a phone with API 21 and it didn't work. It crashed. 
I tried running the app in the simulator with API 19 and it didn't work because the emulator doesn't support Bluetooth. But the app didn't crash...

Anyway, I made some changes and it now works on API 21. I just committed that. So that is already an improvement.

In order to support up to 19 I would have to change the scanning logic and use the 'old' scanner classes. Not sure if that is worth the effort. Do you still see people with device lower than 21 using openScale? Can't be that many...

oliexdev, 2019-05-26:
It is not a huge amount of openScale users. Around 3% (~1500 users) are using Android 4.4. But I have only statistics for GooglePlay not for f-droid.
So it's up to you if you want to lower the minSdkVersion. If you don't want I can also increase the minSdkVersion. But as I said I did not get any bad reports from Android 4.4 users using my old Bluetooth backend.

weliem, 2019-05-28:
I made a release now where minSdkVersion is down to 21. If I can get hold of a phone with Android 4.4 I can investigate supporting that but I don't want to release software that I haven't been able to test properly.

oliexdev, 2019-05-28:
Yeah I can understand that. I will increase my minSdkVersion to 21 then. Thanks for your great library :1st_place_medal:

closed 2019-05-28 by oliexdev

weliem, 2019-06-01:
Cool! Let me know if you run into issues!