weliem/blessed-android #6: Cannot bond BLE device - no popup dialog is shown

Maartinsh, 2019-08-27:
So I started off trying to follow android instructions, but I got the same results. Then I thought that I'm implementing something very bad, so I decided to test it with your library.

`08-27 15:53:52.283 8853-8853/com.welie.blessedexample W/BluetoothCentral: peripheral with address 'E6:F6:05:7E:6A:5D' is not in the Bluetooth cache, hence connection may fail
08-27 15:53:57.308 8853-8853/com.welie.blessedexample I/BluetoothPeripheral: connect to 'null' (...7E:6A:5D) using TRANSPORT_LE
08-27 15:54:02.350 8853-8866/com.welie.blessedexample I/BluetoothPeripheral: connected to 'null' (BOND_NONE) in 5.0s
08-27 15:54:03.167 8853-8866/com.welie.blessedexample I/BluetoothPeripheral: discovered 5 services for 'null'
08-27 15:54:03.170 8853-8853/com.welie.blessedexample I/BluetoothHandler: connected to 'null'
08-27 15:54:03.170 8853-8853/com.welie.blessedexample I/BluetoothHandler: discovered services
08-27 15:54:04.476 8853-8995/com.welie.blessedexample W/BluetoothPeripheral: read needs bonding, bonding in progress`

Well, I get a error 137 which is related with authorization. Kinda makes sense, because I know I have to enter pin for this specific characteristic to be read, but the problem is that I cannot seem to open dialog (it is not shown). 

1. pairingRequestBroadcastReceiver does not receive anything, so I think that text "bonding is in progress" is not correct. I did try to manually do bonding with createBond(); same results.

Back story: When I did this implementation on my side first, I saw a dialog - I entered correct pin and then I unpaired the device to try re-bonding again. After that, there was no pairing dialog shown. I tried different devices - same result. Could it be that device somehow caches bonding or something? Maybe device caches something?

weliem, 2019-08-28:
Hi,

First thing to note from the log is that you are connecting to a device that is not in the phone's cache (see 'is not in the Bluetooth cache' in the log. Try scanning for the device first and then connect to it. Connecting to uncached devices is bad practice and will lead to unpredictable results so don't do it..

The second thing that is weird is that your device has apparently no name ('null'). Is that correct or just because you didn't scan first?

Apparently you are lucky and the device still connected and it's services were discovered. But Android thinks you are not bonded with the device (see BOND_NONE in the log).  

Error 137 is GATT_AUTH_FAIL. You are trying to read an encrypted characteristic and normally this should trigger a dialog. But you say that doesn't happen, right?

So I am not sure what is going on and why the dialog doesn't appear. Have you tried with other devices that need a pin code?

Anyway, try scanning for the device first so that you fix the caching issue....

Maartinsh, 2019-08-29:
Hi, thanks for response.

> Try scanning for the device first and then connect to it. 

How can I do that? I previously used:

`central.startPairingPopupHack()
central.scanForPeripherals()`

> But you say that doesn't happen, right?

Correct.

weliem, 2019-08-31:
Any update on this issue?

Maartinsh, 2019-09-03:
I'm still not sure if the way how I scan and connect is wrong/correct, but I understood why it does not show pin code dialog and respond 137 automatically. Reason being is that my device was coded in such way that it does not remove old bonding information when user decides to disconnect from it. When I did hard reset on the device (cleared bonding information) pin alert dialog works properly and I get correct response code.

weliem, 2019-09-05:
Ok, so it wasn't a issue with the Blessed library then. Will close the issue then...

closed 2019-09-05 by weliem