weliem/blessed-android #7: Connection timeout

PatricioIN, 2019-09-06:
Hi!

We´re trying to connect our scale: "mi scale 2" with a redmi 5 plus (8.1.0) but we can´t do the connection.

>  Try to connect to BLE device D0:96:65:48:E7:F5
> connect to 'MIBCS' (D0:96:65:48:E7:F5) using TRANSPORT_LE
>  device: D0:96:65:48:E7:F5, auto: false
>  registerApp()
> registerApp() - UUID=ca8e1ff7-7c06-4362-9779-27b7e159256e
> onClientRegistered() - status=0 clientIf=7
> connection timout, disconnecting 'MIBCS'
> close()
> unregisterApp() - mClientIf=7
> disconnected 'MIBCS' with status 8
> 
> 

With others devices work fine. ¿Any ideas?

Thank you very much!!!!

weliem, 2019-09-08:
Hi, the Xiaomi scale has a bit strange behaviour. When you stand on it you can connect without problems. But when it is 'sleeping' connections may sometimes fail. It has nothing to do with the library, it is caused by the device. 

What you can do in your app is to look at the advertisement data that is part of the ScanRecord and only connect when the advertisement data indicates the scale is in use or has new data.

PatricioIN, 2019-09-09:
Thank you very much Weliem!!

closed 2019-09-15 by weliem