weliem/blessed-android #11: Cannot connect to devices now and then

Maartinsh, 2019-09-17:
The flow I've is:

1. Start scanning
2. When specific device is found, Stop the scan and do connection 
3. If connection is successful for specific device, do 1.-3. steps again for other devices that are found nearby.

Problem is that, sometimes connection is very easy and works great, but most of the time it fails with error code 8. Is there a way to create bullet-proof solution that connection work 8/10 times? For me it's like 2/10 for now. My logic holds on doing a single connection once to the device to procceed (it does not have to stay in connected state, it's enough for a single successful connection per device to continue). So, making a connection is crucial for my app. Below is the log.

`
2019-09-17 07:20:22.258 7199-7565/com.ae.dnight I/XBeaconManager: XBeacon recognized! Minor ID: 38 recognized!
2019-09-17 07:20:22.326 7199-7566/com.ae.dnight I/XBeaconManager$saveXBeacon: Saving XBeacon: 38 minorId!
2019-09-17 07:20:22.387 7199-7566/com.ae.dnight I/IBDatabaseHelper: Beacon saved to DB: majorId - 0 minorId - 38
2019-09-17 07:20:22.751 7199-7567/com.ae.dnight I/XBeaconManager: XBeacon recognized! Minor ID: 196 recognized!
2019-09-17 07:20:22.789 7199-7567/com.ae.dnight I/XBeaconManager$saveXBeacon: Saving XBeacon: 196 minorId!
2019-09-17 07:20:22.845 7199-7567/com.ae.dnight I/IBDatabaseHelper: Beacon saved to DB: majorId - 0 minorId - 196
2019-09-17 07:20:25.568 7199-7567/com.ae.dnight I/XBeaconManager: XBeacon recognized! Minor ID: 38 recognized!
2019-09-17 07:20:25.602 7199-7567/com.ae.dnight I/XBeaconManager: XBeacon is trying to create a connection! Minor ID: 38!
2019-09-17 07:20:25.616 7199-7567/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:20:25.616 7199-7567/com.ae.dnight D/BluetoothLeScanner: Stop Scan with callback
2019-09-17 07:20:25.622 7199-7567/com.ae.dnight I/BluetoothCentral: scan stopped
2019-09-17 07:20:25.857 7199-7551/com.ae.dnight V/FA: Inactivity, disconnecting from the service
2019-09-17 07:20:30.641 7199-7199/com.ae.dnight I/BluetoothPeripheral: connect to 'null' (E6:F6:05:7E:6A:5D) using TRANSPORT_LE
2019-09-17 07:20:30.647 7199-7199/com.ae.dnight I/BluetoothDevice: removeBond() for device E6F605 called by pid: 7199 tid: 7199
2019-09-17 07:20:30.674 7199-7199/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:20:30.679 7199-7199/com.ae.dnight D/BluetoothGatt: connect() - device: E6:F6:05:7E:6A:5D, auto: false
2019-09-17 07:20:30.679 7199-7199/com.ae.dnight D/BluetoothAdapter: isSecureModeEnabled
2019-09-17 07:20:30.681 7199-7199/com.ae.dnight D/BluetoothGatt: registerApp()
2019-09-17 07:20:30.683 7199-7199/com.ae.dnight D/BluetoothGatt: registerApp() - UUID=278c609f-2e9b-40c4-97c0-40e44dd28114
2019-09-17 07:20:30.690 7199-7244/com.ae.dnight D/BluetoothGatt: onClientRegistered() - status=0 clientIf=4
2019-09-17 07:20:30.691 7199-7199/com.ae.dnight W/ae.dnight: Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (light greylist, reflection)
2019-09-17 07:20:30.691 7199-7199/com.ae.dnight D/BluetoothGatt: refresh() - device: E6:F6:05:7E:6A:5D
2019-09-17 07:20:35.731 7199-7244/com.ae.dnight D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=4 device=E6:F6:05:7E:6A:5D
2019-09-17 07:20:35.760 7199-7244/com.ae.dnight I/BluetoothPeripheral: connection failed with status 'GATT ERROR' (TIMEOUT)
2019-09-17 07:20:35.761 7199-7244/com.ae.dnight D/BluetoothGatt: close()
2019-09-17 07:20:35.775 7199-7244/com.ae.dnight D/BluetoothGatt: unregisterApp() - mClientIf=4
2019-09-17 07:20:35.792 7199-7244/com.ae.dnight I/BluetoothCentral: connection to 'null' (E6:F6:05:7E:6A:5D) failed
2019-09-17 07:20:35.794 7199-7199/com.ae.dnight I/XBeaconManager: Restarting XBeacon Manager
2019-09-17 07:20:35.807 7199-7199/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:20:35.808 7199-7199/com.ae.dnight D/BluetoothLeScanner: could not find callback wrapper
2019-09-17 07:20:35.808 7199-7199/com.ae.dnight I/BluetoothCentral: scan stopped
2019-09-17 07:20:35.813 7199-7199/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:20:35.818 7199-7199/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:20:35.819 7199-7199/com.ae.dnight D/BluetoothLeScanner: Start Scan with callback
2019-09-17 07:20:35.824 7199-7244/com.ae.dnight D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=4 mScannerId=0
2019-09-17 07:20:35.830 7199-7199/com.ae.dnight I/BluetoothCentral: scan started
2019-09-17 07:20:35.834 7199-7199/com.ae.dnight E/XBeaconManager: connection 'E6:F6:05:7E:6A:5D' failed with status 8
2019-09-17 07:20:38.901 7199-7569/com.ae.dnight I/XBeaconManager: XBeacon recognized! Minor ID: 38 recognized!
2019-09-17 07:20:38.934 7199-7569/com.ae.dnight I/XBeaconManager: XBeacon is trying to create a connection! Minor ID: 38!
2019-09-17 07:20:38.943 7199-7569/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:20:38.943 7199-7569/com.ae.dnight D/BluetoothLeScanner: Stop Scan with callback
2019-09-17 07:20:38.947 7199-7569/com.ae.dnight I/BluetoothCentral: scan stopped
2019-09-17 07:20:43.970 7199-7199/com.ae.dnight I/BluetoothPeripheral: connect to 'null' (E6:F6:05:7E:6A:5D) using TRANSPORT_LE
2019-09-17 07:20:43.973 7199-7199/com.ae.dnight I/BluetoothDevice: removeBond() for device E6F605 called by pid: 7199 tid: 7199
2019-09-17 07:20:44.003 7199-7199/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:20:44.006 7199-7199/com.ae.dnight D/BluetoothGatt: connect() - device: E6:F6:05:7E:6A:5D, auto: false
2019-09-17 07:20:44.006 7199-7199/com.ae.dnight D/BluetoothAdapter: isSecureModeEnabled
2019-09-17 07:20:44.008 7199-7199/com.ae.dnight D/BluetoothGatt: registerApp()
2019-09-17 07:20:44.011 7199-7199/com.ae.dnight D/BluetoothGatt: registerApp() - UUID=09031068-e187-4d0d-a583-8fe3258cba9c
2019-09-17 07:20:44.016 7199-7199/com.ae.dnight D/BluetoothGatt: refresh() - device: E6:F6:05:7E:6A:5D
2019-09-17 07:20:44.017 7199-7244/com.ae.dnight D/BluetoothGatt: onClientRegistered() - status=0 clientIf=4
2019-09-17 07:20:45.658 7199-7244/com.ae.dnight D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=4 device=E6:F6:05:7E:6A:5D
2019-09-17 07:20:45.675 7199-7244/com.ae.dnight I/BluetoothPeripheral: connected to 'E6:F6:05:7E:6A:5D' (BOND_NONE) in 1,6s
2019-09-17 07:20:45.691 7199-7199/com.ae.dnight D/BluetoothPeripheral: discovering services of 'null' with delay of 0 ms
2019-09-17 07:20:45.691 7199-7199/com.ae.dnight D/BluetoothGatt: discoverServices() - device: E6:F6:05:7E:6A:5D
2019-09-17 07:20:46.427 7199-7254/com.ae.dnight D/BluetoothGatt: onConnectionUpdated() - Device=E6:F6:05:7E:6A:5D interval=6 latency=0 timeout=500 status=0
2019-09-17 07:20:46.673 7199-7254/com.ae.dnight D/BluetoothGatt: onSearchComplete() = Device=E6:F6:05:7E:6A:5D Status=0
2019-09-17 07:20:46.678 7199-7254/com.ae.dnight I/BluetoothPeripheral: discovered 5 services for 'null'
2019-09-17 07:20:46.680 7199-7199/com.ae.dnight I/XBeaconManager: discovered services
2019-09-17 07:20:46.707 7199-7569/com.ae.dnight I/XBeaconManager$bluetoothCentralCallback$1$onConnectedPeripheral: XBeacon connected! Minor ID: 38! Updating...
2019-09-17 07:20:46.732 7199-7569/com.ae.dnight I/XBeaconManager: Restarting XBeacon Manager
[... 102 lines cut ...]
2019-09-17 07:24:21.184 7199-7244/com.ae.dnight D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=12 mScannerId=0
2019-09-17 07:24:21.197 7199-7199/com.ae.dnight I/BluetoothCentral: scan started
2019-09-17 07:24:52.723 7199-7234/com.ae.dnight I/ae.dnight: Background concurrent copying GC freed 198463(5MB) AllocSpace objects, 0(0B) LOS objects, 50% free, 5MB/11MB, paused 1.182ms total 135.478ms
2019-09-17 07:24:56.266 7199-7199/com.ae.dnight I/XBeaconManager$periodicRunnable: No XBeacon recognized in 30 seconds. Assuming scanner got stuck! Restart it manually!
2019-09-17 07:24:56.287 7199-7199/com.ae.dnight I/XBeaconManager: Restarting XBeacon Manager
2019-09-17 07:24:56.300 7199-7199/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:24:56.300 7199-7199/com.ae.dnight D/BluetoothLeScanner: Stop Scan with callback
2019-09-17 07:24:56.306 7199-7199/com.ae.dnight I/BluetoothCentral: scan stopped
2019-09-17 07:24:56.312 7199-7199/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:24:56.315 7199-7199/com.ae.dnight D/BluetoothAdapter: STATE_ON
2019-09-17 07:24:56.315 7199-7199/com.ae.dnight D/BluetoothLeScanner: Start Scan with callback
[... cut ...]
I also see that there's a message that as the device is not in cache, it's likely to fail connecting - which is weird, because, as I understand to push device into a cache you have to scan before, but my flow is that I do connection with device after scanning i.e. `onDiscoveredPeripheral()` callback.

1. Is there a possibility to enhance the connection (it seems to work worse with 9.0+ android, but it's okay so to say for lower OS versions)?
2. Is it bad to trust in connections of BLE devices as they fail so much? Or is there a way to improve that?

weliem, 2019-09-17:
Can you share relevant parts of your code?

Your log looks weird. It looks like you are calling removeBond() and clearServicesCache() directly after calling connect(). You shouldn't do that....

On Android 9 connecting should just work fine. You must be doing something wrong...

Maartinsh, 2019-09-17:
> Can you share relevant parts of your code?
> 
> Your log looks weird. It looks like you are calling removeBond() and clearServicesCache() directly after calling connect(). You shouldn't do that....
> 
> On Android 9 connecting should just work fine. You must be doing something wrong...

Removed both calls and updated with log. Same happens. I also added the code. 
So, the idea why I restart scanning after I get connection callback is that I thought that it would restart freshly the scanning process (as u can see in the end I stop and start scanning but do not receive any results after connection is successful). Is that bad approach? Any ideas what could work in my scenerio? Thanks in advance.

weliem, 2019-09-17:
First of all, get rid of all the delays. These are blocking your thread and lead to unexpected behavior.

Second you are issuing multiple connect calls for the same device. See warnings in the log.

Third, you are stop and starting scans too quickly. Stopping a scan is an asynchronous method and you have to wait a bit to start scanning again.

You are calling stopScan even when you are not scanning. In restart()....

Maartinsh, 2019-09-17:
Delays were intended to make a pause between stop and start scanning. Obviously, it didn't work as expected. I removed them.

> You are calling stopScan even when you are not scanning. In restart()....

Yes. As I understand, there's no way of checking if there's a scan in progress at the time, am I right? 

> Third, you are stop and starting scans too quickly. Stopping a scan is an asynchronous method and you have to wait a bit to start scanning again.

Yes, I had a doubts about this one. In restart() I added this code fragment for testing:

       ```
central.stopScan()
 handler.postDelayed({

            with (central) {
                startPairingPopupHack()
                scanForPeripherals()
            }

        }, 5000)
```
By the logs I can see that between stop and start there's 5 seconds. Is this pause big enough for normal behaviour?

Log updated,

Maartinsh, 2019-09-17:
Tips resolved this issue.

closed 2019-09-17 by Maartinsh