weliem/blessed-android #14: Make BluetoothPeripheral parceable

moffa90, 2019-10-25:
When i do scan i tried to pass an arraylist of BluetoothPeripheral through intent extra but give error, saying: 

 java.lang.RuntimeException: Parcel: unable to marshal value com.welie.blessed.BluetoothPeripheral@4cb318e
        at android.os.Parcel.writeValue(Parcel.java:1680)
        at android.os.Parcel.writeList(Parcel.java:979)
        at android.os.Parcel.writeValue(Parcel.java:1627)
        at android.os.Parcel.writeArrayMapInternal(Parcel.java:891)
        at android.os.BaseBundle.writeToParcelInner(BaseBundle.java:1579)
        at android.os.Bundle.writeToParcel(Bundle.java:1233)
        at android.os.Parcel.writeBundle(Parcel.java:931)
        at android.content.Intent.writeToParcel(Intent.java:10646)
        at android.app.IActivityManager$Stub$Proxy.broadcastIntent(IActivityManager.java:3729)
        at android.app.ContextImpl.sendBroadcast(ContextImpl.java:1071)
        at android.content.ContextWrapper.sendBroadcast(ContextWrapper.java:449)

Maybe it would be good make that class parceable

weliem, 2019-10-25:
Interesting suggestion! I will look into it....

moffa90, 2019-10-25:
BTW good job with this... I feel BLESSED to found it... hahaha

weliem, 2019-10-29:
Hi...can I ask what you were trying to do? Were you trying to send discovered peripherals to an activity after a scan?

weliem, 2019-12-03:
I have looked into this but it is not a simple thing to do. I could make it work in the case where a device isn't connected yet. But if it is connected it wouldn't work anymore. There are several objects contained in the BluetoothPeripheral that are not parceable themselves....

I rather not implement something that 'only works under certain conditions'. Besides that, there are easy workarounds for this issue for you...I am sure you can think of some.

closed 2019-12-08 by weliem