weliem/blessed-android #15: Scan timeout

oliexdev, 2019-11-05:
Hi Martijn,

is it possible to implement a scan timeout? I have the following problem some devices needs to be turned on, otherwise no Bluetooth connection is possible. So I want to notify the user after a timeout. For a direct connection with `connectPeripheral` it works flawlessly because we have a 30s timeout but if I do `scanForPeripheralsWithAddresses` there is no timeout, it will result in a endless scan.

Best regards
oli

weliem, 2019-11-05:
Hi Oli, it is intentional scanning has no timeout. You can simply set your own timer in your application and call stop scanning. Right?

oliexdev, 2019-11-06:
You are absolutely right, I can mimic this behaviour in my app but that would be a nice option in your Bluetooth library :wink: What do you think?

weliem, 2019-12-03:
I have thought about it but I prefer to keep the interfaces simple/clean and not add an option for a timeout. In addition, doing a timeout on the application level is so simple I don't see any problem to do it there....

closed 2019-12-20 by weliem