weliem/blessed-android #16: Android 10 (Q) does not call BluetoothGattCallback.onConnectionStateChange when turning bluetooth OFF

marian-eremia, 2019-11-19:
On Android 10 (Q), the `BluetoothGattCallback.onConnectionStateChange` callback is never called as opposed to lower versions.

weliem, 2019-11-19:
@marian-eremia , thank you for the PR! I will look into this issue and get back to you....

weliem, 2019-11-19:
I just did a check on Android 9, but I also don't see `onConnectionStateChange` being called...so it doesn't seem specific to Android 10.

On which phone with Android 9 or lower do you see it being called?

weliem, 2019-11-19:
Previous check was on a Nokia 8, and now I also tested on a Samsung S8. On the Samsung `onConnectionStateChange` is indeed called when switching off bluetooth...

So it looks this is a manufacturer specific issue rather than Android version dependent....

marian-eremia, 2019-11-20:
It seems to be manufacturer specific, if it is as you say, but I did tests on a Note 5 (with Android 7), a Huawei P20 (with Android 9) and some other two phones with either Android 7 or 8, and on all these devices `onConnectionStateChange` was called.

The on device with Android 10 on which I noticed the lack of `onConnectionStateChange` call was a Pixel 3a (which should be as standard Android as they should come).

marian-eremia, 2019-11-20:
Either way, in my opinion, a fix should be implemented as, if this is a manufacturer specific issue, then the problem is broader because the library ends up in an unusable state (there is no workaround at this point that can clean-up its state after a bluetooth restart if the `onConnectionStateChange` callback isn't called by the bluetooth stack.

weliem, 2019-11-20:
I totally agree that this issue should be fixed! I would only like to make sure it works on all devices...so I am thinking about how to do that...

marian-eremia, 2019-11-20:
After more investigations, I too came to the conclusion it is device specific this issue.

The callbacks seem to come from the `IBluetoothGatt` implementation, which is most probably vendor specific.

Log from Galaxy Note 5 (Android 7)
```
D/BluetoothAdapter: ondisableBLE
D/BluetoothAdapter: There are no active google scan apps, stop scan
D/BluetoothLeScanner: stopAllScan standalone boolean is value is = false
D/BluetoothLeScanner: Exiting stopAllScan
D/BluetoothGatt: onBluetoothStateChange: up=false
D/BluetoothGatt: Bluetooth is turned off, disconnect all client connections
D/BluetoothGatt: cancelOpen() - device: F7:21:AA:BB:25:XX
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=9 device=F7:21:A
I/BluetoothPeripheral: disconnected 'DEVICE' on request
D/BluetoothAdapter: onBluetoothStateChange: up=false
D/BluetoothAdapter: Bluetooth is turned off, stop adv
D/BluetoothGatt: close()
```

weliem, 2019-11-20:
I just pushed a fix. Can you you test it?

...it uses a timer to detect if the phone does it itself or not. If not it disconnects cleanly and leaving everything in the proper state again.

marian-eremia, 2019-11-21:
Yes! It works perfectly. Thank you!

closed 2019-11-21 by marian-eremia