weliem/blessed-android #19: Disconnecting from a device

pumpkintechlabs, 2019-12-18:
How would you properly disconnect from a periferal?

`central.cancelConnection(periferal)`

disconnects from the device, but several seconds later device reconnects by itself.
Is that a bug or am i doing something wrong?

Cheers

weliem, 2019-12-19:
Calling cancelConnection is the right way to disconnect and should not reconnect itself.

However, in the demo application that is part of Blessed I call an autoConnect after a disconnect. Might that be what you are seeing?

It is this bit of code that does that:

        public void onDisconnectedPeripheral(final BluetoothPeripheral peripheral, final int status) {
            Timber.i("disconnected '%s' with status %d", peripheral.getName(), status);

            // Reconnect to this device when it becomes available again
            handler.postDelayed(new Runnable() {
                @Override
                public void run() {
                    central.autoConnectPeripheral(peripheral, peripheralCallback);
                }
            }, 5000);
        }

weliem, 2020-01-09:
Any update on this issue?

closed 2020-01-18 by weliem