weliem/blessed-android #20: Scan return null

animecomico, 2019-12-20:
Hi, Whe Scan always return name of device null and no connect.

Test in Android 9

weliem, 2019-12-20:
Can you share some code so that I can understand what you are doing?

animecomico, 2019-12-20:
Hi,
I modidy the BluetoothHandler for my device
package com.scarab.blespot_2;

import android.bluetooth.BluetoothGattCharacteristic;
import android.bluetooth.le.ScanResult;
import android.content.Context;
import android.content.Intent;
import android.os.Handler;

import java.util.Calendar;
import java.util.Date;
import java.util.UUID;

import com.scarab.manblue.BluetoothBytesParser;
import com.scarab.manblue.BluetoothCentral;
import com.scarab.manblue.BluetoothCentralCallback;
import com.scarab.manblue.BluetoothPeripheral;
import com.scarab.manblue.BluetoothPeripheralCallback;

import timber.log.Timber;

import static android.bluetooth.BluetoothGatt.CONNECTION_PRIORITY_HIGH;
import static com.scarab.manblue.BluetoothBytesParser.bytes2String;
import static com.scarab.manblue.BluetoothPeripheral.GATT_SUCCESS;

public class BluetoothHandler {
    // Spot AG Service
    private static final UUID SPOT_AG_SERVICE = UUID.fromString("00008601-1D13-34F8-B855-418FFE6B82E3");
    private static final UUID SECURITY_CHARACTERISTIC_UUID = UUID.fromString("00008602-1D13-34F8-B855-418FFE6B82E3");
    private static final UUID STATUS_CHARACTERISTIC_UUID = UUID.fromString("00008603-1D13-34F8-B855-418FFE6B82E3");
    private static final UUID CONTROL_CHARACTERISTIC_UUID = UUID.fromString("00008604-1D13-34F8-B855-418FFE6B82E3");
    private static final UUID DATA_CHARACTERISTIC_UUID = UUID.fromString("00008605-1D13-34F8-B855-418FFE6B82E3");
    private static final UUID CLI_CHARACTERISTIC_UUID = UUID.fromString("00008606-1D13-34F8-B855-418FFE6B82E3");
    //FANSTELS ONLY TEST
    private static final UUID UART_FANSTEL_UUID = UUID.fromString("6e400001-b5a3-f393-e0a9-e50e24dcca9e");
    private static final UUID UART_RX_CHARACTERISTIC_UUID = UUID.fromString("6e400002-b5a3-f393-e0a9-e50e24dcca9e");
    private static final UUID UART_TX_CHARACTERISTIC_UUID = UUID.fromString("6e400003-b5a3-f393-e0a9-e50e24dcca9e");
    //
    private Context context;
    private BluetoothCentral central;
    private Handler handler = new Handler();
    private int currentTimeCounter = 0;
    private static BluetoothHandler instance = null;
    //
    private final BluetoothPeripheralCallback peripheralCallback = new BluetoothPeripheralCallback() {
        @Override
        public void onServicesDiscovered(final BluetoothPeripheral peripheral) {
            Timber.i("discovered services");

            // Request a new connection priority
            peripheral.requestConnectionPriority(CONNECTION_PRIORITY_HIGH);

            // Turn on notifications for Data Characteristic
            if(peripheral.getService(DATA_CHARACTERISTIC_UUID) != null) {
                peripheral.setNotify(peripheral.getCharacteristic(SPOT_AG_SERVICE, DATA_CHARACTERISTIC_UUID), true);
            }

            // Turn on notifications for Status Characteristic
            if(peripheral.getService(STATUS_CHARACTERISTIC_UUID) != null) {
                peripheral.setNotify(peripheral.getCharacteristic(SPOT_AG_SERVICE, STATUS_CHARACTERISTIC_UUID), true);
            }
            // For TEST with UART Fanstel
            // Turn on notification for Health Thermometer Service
            if(peripheral.getService(UART_TX_CHARACTERISTIC_UUID) != null) {
                Timber.i("discovered services UART TX");
                peripheral.setNotify(peripheral.getCharacteristic(UART_FANSTEL_UUID, UART_TX_CHARACTERISTIC_UUID), true);
            }
        }

[... 98 lines cut ...]
//        central.startPairingPopupHack();
//        central.scanForPeripheralsWithServices(new UUID[]{SPOT_AG_SERVICE,
//                STATUS_CHARACTERISTIC_UUID,
//                UART_FANSTEL_UUID});
//    }
    public BluetoothHandler(Context context){
        this.context = context;
        central = new BluetoothCentral(context, bluetoothCentralCallback, new Handler());
    }

    public void onScan(){
        // Scan for peripherals with a certain service UUIDs
        central.startPairingPopupHack();
        central.scanForPeripherals();
        Timber.i("Start Scan");
//        central.scanForPeripheralsWithServices(new UUID[]{SPOT_AG_SERVICE,
//                STATUS_CHARACTERISTIC_UUID,
//                UART_FANSTEL_UUID});
    }

    public void offScan(){
        central.stopScan();
        Timber.i("Stop Scan");
    }
}

animecomico, 2019-12-20:
This is the MainActivity

package com.scarab.blespot_2;

import androidx.appcompat.app.AppCompatActivity;
import androidx.core.app.ActivityCompat;

import android.Manifest;
import android.content.Context;
import android.content.pm.ApplicationInfo;
import android.content.pm.PackageManager;
import android.os.Bundle;
import android.text.method.ScrollingMovementMethod;
import android.view.View;
import android.widget.Button;
import android.widget.TextView;

import com.scarab.manblue.BuildConfig;

import timber.log.Timber;

public class MainActivity extends AppCompatActivity {

    private TextView txtDbg;
    private  Button btnScan;
    private  Button btnConnect;
    private  Button btnTest;
    private BluetoothHandler blueman;
    //
    String[] PERMISSIONS = {
            Manifest.permission.BLUETOOTH,
            Manifest.permission.BLUETOOTH_ADMIN,
            Manifest.permission.ACCESS_COARSE_LOCATION,
            Manifest.permission.ACCESS_FINE_LOCATION,
            Manifest.permission.UPDATE_DEVICE_STATS
    };
    int PERMISSION_ALL = 1;

    boolean isDebug = false;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        if (!hasPermissions(this, PERMISSIONS)) {
            ActivityCompat.requestPermissions(this, PERMISSIONS, PERMISSION_ALL);
        }

        isDebug = ((this.getApplicationInfo().flags &
                ApplicationInfo.FLAG_DEBUGGABLE) != 0);

        if(isDebug){
            Timber.plant(new Timber.DebugTree());
        }
        //
        blueman = new BluetoothHandler(getApplicationContext());

        txtDbg = findViewById(R.id.textView);
        txtDbg.setMovementMethod(new ScrollingMovementMethod());
        //
        btnScan = findViewById(R.id.button);
        btnConnect = findViewById(R.id.button2);
        btnTest = findViewById(R.id.button3);
        //
        btnScan.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                if(btnScan.getText().toString().equals("SCAN")){
                    txtDbg.append("Scan\n");
                    //btnScan.setText("STOP SCAN");
                    blueman.onScan();
                }else{
                    txtDbg.append("Stop Scan\n");
                    btnScan.setText("SCAN");
                    blueman.offScan();
                }
            }
        });
        //
        btnConnect.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                txtDbg.append("Connect\n");
            }
        });
        //
        btnTest.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                txtDbg.append("Test\n");
            }
        });
    }

    public static boolean hasPermissions(Context context, String... permissions) {
        if (context != null && permissions != null) {
            for (String permission : permissions) {
                if (ActivityCompat.checkSelfPermission(context, permission) != PackageManager.PERMISSION_GRANTED) {
                    return false;
                }
            }
        }
        return true;
    }
}

weliem, 2019-12-20:
Well, there are a couple of strange things in your code:

1.  You start a scan **without any filtering** whatsoever. So you will get many results and will see every bluetooth peripheral around you. Many of them will be mobile phones that don't have a name and will give you 'null' as the name.
2. On the **first result** you stop the scan, and you then connect to the first device you find. This is probably not the peripheral you are interested in...

So my suggestion would be to start a scan that will only find your peripheral. I don't know your device but you could try scanning with a service UUID filter that would only let your peripheral through. Something like this:

```
public void onScan(){
    // Scan for peripherals with a certain service UUIDs
    central.startPairingPopupHack();
     central.scanForPeripheralsWithServices(new UUID[]{SPOT_AG_SERVICE});
    Timber.i("Start Scan");
```

That would start a scan and only let through peripherals that have the service SPOT_AG_SERVICE.

weliem, 2020-01-09:
Any update on this issue?

closed 2020-01-18 by weliem