weliem/blessed-android #21: Walking out and back in range doesn't trigger onCharacteristicUpdate

be-bert, 2020-01-08:
Upon initial connection, `onCharacteristicUpdate` is called and I can see the data flowing thru.

When I walk out of range, `onDisconnectedPeripheral` is called.

When I walk back into range, `onConnectedPeripheral` is called, `onServicesDiscovered` is then called, and I again call `peripheral.setNotify(myCharacteristic, true);`.

`onCharacteristicUpdate` is not called again upon that reconnect.

Any ideas on how to fix this?

weliem, 2020-01-08:
Hi, can you share some more information? Perhaps some code? What device are
you using? What phone?

be-bert, 2020-01-08:
Sure, thanks for the response.  I am using your example app and have switched out the UUIDs to connect and stream data to a local BLE device.  On top of that, I've just commented out the services that were in there that were specific to your example.  Then Im just printing out the data instead of updating the labels.

The device Im using is a Nexus 7 with Android version 6.0.1 to test on.

weliem, 2020-01-09:
Hi, the Nexus 7 is notorious for its bluetooth problems. Can you try the same code on a different device?

be-bert, 2020-01-10:
Seems to be the same on a Samsung Galaxy6.  I noticed that when `onDisconnectedPeripheral` happens a status of 8 is returned.

weliem, 2020-01-11:
Can you post the log? Please include the first time and the second time when it fails...

be-bert, 2020-01-13:
Thanks for your help @weliem, here are the logs.

```
2020-01-13 08:27:45.023 6431-6438/com.welie.blessedexample I/art: Starting a blocking GC Instrumentation
2020-01-13 08:27:45.077 6431-6431/com.welie.blessedexample D/BluetoothAdapter: STATE_ON
2020-01-13 08:27:45.078 6431-6431/com.welie.blessedexample D/BluetoothAdapter: STATE_ON
2020-01-13 08:27:45.078 6431-6431/com.welie.blessedexample D/BluetoothLeScanner: Start Scan
2020-01-13 08:27:45.079 6431-6431/com.welie.blessedexample D/BluetoothAdapter: STATE_ON
2020-01-13 08:27:45.079 6431-6431/com.welie.blessedexample D/BluetoothAdapter: STATE_ON
2020-01-13 08:27:45.085 6431-6431/com.welie.blessedexample D/BluetoothAdapter: STATE_ON
2020-01-13 08:27:45.086 6431-6431/com.welie.blessedexample D/BluetoothAdapter: STATE_ON
2020-01-13 08:27:45.216 6431-6443/com.welie.blessedexample D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=7 mClientIf=0
2020-01-13 08:27:45.237 6431-6431/com.welie.blessedexample I/BluetoothCentral: scan started
2020-01-13 08:27:45.307 6431-6431/com.welie.blessedexample D/ViewRootImpl@39df8df[MainActivity]: ThreadedRenderer.create() translucent=false
2020-01-13 08:27:45.317 6431-6431/com.welie.blessedexample D/InputTransport: Input channel constructed: fd=57
2020-01-13 08:27:45.318 6431-6431/com.welie.blessedexample D/ViewRootImpl@39df8df[MainActivity]: setView = DecorView@27b4ff5[MainActivity] touchMode=true
2020-01-13 08:27:45.330 6431-6431/com.welie.blessedexample D/ViewRootImpl@39df8df[MainActivity]: dispatchAttachedToWindow
2020-01-13 08:27:45.405 6431-6431/com.welie.blessedexample D/ViewRootImpl@39df8df[MainActivity]: Relayout returned: oldFrame=[0,0][0,0] newFrame=[0,0][1440,2560] result=0x27 surface={isValid=true 505764226560} surfaceGenerationChanged=true
2020-01-13 08:27:45.413 6431-6431/com.welie.blessedexample D/ViewRootImpl@39df8df[MainActivity]: mHardwareRenderer.initialize() mSurface={isValid=true 505764226560} hwInitialized=true
2020-01-13 08:27:45.428 6431-6444/com.welie.blessedexample D/ScanRecord: parseFromBytes
2020-01-13 08:27:45.458 6431-6451/com.welie.blessedexample D/libEGL: loaded /vendor/lib64/egl/libGLES_mali.so
2020-01-13 08:27:45.499 6431-6451/com.welie.blessedexample I/OpenGLRenderer: Initialized EGL, version 1.4
2020-01-13 08:27:45.499 6431-6451/com.welie.blessedexample D/OpenGLRenderer: Swap behavior 1
2020-01-13 08:27:45.506 6431-6451/com.welie.blessedexample D/mali_winsys: EGLint new_window_surface(egl_winsys_display*, void*, EGLSurface, EGLConfig, egl_winsys_surface**, egl_color_buffer_format*, EGLBoolean) returns 0x3000,  [1440x2560]-format:1
2020-01-13 08:27:45.507 6431-6431/com.welie.blessedexample D/ViewRootImpl@39df8df[MainActivity]: MSG_RESIZED_REPORT: ci=Rect(0, 96 - 0, 0) vi=Rect(0, 96 - 0, 0) or=1
2020-01-13 08:27:45.508 6431-6431/com.welie.blessedexample D/ViewRootImpl@39df8df[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 1
2020-01-13 08:27:45.508 6431-6431/com.welie.blessedexample D/ViewRootImpl@39df8df[MainActivity]: mHardwareRenderer.initializeIfNeeded()#2 mSurface={isValid=true 505764226560}
2020-01-13 08:27:45.511 6431-6431/com.welie.blessedexample I/InputMethodManager: [IMM] startInputInner - mService.startInputOrWindowGainedFocus
2020-01-13 08:27:45.521 6431-6444/com.welie.blessedexample D/InputTransport: Input channel constructed: fd=65
2020-01-13 08:27:46.306 6431-6443/com.welie.blessedexample D/ScanRecord: parseFromBytes

// ... Lots of parseFromBytes calls

2020-01-13 08:27:53.924 6431-6444/com.welie.blessedexample D/ScanRecord: parseFromBytes
2020-01-13 08:27:54.015 6431-6431/com.welie.blessedexample I/BluetoothHandler: Found peripheral 'VPS-A155'
2020-01-13 08:27:54.024 6431-6431/com.welie.blessedexample D/BluetoothAdapter: STATE_ON
2020-01-13 08:27:54.031 6431-6443/com.welie.blessedexample D/ScanRecord: parseFromBytes
2020-01-13 08:27:54.031 6431-6431/com.welie.blessedexample D/BluetoothAdapter: STATE_ON
2020-01-13 08:27:54.032 6431-6431/com.welie.blessedexample D/BluetoothLeScanner: Stop Scan
2020-01-13 08:27:54.048 6431-6431/com.welie.blessedexample I/BluetoothCentral: scan stopped
2020-01-13 08:27:54.182 6431-6431/com.welie.blessedexample I/BluetoothPeripheral: connect to 'VPS-A155' (04:91:62:65:A1:55) using TRANSPORT_LE
2020-01-13 08:27:54.207 6431-6431/com.welie.blessedexample D/BluetoothGatt: connect() - device: 04:91:62:65:A1:XX, auto: false
2020-01-13 08:27:54.208 6431-6431/com.welie.blessedexample D/BluetoothAdapter: isSecureModeEnabled
2020-01-13 08:27:54.214 6431-6431/com.welie.blessedexample D/BluetoothGatt: registerApp()
2020-01-13 08:27:54.220 6431-6431/com.welie.blessedexample D/BluetoothGatt: registerApp() - UUID=39d36fc5-5f8b-4c1a-ba54-faec5f19cb7b
2020-01-13 08:27:54.337 6431-6444/com.welie.blessedexample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2020-01-13 08:27:55.300 6431-6443/com.welie.blessedexample D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=7 device=04:91:62:65:A1:XX
2020-01-13 08:27:55.344 6431-6443/com.welie.blessedexample I/BluetoothPeripheral: connection failed with status 'GATT ERROR' (ERROR)
2020-01-13 08:27:55.344 6431-6443/com.welie.blessedexample D/BluetoothGatt: close()
2020-01-13 08:27:55.353 6431-6443/com.welie.blessedexample D/BluetoothGatt: unregisterApp() - mClientIf=7
2020-01-13 08:27:55.368 6431-6443/com.welie.blessedexample I/BluetoothCentral: retrying connection to 'VPS-A155' (04:91:62:65:A1:55)
[... cut ...]
2020-01-13 08:30:35.302 6431-6431/com.welie.blessedexample D/BluetoothGatt: setCharacteristicNotification() - uuid: 00035b03-58e6-07dd-021a-08123a000301 enable: true
2020-01-13 08:30:35.450 6431-6431/com.welie.blessedexample I/BluetoothHandler: SUCCESS: Notify set to 'on' for 00035b03-58e6-07dd-021a-08123a000301
2020-01-13 08:30:35.656 6431-6491/com.welie.blessedexample D/BluetoothGatt: onClientConnParamsChanged() - Device=04:91:62:65:A1:XX interval=12 status=0

// Now it is connected, but `onCharacteristicUpdate` is never called.
```

weliem, 2020-01-13:
Ok, it all looks normal...apart from the fact that the peripheral doesn't seem to start notifying.

Setting of the descriptor succeeds:
```
2020-01-13 08:30:35.302 6431-6431/com.welie.blessedexample D/BluetoothGatt: setCharacteristicNotification() - uuid: 00035b03-58e6-07dd-021a-08123a000301 enable: true
2020-01-13 08:30:35.450 6431-6431/com.welie.blessedexample I/BluetoothHandler: SUCCESS: Notify set to 'on' for 00035b03-58e6-07dd-021a-08123a000301
```

So then it is really up to the peripheral to start sending the notifications....is the peripheral a device you are making? It really looks like a device issue because it is impossible that the library is at fault. When a notification comes in I send it through immediately...

One potential workaround you could try is to first do a setNotify(xxx, false) to turn it off explicitly and then immediately turn it on setNotify(xxx, true). Perhaps that will trick the device into sending notifications again.

be-bert, 2020-01-16:
It could be an issue with the BLE device.  We noticed it begins streaming every other disconnect/reconnect cycle.  It seems if we completely tear down the and re-initialize the bluetooth handler it will work.  Still have to run some more scenarios to confirm that though.  Thanks for your help.

weliem, 2020-01-18:
I tried to reproduce the issue using 2 different phones and a heartrate sensor. But in all cases it worked fine....

Are you saying that if you destroy the entire BluetoothHandler singleton and re-create it, it will work in your case?

weliem, 2020-02-03:
Any update on this issue?

closed 2020-03-01 by weliem