weliem/blessed-android #22: Bluetooth On/Off and reconnections

BluestormDNA, 2020-01-09:
First of all, congratulations on your library. It certainly makes working with Ble much easier.

On the topic:

My use case relies on sending specific packets to the ble device before succesfully disconnecting (I use its not a rare case) so i need to take in account some user behaviour like turning off BT on purpose to try to turn it on again and reconnect.

Have you considered exposing the On/Off status from the adapterStateReceiver to the BluetoothCentralCallback?

weliem, 2020-01-09:
Hi, you can just use the regular Android adapter classes to register a broadcast receiver and get the current state in your app. It wont affect Blessed code...

BluestormDNA, 2020-01-09:
Yeah after thinking about the possibilities and before implementing my own broadcastReceiver i realized that it was easier to fork and add 2 lines to the central. Then i thought that maybe it wasnt an obscure corner case and that it made sense (to me) that maybe the library as a ble library itself maybe should have the BT status exposed.

It is ok if you dont want to expose it for wathever reason it was only a suggesion,

Keep the good work!

weliem, 2020-01-09:
It is certainly not a strange idea. Can you share how you solved it? If it is really 2 lines I could probably add it...

BluestormDNA, 2020-01-10:
It ended being 5 lines of code because i went ahead and couldnt resist adding also the transient states...

weliem, 2020-01-10:
Thanks for the PR! I accepted it and made a couple improvements.

Version 1.2 has now been released.

BluestormDNA, 2020-01-10:
Thanks! closing!

closed 2020-01-10 by BluestormDNA