weliem/blessed-android #24: scanForPeripheralsWithServices does never return any result

dilmash, 2020-01-15:
Martijn, I am trying to use your code to connect to a peripheral device with a specific service uuid. I installed your code as is and modified the scan initiation as follows:
```

    static private String HSS_GATE_SERVICE              = "79ceffa8-cb5c-4b03-a6c5-8e8282928067";
    final  private static UUID UUID_HSS_GATE_SERVICE    = UUID.fromString(HSS_GATE_SERVICE);
    private BluetoothHandler(Context context) {
        this.context = context;

        // Create BluetoothCentral
        central = new BluetoothCentral(context, bluetoothCentralCallback, new Handler());

        // Scan for peripherals with a certain service UUIDs
        central.startPairingPopupHack();
        central.scanForPeripheralsWithServices(new UUID[]{UUID_HSS_GATE_SERVICE});
        // central.scanForPeripherals();
    }
```

The device (only one) is definitely ready for connection but there is no response to this call. If I scan by MAC or without a filter, I get the response.

Do have an idea what needs to be done to scan for service UUID's

weliem, 2020-01-15:
In order for the scanbyUUID to work the peripheral needs to specify the UUID in the advertisement packages it sends out. In order words, what it advertises can be something totally different than what it has.

The easiest way to check if it advertises the UUID is to use an app like 'nRF Connect' which can list all the fields that are advertised.

dilmash, 2020-01-19:
Thanks for responding. The peripheral definitely does not advertise it's service UUID's. I changed the the code to search for any devices and check names and MAC instead.

weliem, 2020-01-20:
OK, that explains it then. 

Instead of checking names or mac address yourself you can also use the methods `scanForPeripheralsWithAddresses` or `scanForPeripheralsWithNames`.

Closing issue

closed 2020-01-20 by weliem