weliem/blessed-android #26: connectPeripheral without stopping the scan

ahimsadas, 2020-01-21:
Hi,

I'm quite new to BLE and Android development so please excuse my naiveness!

I'm trying to read a string stored as a characteristic value being advertised by an iOS device.
The [issue](https://stackoverflow.com/q/58864679/9971602) is the app on the iOS device is in the background state and as the Apple Core Bluetooth documentation goes, it doesn't advertise the local name or any service UUID.

So I decided to connect with every Apple device (filtered by the manufacturer identifier), look for the specific **SERVICE_UUID**, then look for the **CHARACTERISTIC_UUID**, and then finally read the characteristic value. Which works pretty neatly using BLESSED actually!

The requirement is that I read and show the string value in 'near' realtime, as and when any iOS peripheral is discovered. So I couldn't start and stop the scan every time a peripheral is discovered, because of this...

> [As of Android 7, starting and stopping scans more than 5 times in a window of 30 seconds will temporarily disable scanning.](https://blog.classycode.com/undocumented-android-7-ble-behavior-changes-d1a9bd87d983)

Looking at my log, I _think_ calling **connectPeripheral** multiple times (once for every **onDiscoveredPeripheral** callback) without stopping the scan overrides some of the connections, leaves some hanging without reaching the **onCharacteristicUpdate** callback and causes some havoc.

I have called for **cancelConnection** in the **onCharacteristicUpdate** callback after reading the value.

I do get the values I need but I just want to make sure that the process is clean. The log maybe showing me 2-3 different connection threads together, mixing them up?!

Does the overriding / overlapping happen technically? Do I need to queue each **connectPeripheral**?
Does the "Queueing commands" feature of BLESSED take care of **connectPeripheral**?

weliem, 2020-01-21:
Indeed, you cannot call **connectPeripheral** multiple times as they will start cancelling out each other. This is due to an Android limitation. However, this limitation does not exist for **autoConnectPeripheral**. I will make this more explicit in the documentation.

If you keep scanning while connecting it would be best if you at least manage what devices you are trying to connect to so that you don't try to call connect multiple times for the same device. Blessed has some protections against it, but it is good practice not to do it in the first place. You will receive multiple callbacks on onDiscoveredPeripheral if the device advertises fast...

And yes, in your case I would also queue the connectPeripheral commands....Blessed doesn't queue them.

ahimsadas, 2020-01-23:
Thanks so much for your quick response :)
I am implementing a queue.

Something like in this [video](https://www.stkent.com/2017/09/18/ble-on-android.html) (from 31:06), the link I found from your Medium post!

closed 2020-02-03 by weliem