weliem/blessed-android #27: Connection failed with status 8

Maartinsh, 2020-01-31:
Hi, I'll drop some logs that I got that might give you some idea of what's happenning:

> Starting scanner
> Device is recognized
> Scanning stopped
> Manually connecting device X
> Manually connecting device X
> Manually connecting device X
> $bluetoothCentralCallback: connection 'X' failed with status 8
> Manually connecting device X
> Manually connecting device X
> $bluetoothCentralCallback: connection 'X' failed with status 8
> Manually connecting device X

I know that you wrote in your blog post that this can be normal, but is there anything that I can do? You mentioned that I could do a clean-up. Can you please show me an example of that?
I also can see that I'm doing manual connection more than once (it's with few seconds interval) - could this cause this error? Obv, this I could fix. 

> Manually connecting:

`central.connectPeripheral(peripheral, peripheralCallback)`

weliem, 2020-02-03:
Status 8 is a timeout. So apparently the connection times out, which could have many reasons....

But why are you connecting all the time to the same device X? Connecting once should be enough, right?

Maartinsh, 2020-02-10:
Yes, I fixed on my end the multiple connection thing - it should do it only once, but that did not solve the problem. Sometimes with some devices it gives me status 8. What's the best thing to do when that happens? Could clearing BLE cache help?

weliem, 2020-03-01:
Every connection can have a timeout. If you think you should stay connected you can reconnect again. But it seems the device thinks there is no reason to be connected...

weliem, 2020-04-04:
closing due to inactivity...

closed 2020-04-04 by weliem