weliem/blessed-android #28: How to determine which peripheral service UUID when peripheral is discovered

yosoycody, 2020-02-08:
I'm scanning for bluetooth devices with 3 different service UUIDs (cycling power, fitness machine, and heart rate monitors). In the onDiscoveredPeripheral() method, I need to differentiate which service UUID the peripheral belongs to. I don't see a way to do that. peripheral.getService(UUID) returns null in all cases. Is there something else I need to do to see the advertising UUID for each peripheral when it's discovered?

yosoycody, 2020-02-08:
I think I just found the answer to my question, but correct me if there's a better way. It looks like I can read the value from the ScanResult object in the onDiscoveredPeripheral() method, like this: scanResult.getScanRecord().getServiceUuids()

weliem, 2020-02-09:
Hi, yes indeed. You need to use the ScanResult object and get the service
uuid from there...

closed 2020-02-09 by yosoycody