weliem/blessed-android #30: Not getting onCharacteristicUpdate callback with Bluno Beetle

yosoycody, 2020-03-02:
I have a Samsung Galaxy S7 running Android 8, and trying to read from a Bluno beetle using the following service UUID and characteristic UUIDs:
```
private static final UUID REWIRE_SERVICE_UUID = UUID.fromString("0000dfb0-0000-1000-8000-00805f9b34fb");
private static final UUID REWIRE_CHARACTERISTIC_UUID = UUID.fromString("0000dfb1-0000-1000-8000-00805f9b34fb");
```

It's able to connect to the device and discover the service. However, the characteristic callback is never called for this characteristic.

Within the onServicesDiscovered, I'm setting up the notification like this:
```
if (peripheral.getService(REWIRE_SERVICE_UUID) != null) {
   peripheral.setNotify(peripheral.getCharacteristic(REWIRE_SERVICE_UUID, REWIRE_CHARACTERISTIC_UUID), true);
}
```

The only way I can read the characteristic is if I force it like this:
`peripheral.readCharacteristic(peripheral.getCharacteristic(REWIRE_SERVICE_UUID, REWIRE_CHARACTERISTIC_UUID));`

But I need to get it automatically every time with notifications. I have an iOS app that's able to read the Bluno beetle fine, so something is different either with the Android device, or possibly the library. I'm also able to read heart rate monitors and other devices fine with the Blessed library. Any ideas?

yosoycody, 2020-03-02:
I see this in the logs: BluetoothPeripheral: could not get CCC descriptor for characteristic 0000dfb1-0000-1000-8000-00805f9b34fb

weliem, 2020-03-02:
Well, if the CCC descriptor cannot be found, then it doesn't support notifying....

yosoycody, 2020-03-02:
Wonder why iOS is able to get it. I see the following topics specifically about Bluno beetle and notification descriptor:
https://www.dfrobot.com/forum/viewtopic.php?t=2035
https://github.com/DFRobot/BLE_firmware_V1.9/issues/5

weliem, 2020-03-02:
I see on those pages that indeed the CCC descriptor is missing and that others are running into the same issue. So your device does not conform to the Bluetooth standard. 

You could try to fork Blessed and remove the checks on the CCC....perhaps it works then.

yosoycody, 2020-03-02:
Do you think I can just extend the BluetoothPeripheral class from Blessed and override the setNotify function?

weliem, 2020-03-02:
Don't think that'll work. You have to refactor the setNotify function so that it doesn't try to write the descriptor. It is not hard...

yosoycody, 2020-03-02:
Thanks for the help. Yeah, I was trying to avoid modifying the library but looks like I can't extend BluetoothPeripheral like you said.

yosoycody, 2020-03-03:
Got it to work doing what you said. Within the setNotify() function in BluetoothPeripheral, I updated the code where it checks to see if the CCC descriptor is null. If it is null, then I call this:
`bluetoothGatt.setCharacteristicNotification(characteristic, enable);` and return true. Else I proceed with the rest of the code as you had it.

closed 2020-03-04 by yosoycody