weliem/blessed-android #31: IntentReceiver leaking in BluetoothCentral

evstep, 2020-03-10:
> E/ActivityThread: Service com.blebackground.android.app.BleService has leaked IntentReceiver com.welie.blessed.BluetoothCentral$10@52c4df4 that was originally registered here. Are you missing a call to unregisterReceiver()?
>     android.app.IntentReceiverLeaked: Service com.blebackground.android.app.BleService has leaked IntentReceiver com.welie.blessed.BluetoothCentral$10@52c4df4 that was originally registered here. Are you missing a call to unregisterReceiver()?
>         at android.app.LoadedApk$ReceiverDispatcher.<init>(LoadedApk.java:1669)
>         at android.app.LoadedApk.getReceiverDispatcher(LoadedApk.java:1446)
>         at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1614)
>         at android.app.ContextImpl.registerReceiver(ContextImpl.java:1587)
>         at android.app.ContextImpl.registerReceiver(ContextImpl.java:1575)
>         at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:627)
>         at com.welie.blessed.BluetoothCentral.<init>(BluetoothCentral.java:337)
>         at com.blebackground.android.app.BleService.onCreate(BleService.kt:57)
>         at android.app.ActivityThread.handleCreateService(ActivityThread.java:4521)
>         at android.app.ActivityThread.access$2600(ActivityThread.java:296)
>         at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2243)
>         at android.os.Handler.dispatchMessage(Handler.java:107)
>         at android.os.Looper.loop(Looper.java:213)
>         at android.app.ActivityThread.main(ActivityThread.java:8178)
>         at java.lang.reflect.Method.invoke(Native Method)
>         at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
>         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1101)
Looks like we need some api to stop broadcast receivers to prevent leaking. I think it's caused because  `adapterStateReceiver` is never unregistered.

weliem, 2020-03-10:
Just added a method close() to clean up, see acd5179

Does that work for you?

evstep, 2020-03-10:
Seems good to me, thank you.

closed 2020-03-10 by evstep