weliem/blessed-android #33: Best method for writing to characteristic outside of callbacks

ptran49, 2020-03-26:
Hi Martijn,
I have been looking through your example app and I noticed that you only write to characteristics through callbacks in BluetoothHandler.java. If I want to write to a characteristic outside of a callback, like in response to a user pressing a button, what would be the best way to go about doing that? Currently, the best solution that I have thought of is to create a function within BluetoothHandler.java that uses getConnectedPeripherals, select the peripheral I want, and then use writeCharacteristic. Any help would be appreciated.

weliem, 2020-03-26:
Hi, what you are doing is fine. You just need a reference to a peripheral. So you can use getConnectedPeripherals(), or getPeripheral(), or just keep track of it yourself.

closed 2020-03-26 by ptran49

Koxx3, 2020-09-08:
Hello,
I would be very interested by an example ;)
would it be possible please ?

ptran49, 2020-09-13:
Hi,
Sorry for the late response. The way I did it was create a custom function inside my own BluetoothHandler class:
`fun writeCharacteristic(command: ByteArray){
        macAddress?.let { mac ->
            val peripheral = central.getPeripheral(mac)
            Timber.d("MAC Address is valid")
            if (peripheral.getService(ROBO_BLE_UUID) != null) {
                Timber.d("Service is valid")
                peripheral.getCharacteristic(
                    ROBO_BLE_UUID,
                    ROBO_BLE_TXCHARACTERISTIC_UUID
                ).let { peri ->
                    Timber.d("Writing characteristic")
                    peri?.let {
                        peripheral.writeCharacteristic(it, command, BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT)
                    }
                }
            }
        }
    }`

I use the MAC address of the peripheral (which is known by me since I know what I am interfacing with, then I get the characteristic that I want from the peripheral (also known), and then I directly use writeCharacteristic() for the BLEPeripheral class.