weliem/blessed-android #34: No data returned - can't find device

nedimf, 2020-03-30:
I have been trying to implement your library to a project but having problem with getting any devices found?

```java
    BluetoothCentral central;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        central = new BluetoothCentral(getApplicationContext(), bluetoothCentralCallback, new Handler(Looper.getMainLooper()));

        central.scanForPeripherals();

    }
    private final BluetoothCentralCallback bluetoothCentralCallback = new BluetoothCentralCallback() {
        @Override
        public void onDiscoveredPeripheral(BluetoothPeripheral peripheral, ScanResult scanResult) {
            Timber.d(peripheral.getAddress());

        }
    };
    private final BluetoothPeripheralCallback peripheralCallback = new BluetoothPeripheralCallback() {
        @Override
        public void onServicesDiscovered(BluetoothPeripheral peripheral) {
            super.onServicesDiscovered(peripheral);
            Timber.d(peripheral.getAddress());
        }
    };

```

nedimf, 2020-03-30:
Seems like problem was at permission enabling on device. I had to give permission for location service to work. One more question before I close this, what is ```scanResult``` returning, I'm having issue to get any scanned results still (I can only access ones that are called by peripheralCallback).

nedimf, 2020-03-30:
I'm not using this library anymore, so I'm closing the issue.
P.S
Never solve it. But my investigation showed ```checkForPermission > 23```is needed.

```java
    @RequiresApi(api = Build.VERSION_CODES.M)
    private void checkBTPermissions() {
        if(Build.VERSION.SDK_INT > Build.VERSION_CODES.LOLLIPOP){
            int permissionCheck = this.checkSelfPermission("Manifest.permission.ACCESS_FINE_LOCATION");
            permissionCheck += this.checkSelfPermission("Manifest.permission.ACCESS_COARSE_LOCATION");
            if (permissionCheck != 0) {

                this.requestPermissions(new String[]{Manifest.permission.ACCESS_FINE_LOCATION, Manifest.permission.ACCESS_COARSE_LOCATION}, 1001); //Any number
            }
        }else{
            Log.d("TAGG", "checkBTPermissions: No need to check permissions. SDK version < LOLLIPOP.");
        }
    }

```

closed 2020-03-30 by nedimf

weliem, 2020-03-30:
Maybe you should check out the testapp that is part of this repo and familiarize yourself with that. That shows how to scan and connect to devices as well as asking for the right permissions.