weliem/blessed-android #35: GATT CONN TERMINATE PEER USER error

daivatsoni, 2020-04-03:
Hello Team,

I am trying to extend your code for Blood Sugar reading. I have "Accu-Chek Instant" device. I have set Services and Characteristics in your code but it always gets disconnected. Find below the log:

===============
2020-04-03 18:29:04.594 10737-10737/com.welie.blessedexample D/BluetoothGatt: registerApp() - UUID=235986d8-9cd4-485d-bc21-fc2629117da7
2020-04-03 18:29:04.597 10737-10750/com.welie.blessedexample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
2020-04-03 18:29:12.762 10737-10750/com.welie.blessedexample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=90:9A:77:EC:53:1A
2020-04-03 18:29:12.768 10737-10750/com.welie.blessedexample I/BluetoothPeripheral: connected to 'meter+02252924' (BOND_BONDED) in 8.2s
2020-04-03 18:29:12.773 10737-10737/com.welie.blessedexample D/BluetoothPeripheral: discovering services of 'meter+02252924' with delay of 0 ms
2020-04-03 18:29:12.773 10737-10737/com.welie.blessedexample D/BluetoothGatt: discoverServices() - device: 90:9A:77:EC:53:1A
2020-04-03 18:29:13.028 10737-10750/com.welie.blessedexample D/BluetoothGatt: onConnectionUpdated() - Device=90:9A:77:EC:53:1A interval=6 latency=0 timeout=500 status=0
2020-04-03 18:29:13.555 10737-10750/com.welie.blessedexample D/BluetoothGatt: onSearchComplete() = Device=90:9A:77:EC:53:1A Status=0
2020-04-03 18:29:13.559 10737-10750/com.welie.blessedexample I/BluetoothPeripheral: discovered 4 services for 'meter+02252924'
2020-04-03 18:29:13.563 10737-10737/com.welie.blessedexample I/BluetoothHandler: connected to 'meter+02252924'
2020-04-03 18:29:13.564 10737-10737/com.welie.blessedexample I/BluetoothHandler: discovered services
2020-04-03 18:29:13.566 10737-10737/com.welie.blessedexample D/BluetoothGatt: requestConnectionPriority() - params: 1
2020-04-03 18:29:13.569 10737-10737/com.welie.blessedexample D/BluetoothPeripheral: requesting connection priority 1
2020-04-03 18:29:13.573 10737-10737/com.welie.blessedexample D/BluetoothPeripheral: reading characteristic <00002a29-0000-1000-8000-00805f9b34fb>
2020-04-03 18:29:13.587 10737-10737/com.welie.blessedexample I/BluetoothHandler: Received manufacturer: Roche
2020-04-03 18:29:13.590 10737-10737/com.welie.blessedexample D/BluetoothPeripheral: reading characteristic <00002a24-0000-1000-8000-00805f9b34fb>
2020-04-03 18:29:13.592 10737-10750/com.welie.blessedexample D/BluetoothGatt: onConnectionUpdated() - Device=90:9A:77:EC:53:1A interval=36 latency=0 timeout=500 status=0
2020-04-03 18:29:13.705 10737-10737/com.welie.blessedexample I/BluetoothHandler: Received modelnumber: 960
2020-04-03 18:29:13.705 10737-10737/com.welie.blessedexample D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a18-0000-1000-8000-00805f9b34fb enable: true
2020-04-03 18:29:13.795 10737-10737/com.welie.blessedexample I/BluetoothHandler: SUCCESS: Notify set to 'on' for 00002a18-0000-1000-8000-00805f9b34fb
2020-04-03 18:29:13.837 10737-10750/com.welie.blessedexample D/BluetoothGatt: onConnectionUpdated() - Device=90:9A:77:EC:53:1A interval=6 latency=0 timeout=2000 status=0
2020-04-03 18:29:33.336 10737-10750/com.welie.blessedexample D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=6 device=90:9A:77:EC:53:1A
2020-04-03 18:29:33.340 10737-10750/com.welie.blessedexample I/BluetoothPeripheral: peripheral 'meter+02252924' disconnected with status 'GATT CONN TERMINATE PEER USER'
2020-04-03 18:29:33.340 10737-10750/com.welie.blessedexample D/BluetoothGatt: close()
2020-04-03 18:29:33.341 10737-10750/com.welie.blessedexample D/BluetoothGatt: unregisterApp() - mClientIf=6
2020-04-03 18:29:33.364 10737-10737/com.welie.blessedexample I/BluetoothHandler: disconnected 'meter+02252924' with status 19
2020-04-03 18:29:38.376 10737-10737/com.welie.blessedexample I/BluetoothPeripheral: autoConnect to 'meter+02252924' (90:9A:77:EC:53:1A) using TRANSPORT_LE
2020-04-03 18:29:38.377 10737-10737/com.welie.blessedexample D/BluetoothGatt: connect() - device: 90:9A:77:EC:53:1A, auto: true
2020-04-03 18:29:38.377 10737-10737/com.welie.blessedexample D/BluetoothGatt: registerApp()
2020-04-03 18:29:38.378 10737-10737/com.welie.blessedexample D/BluetoothGatt: registerApp() - UUID=4b613ce0-bc83-49b6-8275-7dda7cf475fd
2020-04-03 18:29:38.380 10737-10750/com.welie.blessedexample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
===============
Let me know what's wrong going on.

Thank you so much in advance.

weliem, 2020-04-03:
Doesn't look like anything is wrong at all. You are reading the manufacturer and model number. Then you setNotify on the glucose measurement characteristic. Then you do nothing and after 20 seconds the accucheck disconnects due to inactivity....

What were you expecting to happen?

daivatsoni, 2020-04-06:
I want to have a Glucose value. I have set the Glucose service and measurement characteristics. I replicated the code which you wrote for other measurements. Now I am not sure what I have missed. Can I push changes by creating separate branch of it?

weliem, 2020-04-06:
I worked with a different AccuChek glucose meter in the past. It also didn't send measurements upon setting notify on. On the device there is an option somewhere in the menu's to 'auto send' measurement  after a measurement is taken. Perhaps yours has that option as well.

Nonetheless I ended up using the RACP characteristic to send a command so that it would send the measurements. With the RACP characteristic you can ask it to send some or all measurements.

daivatsoni, 2020-04-06:
Okay. Let me try to read measurements using the RACP characteristics. I will keep you posted on this. 
Thank you for your help.

daivatsoni, 2020-04-06:
I am a bit confused about where should I call readCharacteristic method. I tried to call it with setNotify() method. It responded with "characteristic does not have read property". So could you please help me with this?

weliem, 2020-05-07:
Hi...you should not use readCharacteristic on the Control Characteristic. You basically write commands to it like 'give me all records' and then you will receive them via the measurement characteristic...see the Bluetooth SIG documentation on how to build the command byte arrays

weliem, 2020-07-07:
Any update on this issue?

daivatsoni, 2020-07-07:
I'm sorry mate, my knowledge is lacking now. so no any update.

If you could help me in this, then that would be great.

Thank you.

-- 
[image: Hardy Infotech]
[image: Facebook] <https://www.facebook.com/hardyinfotech/> [image:
Instagram] <https://www.instagram.com/hardy.infotech> [image: Twitter]
<https://twitter.com/HardyInfotech> [image: LinkedIn]
<https://www.linkedin.com/company/hardy-infotech/>
Daivat Soni
Founder / Director
M +91 95-58-131100
E info@hardyinfotech.com
W www.hardyinfotech.com
E-1012, Titanium City Centre, 100 Feet Road, Anandnagar,
Satellite, Ahmedabad Gujarat Bharat - 380015
This message contains information intended only for the use of the
addressee named above. If you are not the intended recipient of this
communication please delete and destroy all copies and telephone Hardy
Infotech on +91 95-58-131100 immediately. If you are the intended recipient
of the communication you should not copy, disclose or distribute this
information without the authority of Hardy Infotech. Any views expressed in
the communication are those of the individual sender, except where the
sender specifically states them to be the views of Hardy Infotech. Except
as required by law, Hardy Infotech does not represent, warrant and/or
guarantee that the integrity of this communication has been maintained nor
that the communication is free of errors, virus, interception or
interference.

weliem, 2020-11-06:
I'll close this issue for now. If I have time I may dive into this and add some example code....

closed 2020-11-06 by weliem

weliem, 2021-01-02:
Hi, I added support for Glucose Meters in the example app.....in case you are still interested.