weliem/blessed-android #37: IllegalArgumentException: Receiver not registered

VladKeda, 2020-04-05:
Hello! Thanks for this hard work.
Your library works great. This makes using BLE easier for me.
But sometimes I come across this problem: java.lang.IllegalArgumentException: Receiver not registered.
One of several method sequences is as follows: BluetoothCentral.cancelAllConnectionsWhenBluetoothOff -> BluetoothPeripheral.disconnectWhenBluetoothOff -> BluetoothPeripheral.completeDisconnect -> android.content.ContextWrapper.unregisterReceiver -> android.app.ContextImpl.unregisterReceiver -> android.app.LoadedApk.forgetReceiverDispatcher -> java.lang.IllegalArgumentException: Receiver not registered.
Please can you investigate this case? Perhaps you just need to add a try/catch block to the BluetoothPeripheral.completeDisconnect method when calling Context.unregisterReceiver.

weliem, 2020-04-05:
Hi, yes that shouldn't happen. Looks like this happens when bluetooth is being switched off...

I'll have a look. Which phone and android version are you using?

VladKeda, 2020-04-05:
This happened once on a Samsung Galaxy S10 (Android 10).
And once on the Nexus 5X (Android 8.1), but with a different sequence of methods, which also leads to  BluetoothPeripheral.completeDisconnect -> android.content.ContextWrapper.unregisterReceiver -> android.app.ContextImpl.unregisterReceiver -> android.app.LoadedApk.forgetReceiverDispatcher -> java.lang.IllegalArgumentException: Receiver not registered.

VladKeda, 2020-04-05:
Also I can't reproduce this exception intentionally

weliem, 2020-04-10:
Hi, I did some reading on this. Apparently this can happen if Android cleans up the broadcast receivers before I do. For protection I added try/catch block.

Thanks for reporting this issue!

VladKeda, 2020-04-10:
Hi, thanks for your time and fix!

closed 2020-05-07 by weliem