weliem/blessed-android #39: Need explicit disconnection

SuperG33K, 2020-05-07:
Hello and thanks for this API !

We use a foreground service to record bluetooth data from multiple BLE devices and when we stop the recording, we need to explicitly disconnect from all devices. After calling central.close(), we continue to receive callbacks from updated characteristic, even when trying to stop thread and disconnecting callbacks (we are using our own callbacks on top on blessed callbacks)

It seems central.close() do not disconnect connected devices but only clear some device list and callbacks and unregister. Maybe we need an explicit disconnection to stop all BLE gatt/devices connection opened with blessed ?

weliem, 2020-05-07:
You can call cancelConnection() on every connected device. That would solve your issue.

weliem, 2020-05-10:
Did calling cancelConnection solve your problem?

SuperG33K, 2020-05-13:
Yes it seems to work.

However, I get a warning with this exception below.

`2020-05-13 15:33:07.391 23153-23192/com.scalian.octopus W/MessageQueue: Handler (android.os.Handler) {74ad000} sending message to a Handler on a dead thread
    java.lang.IllegalStateException: Handler (android.os.Handler) {74ad000} sending message to a Handler on a dead thread
        at android.os.MessageQueue.enqueueMessage(MessageQueue.java:558)
        at android.os.Handler.enqueueMessage(Handler.java:754)
        at android.os.Handler.sendMessageAtTime(Handler.java:703)
        at android.os.Handler.sendMessageDelayed(Handler.java:673)
        at android.os.Handler.post(Handler.java:403)
        at com.welie.blessed.BluetoothCentral$4.disconnected(BluetoothCentral.java:287)
        at com.welie.blessed.BluetoothPeripheral.completeDisconnect(BluetoothPeripheral.java:985)
        at com.welie.blessed.BluetoothPeripheral.successfullyDisconnected(BluetoothPeripheral.java:573)
        at com.welie.blessed.BluetoothPeripheral.access$600(BluetoothPeripheral.java:77)
        at com.welie.blessed.BluetoothPeripheral$1.onConnectionStateChange(BluetoothPeripheral.java:335)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:272)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:780)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:41)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:267)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:192)
        at android.os.Binder.execTransactInternal(Binder.java:1021)
        at android.os.Binder.execTransact(Binder.java:994)
`

I think it's probably because I use a handler thread to get all callbacks to a background thread and I close the thread before queue is empty so it's probably from my side

weliem, 2020-05-13:
Yes, you'll have to keep that thread alive a bit longer....

closed 2020-05-14 by SuperG33K

maartenvandillen, 2020-10-02:
Is there any way to remove the peripheral from the internal list? I would like to make use of the autoConnectPeripheral feature, but not when disconnected manually by the user.

weliem, 2020-10-02:
autoConnectPeripheral only 'works' once. After connecting and disconnecting, it won't autoconnect again unless you call autoConnectPeripheral again.

maartenvandillen, 2020-10-02:
Thanks, solved it by conditionally calling autoConnectPeripheral based on whether there was a user initiated disconnect or not.