weliem/blessed-android #42: Use writeCharacteristics from MainActivity

rsostar, 2020-07-07:
I am having trouble writing a characteristics from main activity. How can I get peripheral in MainActivity so I can use writeCharacteristics?

gocamiot, 2020-07-07:
I have the same issue. Please help with example code.

weliem, 2020-07-07:
Well, this is not really a question related to the library, right? You just have to make BluetoothCentral accessible to the MainActivity.

Assuming you are using the example app, you could add a method to the BluetoothHandler:

    public BluetoothCentral getCentral() {
        return central;
    }

Then you can call that from the MainActivity like this;

        BluetoothCentral central = BluetoothHandler.getInstance(getApplicationContext()).getCentral();

Using the central object you could ask for all connected peripherals:

        List<BluetoothPeripheral> peripheralList = central.getConnectedPeripherals();

Or ask for a specific peripheral by providing the device address:

        BluetoothPeripheral peripheral = central.getPeripheral(deviceAddress);

weliem, 2020-07-07:
But nevertheless, it is generally better to do all BLE operations in a dedicated class like the BluetoothHandler in the example. The main activity should generally only update the UI based on some sort of 'model' and should not do BLE operations.

gocamiot, 2020-07-08:
Thank you Martijn. I am going to try and use this example in the BluetoothHandler Class, like you suggested above, maybe add observer to update my recyclerviews. Like you did in onCharacteristicUpdate. I am a newby Android developer, and a soso ble firmware developer, trying to do my own wearable. A good example function to do some versions of writecharacteristic would be nice, to see what error handling and options one has using your libraries. (if any) Your libraries really help a lot. It is impressive and undervalued. Well done. And thank you for prompt response. :)

weliem, 2020-07-08:
There is code in the BluetoothHandler class of the example that does a write:

```
 BluetoothBytesParser parser = new BluetoothBytesParser();
 parser.setCurrentTime(Calendar.getInstance());
 peripheral.writeCharacteristic(currentTimeCharacteristic, parser.getValue(), WRITE_TYPE_DEFAULT);
```

After each write you will get a callback on `onCharacteristicWrite` with information about success or failure:

```
        public void onCharacteristicWrite(BluetoothPeripheral peripheral, byte[] value, BluetoothGattCharacteristic characteristic, int status) {
            if( status == GATT_SUCCESS) {
                Timber.i("SUCCESS: Writing <%s> to <%s>", bytes2String(value), characteristic.getUuid().toString());
            } else {
                Timber.i("ERROR: Failed writing <%s> to <%s>", bytes2String(value), characteristic.getUuid().toString());
            }
        }
```

Good luck!

gocamiot, 2020-07-09:
Great help. Thank you. I will give you some feedback.:)

gocamiot, 2020-07-10:
Hey Martijn. I hope you have the energy to help me again. :) So I found it the easy way to make an object in MainActivity of 'çentral'. Like you suggested above. I can then access the peripheral.address and print on screen. Everything works great. Now to make an object of chracteristic(like you did in BluetoothHandler class) from MainActivity from this centrel object, Im need some advice. Here is the function I wrote in MainActivity that gets called from recyclerview clicks, and this code works, so click and I get displayed the periphiral adress on screen. 
private fun tagItemClicked(tag : BleTag) {
            val parser = BluetoothBytesParser()
            val central = getInstance(applicationContext)!!.getCentral()
            //val characteristic = central.BluetoothGattCharacteristic.
            val peripheral: BluetoothPeripheral = central!!.getPeripheral(tag.tagAddress)
            parser.setString("0")
            //peripheral.writeCharacteristic(characteristic, parser.value, BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT)
            //peripheral.writeCharacteristic(central.smartLight_LED1_CHARACTERISTIC_UUID, parser.value, BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT)
            
            Toast.makeText(this, "Clicked: ${tag.name}-> ${tag.tagAddress}", Toast.LENGTH_LONG).show()

            when (tag.name) {
//                "Kitchen" -> {print("x == 1"); peripheral.writeCharacteristic(characteristic, parser.value, BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT)}
                "Tag01" -> {print("x == 1");}
                "Lounge" -> {print("x == 1")}
                "Bath room" -> {print("x == 1")}
                "Main room" -> {print("x == 1")}
                "Spare room" -> {print("x == 1")}
                else -> { // Note the block
                    print("x is neither 1 nor 2")
                }
            }
        }

I need to create and pass the correct "character" object in your writeCharacteristic. I would also like to check if writeable characteristic for error checcking purposes.

(I think  Ill be OK to broadcast with callback function after writeCharacteristic to update screen again)

I apologise for amateur code and help needed here. And I thank you again for any help. Jakes

gocamiot, 2020-07-13:
Hey Martijn. All working well. I just had to get another service, and getCharacteristic, and write. Thank you for help. Warm Regards. Jakes

weliem, 2020-07-17:
Great! Closing the issue...

closed 2020-07-17 by weliem