weliem/blessed-android #43: autoConnect() takes too long to connect with device

Maartinsh, 2020-07-08:
Hi, so I'm using your library and what I do is I create connections with BLE peripheral periodically. 

For first implementation I used `connect()` and it worked very well. Connection was made very fast. Then bug was found - if user toogled bluetooth on/off, bluetooth cache stack was cleared and then `connect()` did not work anymore. I went through your documentation and found that `autoConnect()` can be used and with this implementation toogling bluetooth on/off still worked, which was a blast. But then reports came in that sometimes autoConnect() takes up to 10 minutes, which is big delay for our logic. 

Question is, is there something that we can do? Is there a possibility to fix bluetooth toogle on/off cache clearing for manualConnection()? Or is there a way to reduce autoConnect() delay to minimum?

weliem, 2020-07-08:
When you toggle bluetooth on/off, Android itself clears the cached. That's what all Android phones do and there is nothing I can do about it. That is why I introduced the `autoConnect` function so that you wouldn't have to worry about it. That functions first scans for the device if it is not cached and otherwise issues an Android-level autoconnect.

It is true that this sometimes takes longer for getting a connecting, altough 10 min sounds strange. In my experience it may take up to 30 seconds but not longer. How long it takes usually depends on how often the peripheral advertises. With an advertisement interval of 200-400ms connections happen fast. But if the interval is > 1000ms it may take very long. I also found that some Huawei phones have problems with autoConnecting due to very heavy battery saving customizations Huaweu did...

Can you control the advertising interval of the peripheral you are using? If so, set it to the 200-400ms range.

Maartinsh, 2020-07-15:
I found out that we do not receive any callback method of connection status (onConnectedPeripheral/onConnectionFailed/onDisconnectedPeripheral is no called), when connection is establishing for longer period of time. What could cause this? I did notice that sometimes we try to call "connect" one or two times more before we have received any response from previous connection - could that somehow block the callback response (reason why we dont get any)? This happens in some rare scenerios (no specific pattern for now) though.

weliem, 2020-07-15:
You can call autoConnect for several peripherals, but there can be only 1 outstanding connectPeripheral call. The library doesnt prevent you from doing it. It only prevents multiple connects to the same peripheral...

Maartinsh, 2020-07-16:
Well, I've a 30 seconds timer in which I call connectPeripheral to single device. Usually second time it says that we are still waiting for connection - which you just confirmed. Problem is that it usually never timeouts - no response in hours, just a message that trying to do connection, but it's still waiting/pending.

weliem, 2020-07-17:
So are you saying that the connectPeripheral call never times out? There should be no need for having your own timer...

weliem, 2020-07-17:
Normally a connectPeripheral() will time out in 30 or less. Those are the Android values. Blessed has an extra timer that will time out after 35 seconds. So one way or another, after 35 seconds or less you should receive a ‘connectionFailed’ event...

Maartinsh, 2020-07-24:
Usually it starts with normal connection -> connection response logs like this:

`
07 21 07:48:29:183 BluetoothPeripheral: connection failed with status 'GATT ERROR' (TIMEOUT)
07 21 07:48:29:189 BluetoothCentral: connection to 'Beacon' (D7:29:56:68:7E:38) failed
07 21 07:48:29:201 SCPBluetoothScannerManager$bluetoothCentralCallback: connection 'D7:29:56:68:7E:38' failed with status 8
07 21 07:48:44:053 ...: tryAutomaticallyConnectToBeacons
07 21 07:48:44:175 BluetoothPeripheral: connect to 'Beacon' (D7:29:56:68:7E:38) using TRANSPORT_LE
07 21 07:48:57:154 BluetoothPeripheral: connection failed with status 'GATT ERROR' (ERROR)
07 21 07:48:57:163 BluetoothCentral: retrying connection to 'Beacon' (D7:29:56:68:7E:38)
07 21 07:48:57:168 BluetoothPeripheral: autoConnect to 'Beacon' (D7:29:56:68:7E:38) using TRANSPORT_LE
...
07 21 07:49:29:093 BluetoothPeripheral: force disconnect 'SBeacon' (D7:29:56:68:7E:38) 
`

and then it procceeds trying to establish connection, but just cannot and does not receive anything.

`
07 22 06:40:18:717 BluetoothCentral: already connecting to D7:29:56:68:7E:38'
07 22 06:40:19:444 BluetoothCentral: scanning timeout, restarting scan
07 22 06:40:19:451 BluetoothCentral: scan stopped
07 22 06:40:20:472 BluetoothCentral: scan started
....
07 22 06:41:03:735 BluetoothCentral: already connecting to D7:29:56:68:7E:38'
07 22 06:41:20:463 BluetoothCentral: scanning timeout, restarting scan
07 22 06:41:20:471 BluetoothCentral: scan stopped
07 22 06:41:21:491 BluetoothCentral: scan started
....
07 22 06:41:48:752 BluetoothCentral: already connecting to D7:29:56:68:7E:38'
07 22 06:42:21:479 BluetoothCentral: scanning timeout, restarting scan
07 22 06:42:21:487 BluetoothCentral: scan stopped
07 22 06:42:22:504 BluetoothCentral: scan started
`

weliem, 2020-07-25:
The last 'force disconnect', where does that come from? Is that your code that executes when your timer times out? Can you share your code?

Maartinsh, 2020-07-27:
> The last 'force disconnect', where does that come from? Is that your code that executes when your timer times out? Can you share your code?

Well, previously I had the same problem so I thought before actually doing connection, it might be useful to force disconnect by myself if the connection was "stuck". This as we can see did not help at all.

Maartinsh, 2020-07-27:
Also as I side note. 

We try to do connection every 45 seconds. 

```
private fun doManualConnection(
        beacon: Beacon,
        peripheral: BluetoothPeripheral
    ) {
        val alreadyConnected = connectedPeripherals.any { it.address == peripheral.address }
        val alreadyBonded = isPeripheralBondedAlready(peripheral.address)

        Timber.i("Peripheral state: ${peripheral.state.toString()}, already bonded: $alreadyBonded")

        if (peripheral.state == STATE_CONNECTING) {
            instantReconnectRequired = true
            baseScanner.central.cancelConnection(peripheral)
            return
        }

        if (!alreadyConnected) {
            if (alreadyBonded) {
                defaultConnectPeripheral(beacon, peripheral)  //connectPeripheral(peripheral, peripheralCallback)
            } else {
                autoConnectPeripheral(beacon, peripheral)  //autoConnectPeripheral(peripheral, peripheralCallback)
            }
        }
    }
```

What we kinda added is that before we start connecting to a peripheral and if we see that it is actually trying to "connect" for that moment, we assume the connection has been stuck (45 seconds passed) and we disconnect device and try to redo the connection with the same method. Problem that I also noticed is that we stuck in peripheral state `DISCONNECTING`. Should disconnection be done other way?

weliem, 2020-07-29:
Looking at the code, it may well be true that the state remains DISCONNECTING in your situation. I will fix it when I get back from holidays.

Maartinsh, 2020-07-31:
> Looking at the code, it may well be true that the state remains DISCONNECTING in your situation. I will fix it when I get back from holidays.

Thanks, I'd love to see solution 👍

weliem, 2020-08-02:
Ok, bug has been fixed in release 1.21. Have a look if it solves your problems....

weliem, 2020-08-02:
I think I also found the bug causing the 'already connecting to D7:29:56:68:7E:38' issue. In some cases the disconnected peripheral wouldn't be cleaned up properly.

The fix is in release 1.22

weliem, 2020-08-06:
Any update? Did the fixes work?

weliem, 2020-08-17:
Any news on this issue?

closed 2020-08-31 by weliem