weliem/blessed-android #44: writeCharacteristic Success but not received on target device

gocamiot, 2020-07-15:
Dear Martijn. I use the following command sequence:
val parserValue1 = BluetoothBytesParser()
                parserValue1.setIntValue(1, FORMAT_UINT8)
                //parserValue1.setString("01", FORMAT_UINT8)
                tagPeripheral.writeCharacteristic(tagCharacteristicList[1], parserValue1.value ,BluetoothGattCharacteristic.WRITE_TYPE_DEFAULT)

The tagCharacteristic list is what I get from:
val tagCharacteristicList: MutableList<BluetoothGattCharacteristic> = tagService.characteristics

LogCat shows on this writeCharacteristic command, from callback:
2020-07-15 06:45:08.346 22426-22426/com.itesseract.smartTag D/BluetoothPeripheral: writing <01> to characteristic <00002b29-0000-1000-8000-00805f9b34fb>
2020-07-15 06:45:08.352 5801-5913/? V/MARsPolicyManager: onAlertToastWindowStarted pkgName = com.itesseract.smartTag, userId = 0
2020-07-15 06:45:08.355 22426-22426/com.itesseract.smartTag D/ViewRootImpl@285206b[Toast]: setView = android.widget.LinearLayout{5197c8 V.E...... ......I. 0,0-0,0} TM=true MM=false
2020-07-15 06:45:08.355 22426-22426/com.itesseract.smartTag V/Toast: Text: Twit in android.widget.Toast$TN@bacc061
2020-07-15 06:45:08.367 22426-22426/com.itesseract.smartTag D/ViewRootImpl@285206b[Toast]: Relayout returned: old=[0,110][1080,2154] new=[86,1827][993,1986] req=(907,159)0 dur=6 res=0x7 s={true 487576141824} ch=true
2020-07-15 06:45:08.367 22426-22456/com.itesseract.smartTag D/OpenGLRenderer: createReliableSurface : 0x71eb96f140, 0x7185cd8000
2020-07-15 06:45:08.369 22426-22456/com.itesseract.smartTag I/mali_winsys: new_window_surface() [907x159] return: 0x3000
2020-07-15 06:45:08.372 22426-22456/com.itesseract.smartTag D/OpenGLRenderer: SurfaceChanged : 0x71eb9ae300 -> 0x727a462980
2020-07-15 06:45:08.384 22426-22426/com.itesseract.smartTag D/ViewRootImpl@285206b[Toast]: MSG_RESIZED: frame=[86,1827][993,1986] ci=[0,0][0,0] vi=[0,0][907,159] or=1
2020-07-15 06:45:08.416 22426-22426/com.itesseract.smartTag I/BluetoothHandler$peripheralCallback: SUCCESS: Writing <01> to <00002b29-0000-1000-8000-00805f9b34fb>

So: 
- Blessed callback says received OK.
- Callback on nrf52 do not get triggered. Does not show receive.
- I can use Blue Light app from punchthrough, send hex 01 and 00, same nrf device and code. It changes LED. It works.

I guess it is how I make up the packet with Blessed, I tried the other formats in BluetoothParser, string, long, UINT and SINT, 8 to 32 bits, offsets, same result.

Is there anything you can suggest? Please help.
Jakes

weliem, 2020-07-15:
The log looks totally normal. It wrote 1 to that characteristic successfully. Perhaps your devixe requires a WRITE_TYPE_NO_RESPONSE rather than a WRITE_TYPE_DEFAULT?

gocamiot, 2020-07-15:
Thanks for replying. I tried WRITE_TYPE_NO_RESPONSE. Comes back with error immediately.
2020-07-15 10:06:04.471 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked called. tagCharacteristic: Tag01
2020-07-15 10:06:04.472 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. tagCharacteristic ServiceUUID: 00001801-0000-1000-8000-00805f9b34fb
2020-07-15 10:06:04.472 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. tagCharacteristic ledUUID: 65210002-28d5-4b7b-badf-7dee1e8d1b6d
2020-07-15 10:06:04.473 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. tagPeripheral: FC:B4:1D:D8:3E:AD
2020-07-15 10:06:04.473 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. tagPeripheralName: smartTag02
2020-07-15 10:06:04.474 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. tagService: 00001801-0000-1000-8000-00805f9b34fb
2020-07-15 10:06:04.474 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. characteristicsUUID: 00002a05-0000-1000-8000-00805f9b34fb
2020-07-15 10:06:04.474 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. characteristicsService: android.bluetooth.BluetoothGattService@5197c8
2020-07-15 10:06:04.474 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. characteristicsValue: null
2020-07-15 10:06:04.474 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. characteristicsUUID: 00002b29-0000-1000-8000-00805f9b34fb
2020-07-15 10:06:04.474 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. characteristicsService: android.bluetooth.BluetoothGattService@5197c8
2020-07-15 10:06:04.474 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. characteristicsValue: null
2020-07-15 10:06:04.475 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. characteristicsUUID: 00002b2a-0000-1000-8000-00805f9b34fb
2020-07-15 10:06:04.475 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. characteristicsService: android.bluetooth.BluetoothGattService@5197c8
2020-07-15 10:06:04.475 12828-12828/com.itesseract.smartTag I/MainActivity: TagItemClicked. characteristicsValue: null
2020-07-15 10:06:04.475 12828-12828/com.itesseract.smartTag E/BluetoothPeripheral: characteristic <00002b29-0000-1000-8000-00805f9b34fb> does not support writeType 'WRITE_TYPE_NO_RESPONSE'

weliem, 2020-07-15:
Ok so your precious attempt was correct. No idea what it doesnt work. Android reported everything went well...

gocamiot, 2020-07-16:
Solved. I referenced incorrect characteristicUUID, so updated the incorrect characteristic.
All working now.
Thanks for help. Take care.
Ps. Again thank you for updated Blessed Library, you make Android BLE a breeze.