weliem/blessed-android #45: Notifications not working

darrinps, 2020-08-03:
I have a BLE device that has notifications working for Apple and am using blessed-android to attempt to do the same on an Android device.

Scanning, pairing, service discovery, and the write to turn on the notification service appear to function flawlessly, but no notification is ever received.

Here is some output from a run I did with added logging to help see what the issue may be:

2020-08-02 20:03:58.151 8672-8672/com.welie.blessedexample I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-02 20:03:58.151 8672-8672/com.welie.blessedexample I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-02 20:03:58.153 8672-8672/com.welie.blessedexample D/BluetoothGatt: connect() - device: 00:A0:50:30:C9:6B, auto: false
2020-08-02 20:03:58.153 8672-8672/com.welie.blessedexample D/BluetoothGatt: registerApp()
2020-08-02 20:03:58.153 8672-8672/com.welie.blessedexample D/BluetoothGatt: registerApp() - UUID=e552d944-1891-4522-9d24-97a00f53e690
2020-08-02 20:03:58.156 8672-8702/com.welie.blessedexample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2020-08-02 20:03:58.249 8672-8702/com.welie.blessedexample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=00:A0:50:30:C9:6B
2020-08-02 20:03:58.254 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: starting bonding with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-02 20:03:58.254 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: starting bonding with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-02 20:03:58.257 8672-8702/com.welie.blessedexample I/BluetoothPeripheral: connected to 'Stalker Pro IIs 1019' (BOND_BONDING) in 0.1s
2020-08-02 20:03:58.257 8672-8702/com.welie.blessedexample I/BluetoothPeripheral: connected to 'Stalker Pro IIs 1019' (BOND_BONDING) in 0.1s
2020-08-02 20:03:58.257 8672-8702/com.welie.blessedexample I/BluetoothPeripheral: waiting for bonding to complete
2020-08-02 20:03:58.258 8672-8702/com.welie.blessedexample I/BluetoothPeripheral: waiting for bonding to complete
2020-08-02 20:03:58.258 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: pairing request received , pairing variant: PAIRING_VARIANT_CONSENT (3)
2020-08-02 20:03:58.258 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: pairing request received , pairing variant: PAIRING_VARIANT_CONSENT (3)
2020-08-02 20:03:58.713 8672-8705/com.welie.blessedexample D/BluetoothGatt: onConnectionUpdated() - Device=00:A0:50:30:C9:6B interval=6 latency=0 timeout=500 status=0
2020-08-02 20:03:58.878 8672-8672/com.welie.blessedexample D/BluetoothCentral: popup hack completed
2020-08-02 20:03:58.878 8672-8672/com.welie.blessedexample D/BluetoothCentral: popup hack completed
2020-08-02 20:03:59.029 8672-8705/com.welie.blessedexample D/BluetoothGatt: onConnectionUpdated() - Device=00:A0:50:30:C9:6B interval=36 latency=0 timeout=500 status=0
2020-08-02 20:03:59.954 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: pairing request received , pairing variant: PAIRING_VARIANT_PIN (0)
2020-08-02 20:03:59.954 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: pairing request received , pairing variant: PAIRING_VARIANT_PIN (0)
2020-08-02 20:04:08.386 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: bonded with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-02 20:04:08.386 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: bonded with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-02 20:04:08.388 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: discovering services of 'Stalker Pro IIs 1019' with delay of 0 ms
2020-08-02 20:04:08.388 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: discovering services of 'Stalker Pro IIs 1019' with delay of 0 ms
2020-08-02 20:04:08.388 8672-8672/com.welie.blessedexample D/BluetoothGatt: discoverServices() - device: 00:A0:50:30:C9:6B
2020-08-02 20:04:08.401 8672-8702/com.welie.blessedexample D/BluetoothGatt: onSearchComplete() = Device=00:A0:50:30:C9:6B Status=0
2020-08-02 20:04:08.407 8672-8702/com.welie.blessedexample I/BluetoothPeripheral: discovered 4 services for 'Stalker Pro IIs 1019'
2020-08-02 20:04:08.407 8672-8702/com.welie.blessedexample I/BluetoothPeripheral: discovered 4 services for 'Stalker Pro IIs 1019'
2020-08-02 20:04:08.408 8672-8672/com.welie.blessedexample I/BluetoothHandler: connected to 'Stalker Pro IIs 1019'
2020-08-02 20:04:08.408 8672-8672/com.welie.blessedexample I/BluetoothHandler: connected to 'Stalker Pro IIs 1019'
2020-08-02 20:04:08.409 8672-8672/com.welie.blessedexample I/BluetoothHandler: discovered services
2020-08-02 20:04:08.409 8672-8672/com.welie.blessedexample I/BluetoothHandler: discovered services
2020-08-02 20:04:08.409 8672-8672/com.welie.blessedexample I/BluetoothHandler: Here is the list of supported services
2020-08-02 20:04:08.409 8672-8672/com.welie.blessedexample I/BluetoothHandler: Here is the list of supported services
2020-08-02 20:04:08.409 8672-8672/com.welie.blessedexample I/BluetoothHandler: Service: 00001800-0000-1000-8000-00805f9b34fb
2020-08-02 20:04:08.409 8672-8672/com.welie.blessedexample I/BluetoothHandler: Service: 00001800-0000-1000-8000-00805f9b34fb
2020-08-02 20:04:08.409 8672-8672/com.welie.blessedexample I/BluetoothHandler: Service: 00001801-0000-1000-8000-00805f9b34fb
2020-08-02 20:04:08.410 8672-8672/com.welie.blessedexample I/BluetoothHandler: Service: 00001801-0000-1000-8000-00805f9b34fb
2020-08-02 20:04:08.410 8672-8672/com.welie.blessedexample I/BluetoothHandler: Service: 4880c12c-fdcb-4077-8920-a450d7f9b907
2020-08-02 20:04:08.410 8672-8672/com.welie.blessedexample I/BluetoothHandler: Service: 4880c12c-fdcb-4077-8920-a450d7f9b907
2020-08-02 20:04:08.410 8672-8672/com.welie.blessedexample I/BluetoothHandler: Service: 0000180a-0000-1000-8000-00805f9b34fb
2020-08-02 20:04:08.410 8672-8672/com.welie.blessedexample I/BluetoothHandler: Service: 0000180a-0000-1000-8000-00805f9b34fb
2020-08-02 20:04:08.410 8672-8672/com.welie.blessedexample I/BluetoothHandler: Attempting to read manufacturer name
2020-08-02 20:04:08.410 8672-8672/com.welie.blessedexample I/BluetoothHandler: Attempting to read manufacturer name
2020-08-02 20:04:08.411 8672-8672/com.welie.blessedexample I/BluetoothHandler: Searching the services list for 4880c12c-fdcb-4077-8920-a450d7f9b907
2020-08-02 20:04:08.411 8672-8672/com.welie.blessedexample I/BluetoothHandler: Searching the services list for 4880c12c-fdcb-4077-8920-a450d7f9b907
2020-08-02 20:04:08.411 8672-8672/com.welie.blessedexample I/BluetoothHandler: Attempting to turn on notifications
2020-08-02 20:04:08.411 8672-8672/com.welie.blessedexample I/BluetoothHandler: Attempting to turn on notifications
2020-08-02 20:04:08.411 8672-8672/com.welie.blessedexample D/BluetoothGatt: requestConnectionPriority() - params: 1
2020-08-02 20:04:08.412 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: requesting connection priority 1
2020-08-02 20:04:08.412 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: requesting connection priority 1
2020-08-02 20:04:08.413 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: reading characteristic <00002a29-0000-1000-8000-00805f9b34fb>
2020-08-02 20:04:08.413 8672-8672/com.welie.blessedexample D/BluetoothPeripheral: reading characteristic <00002a29-0000-1000-8000-00805f9b34fb>
2020-08-02 20:04:08.571 8672-8672/com.welie.blessedexample I/BluetoothHandler: Received manufacturer: Stalker
2020-08-02 20:04:08.571 8672-8672/com.welie.blessedexample I/BluetoothHandler: Received manufacturer: Stalker
2020-08-02 20:04:08.572 8672-8672/com.welie.blessedexample D/BluetoothGatt: setCharacteristicNotification() - uuid: fec26ec4-6d71-4442-9f81-55bc21d658d6 enable: true
2020-08-02 20:04:08.750 8672-8672/com.welie.blessedexample I/BluetoothHandler: SUCCESS: Notify set to 'on' for fec26ec4-6d71-4442-9f81-55bc21d658d6
2020-08-02 20:04:08.751 8672-8672/com.welie.blessedexample I/BluetoothHandler: SUCCESS: Notify set to 'on' for fec26ec4-6d71-4442-9f81-55bc21d658d6
2020-08-02 20:04:08.883 8672-8702/com.welie.blessedexample D/BluetoothGatt: onConnectionUpdated() - Device=00:A0:50:30:C9:6B interval=12 latency=0 timeout=500 status=0

Anything I can do to help you debug this?
[image]
[image]

darrinps, 2020-08-03:
Note that I added a logging statement in the onCharacteristicUpdate and it never gets hit (as can be seen in the log above).

@Override
        public void onCharacteristicUpdate(BluetoothPeripheral peripheral, byte[] value, BluetoothGattCharacteristic characteristic, int status) {
            Timber.i("onCharacteristicUpdate received. Value is: %s", value.toString());

            if(status != GATT_SUCCESS) return;
            UUID characteristicUUID = characteristic.getUuid();
            BluetoothBytesParser parser = new BluetoothBytesParser(value);

weliem, 2020-08-03:
Everything looks totally fine. The last log statement confirms that turning on the notification has completed successfully. But then your log stops. Are you saying that after that nothing happens? What happens after?

I am not familiar with your device but perhaps you need to use it while being connected to get data...

darrinps, 2020-08-03:
Yes, up until the point where a notification should have been received it's perfect in every respect that I can tell.

If you look at the second post above you will see that there should have been a log statement for when a notification was received. I forced that notification on the device. It shows on Apple, but doesn't get there on Android.

If you look at the images, it appears that everything is set correctly (right UUIDs being used).

weliem, 2020-08-03:
Well, I don't think the library is causing any issue. Turning on notifications is quite straightforward and seems to work fine.

Are you sure the device is sending a notification at all? Does it work with nRF Connect?

darrinps, 2020-08-03:
It is sending them to iOS OK.

I hooked up nRF to it and set it to send notifications (I think that's the three downward pointing arrows). See screen shot:

[image]

It didn't automatically pick that data up though. I hit the single down arrow which I think does a read on the characteristic.

I'll verify with a similar app in iOS to see what's going on there and report back here.

darrinps, 2020-08-03:
OK, so I just loaded nRF on an iOS device, turned on notifications, and it works perfectly there.

So, there must be some difference between how iOS and Android is handling notifications as the same app (same developers at least) works on iOS and not on Android.

This is very odd.

weliem, 2020-08-04:
Can you grab the HCI log on iOS? My guess is that you are missing a command or something. It is highly unlikely that it works with iOS and not on Android. Usually there is a simple explanation...

darrinps, 2020-08-04:
Absolutely. Anything I can do to help. Will do this ASAP and post results here.

darrinps, 2020-08-04:
Not sure which format you wanted so posting in both inside of this zip

[HCILog.zip](https://github.com/weliem/blessed-android/files/5024709/HCILog.zip)

gocamiot, 2020-08-06:
I have samsung S10. bleTag with two notifications. Works perfectly. NRFdk840. Using android SDK 26 and up. Phone and SDK might make difference. Try a known working Android app first, like LightBlue or something maybe? Android sometimes moody.. Good luck.

darrinps, 2020-08-06:
Yeah, I even tried the "reference" implementation from Google and have this issue. It's odd because this device is working perfectly for iOS every time yet never works for Android. I can scan it, list its services, and connect, and pair with it perfectly, but unlike on iOS, Android just never gets any notifications. May not be this library though as the reference implementation and nRF both show the same thing. I just don't know what it could be. I mean, does the BLE spec deviate by device type? I sure wouldn't think so.

darrinps, 2020-08-07:
Just tried LightBlue on both iOS and Android. The notifications come over on iOS just fine and not on Android.

Guessing you can close this bug as I think it's just something out of spec with the peripheral not working for Android, although I am at a loss as to what that might be. I posted a question on Stack Overflow about this and if I get some good input, I'll reply back here so you can incorporate it into the library assuming it's something that can be done on the client side.

gocamiot, 2020-08-07:
What Peripheral is it? What Android phone? Do you have the Pheripheral code?
Try this code in onCharacteristicUpdate":
else if (characteristicUUID == SMARTTAG_LED1_CHARACTERISTIC_UUID) {
                val characteristicValue = parser.getIntValue(BluetoothBytesParser.FORMAT_UINT8)
                Timber.i("Received pnp LED1: %s", characteristicValue)

Take out GATT_SUCCESS return. This might be the issue.

Also see try

darrinps, 2020-08-07:
It's a radar gun...when the trigger is pulled, it emits a notification of the speed and various other aspects it registers...not a common thing.

darrinps, 2020-08-07:
I do not have the code...wish I did! LOL

I'll give that a try and report back...thanks

darrinps, 2020-08-07:
Don't see a GATT_SUCCESS return, but do see a 

     if(status != GATT_SUCCESS) return;

That's at line 134 of the BluetoothHandler class

Is that what you want me to comment out?

I searched on tagPeripheral and got no hits so not sure where you want that logging snippet to go.

gocamiot, 2020-08-07:
"if(status != GATT_SUCCESS) return; "

I have this nowhwere in my bluetooth handler. Yes comment out if it is in onCharacteristicUpdate.

Then add at the bottom of the else ifs in onCharacteristicUpdate

else if (characteristicUUID == YOUR_UUID_CHARACTERISTIC_UUID) {
                val characteristicValue = parser.getIntValue(BluetoothBytesParser.FORMAT_UINT8)
                Timber.i("Received pnp TEST: %s", characteristicValue) }
              
The UUID must be whatever UUID you set notification on, i think you set fec26ec4-6d71-4442-9f81-55bc21d658d6

Do another else if for the other UUID with above code.

Just a test.
Send logcat.

gocamiot, 2020-08-07:
In my mind your Notification set works. Its the logic when received notification that fails.

gocamiot, 2020-08-07:
Also see try
if (central!!.connectedPeripherals.contains(tagPeripheral)) {
Timber.i("TagItemClicked. tagPeripheral: ${tagPeripheral.address}")
Timber.i("TagItemClicked. tagPeripheralName: ${tagPeripheral.name}")
}

So.
There is a list kept with connected and not connected peripherals by blessed, if I understand this correctly. After deep dive in blessed libraries...
Very conveniently I can check that list, and see if perihperal moved out of range.. (disconneted and found in another list).
Martijn brilliance.. :)

You could see in that list if your radar gun is connected. Based on your above logcat, you will find it there.
:)

weliem, 2020-08-07:
I had a quick look at the HCI snoop log. What I see is that directly after connecting, an MTU of 185 is requested. Then the flow is:

- Then it enables an indication on a descriptor, probably the services changed descriptor...but I can't see uuid in the log
- It then reads the device name
- Then it enables a notification on a descriptor, probably the custom one
- Then notifications come in.

So probably you should try request the MTU as well. You must do this as the first thing in **onServicesDiscovered** and then **peripheral.requestMtu(185)**. Then add a callback onMtuChanged() and see if the 185 is accepted.

Maybe the device doesn't send notifications unless the MTU has been set to 185....

darrinps, 2020-08-07:
BINGO!!!! Verified it at lunch.

All because they (the BLE chip maker) didn't supply a proper GATT.

I wonder if there is any way to detect this in Blessed-Android to apply that automatically in cases like this?

FWIW, here is what I did:

`private final BluetoothPeripheralCallback peripheralCallback = new BluetoothPeripheralCallback() {
        @Override
        public void onServicesDiscovered(@NotNull BluetoothPeripheral peripheral) {
            Timber.i("discovered services");

            peripheral.requestMtu(185);

            // Request a new connection priority
            peripheral.requestConnectionPriority(CONNECTION_PRIORITY_HIGH);
...`
Then...

`@Override
        public void onMtuChanged(@NotNull BluetoothPeripheral peripheral, int mtu, int status) {
            super.onMtuChanged(peripheral, mtu, status);
            Timber.i("onMtuChanged for peripheral " + peripheral.getName() + " mtu=" + mtu + " status=" + status);

            // Turn on notification for Stalker Radar Service
            Timber.i("Searching the services list for %s", STALKER_SERVICE_UUID.toString());
            if(peripheral.getService(STALKER_SERVICE_UUID) != null) {
                Timber.i("Attempting to turn on notifications for service " + STALKER_SERVICE_UUID + "/nCharacteristic: " + STALKER_CHARACTERISTIC_UUID);
                peripheral.setNotify(peripheral.getCharacteristic(STALKER_SERVICE_UUID, STALKER_CHARACTERISTIC_UUID), true);
            }
        }`

weliem, 2020-08-07:
Nice! Glad that the problem was solved....

It is not that the manufacturer did something wrong. But if you don't have the official documentation, it takes some experimentation to figure out what the device is expecting...If requesting a mtu of 185 is crucial, it would normally be stated in the documentation.

darrinps, 2020-08-07:
Agreed. I just needed to be told.

I suppose this is normally just done on iPhone as standard procedure as every app I used on iPhone functioned perfectly.

Anyway, MANY thanks...the library seems to work very well BTW.

closed 2020-08-07 by darrinps