weliem/blessed-android #46: Multiple connects at the same time.

gocamiot, 2020-08-07:
Afternoon Martijn. Am I correct to say your sample app only connects one ble device at a time? How can I connect to multiple? And how many can I connect?

weliem, 2020-08-07:
Hi Jake, the sample app indeed only connects 1 device at a time. There can only be 1 central.connectPeripheral(peripheral) call active. This is due to lower level Android limitations. If you want to do multiple calls, you would have to queue them up.

However, you can try to do central.autoConnectPeripheral(peripheral) and then Android will queue them up so that you can have many outstanding connect requests.

Android has limitations on how many devices can be connected. The limit is 7 on most Android versions.

gocamiot, 2020-08-07:
Thank you Martijn. Will give it a go. Regards

closed 2020-08-07 by gocamiot