weliem/blessed-android #47: Crash on connection timeout in BluetoothPeripheral

yuvalSure, 2020-08-11:
Crash log:
2020-08-10 19:41:55.658 6704-6903/com.x.x.x.debug D/BluetoothGatt: close()
2020-08-10 19:41:55.658 6704-6903/com.x.x.x.debug D/BluetoothGatt: unregisterApp() - mClientIf=11
2020-08-10 19:41:55.660 7163-7202/? I/bt_stack: [INFO:gatt_api.cc(1212)] GATT_Disconnect conn_id=0
2020-08-10 19:41:55.660 7163-7184/? E/BtGatt.ContextMap: Context not found for ID 11
 --------- beginning of crash
2020-08-10 19:41:55.663 6704-6917/com.x.x.x.x E/AndroidRuntime: FATAL EXCEPTION: ThreadHandlerWrapper14
    Process: com.tekoia.sure.server.debug, PID: 6704
    java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGatt.disconnect()' on a null object reference
        at com.welie.blessed.BluetoothPeripheral$9.run(BluetoothPeripheral.java:996)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)

Triggered by the execution of "timeoutRunnable" in startConnectionTimer where the BluetoothGatt.close()  invoked and set to null just before BluetoothGatt.disconnect() was executed asynchronously.

weliem, 2020-08-11:
Which version of the library are you running? I think I have already fixed that issue....

weliem, 2020-08-11:
I think the issue was still there. Pushed a fix...

weliem, 2020-08-11:
Can you test?

weliem, 2020-08-17:
Any update? Did the fix work?

commit 5e4b7a90 (2020-08-11): Fix #47, make sure NPE is avoided
closed 2020-08-31 by weliem