weliem/blessed-android #48: Connection after initial pairing fails

darrinps, 2020-08-12:
I'm running the very latest version of the library and can only connect with an initial pairing. That is working perfectly. If I start the app up again though, eventually it will time out, try to connect again, and that too will time out. This same device works on iPhone after having been paired initially if that helps.

Here is the log file. Note that I added some additional logging.

2020-08-12 17:39:09.023 17697-17697/com.android.stalkerradar I/PeripheralListAdapter: Firing up RadarEnabled to connect to Radar gun: Stalker Pro IIs 1019
2020-08-12 17:39:09.031 17697-17697/com.android.stalkerradar I/BluetoothCentral: no scan to stop because no scan is running
2020-08-12 17:39:09.048 17697-17782/com.android.stalkerradar E/FA: Missing google_app_id. Firebase Analytics disabled. See https://goo.gl/NAOOOI
2020-08-12 17:39:09.118 17697-17697/com.android.stalkerradar I/RadarEnabledActivity: Checking permissions
2020-08-12 17:39:09.118 17697-17697/com.android.stalkerradar I/RadarEnabledActivity: Has location permission granted
2020-08-12 17:39:09.119 17697-17697/com.android.stalkerradar D/BluetoothHandler: Creating Central
2020-08-12 17:39:09.120 17697-17697/com.android.stalkerradar D/BluetoothHandler: Initiating popup hack
2020-08-12 17:39:09.120 17697-17697/com.android.stalkerradar D/BluetoothCentral: Not on a Samung device...hack needed
2020-08-12 17:39:09.123 17697-17697/com.android.stalkerradar I/BluetoothHandler: Alive and about to scan for Stalker Pro IIs 1019
2020-08-12 17:39:09.123 17697-17697/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-12 17:39:09.125 17697-17744/com.android.stalkerradar D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2020-08-12 17:39:09.127 17697-17697/com.android.stalkerradar I/BluetoothCentral: scan started
2020-08-12 17:39:09.358 17697-17697/com.android.stalkerradar I/BluetoothHandler$bluetoothCentralCallback: Found peripheral Stalker Pro IIs 1019
2020-08-12 17:39:09.358 17697-17697/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-12 17:39:09.360 17697-17697/com.android.stalkerradar I/BluetoothCentral: scan stopped
2020-08-12 17:39:09.371 17697-17782/com.android.stalkerradar D/FA: Connected to remote service
2020-08-12 17:39:09.463 17697-17697/com.android.stalkerradar I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-12 17:39:09.464 17697-17697/com.android.stalkerradar D/BluetoothGatt: connect() - device: 00:A0:50:30:C9:6B, auto: false
2020-08-12 17:39:09.464 17697-17697/com.android.stalkerradar D/BluetoothGatt: registerApp()
2020-08-12 17:39:09.464 17697-17697/com.android.stalkerradar D/BluetoothGatt: registerApp() - UUID=623b0bf7-061a-45b0-bc85-52ed6ea482e9
2020-08-12 17:39:09.468 17697-17744/com.android.stalkerradar D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2020-08-12 17:39:09.789 17697-17697/com.android.stalkerradar D/SelectRadarGunActivity: BLE: ScanDeviceActivity onStop()
2020-08-12 17:39:09.789 17697-17697/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-12 17:39:09.789 17697-17697/com.android.stalkerradar D/BluetoothLeScanner: could not find callback wrapper
2020-08-12 17:39:10.124 17697-17697/com.android.stalkerradar D/BluetoothCentral: popup hack completed
2020-08-12 17:39:10.178 17697-17744/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=00:A0:50:30:C9:6B
2020-08-12 17:39:10.180 17697-17744/com.android.stalkerradar I/BluetoothPeripheral: connected to 'Stalker Pro IIs 1019' (BOND_BONDED) in 0.7s
2020-08-12 17:39:10.182 17697-17697/com.android.stalkerradar D/BluetoothPeripheral: discovering services of 'Stalker Pro IIs 1019' with delay of 0 ms
2020-08-12 17:39:10.182 17697-17697/com.android.stalkerradar D/BluetoothGatt: discoverServices() - device: 00:A0:50:30:C9:6B
2020-08-12 17:39:10.484 17697-17697/com.android.stalkerradar D/BluetoothPeripheral: starting bonding with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-12 17:39:10.490 17697-17697/com.android.stalkerradar D/BluetoothPeripheral: bonded with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-12 17:39:10.491 17697-17697/com.android.stalkerradar D/BluetoothPeripheral: discovering services of 'Stalker Pro IIs 1019' with delay of 0 ms
2020-08-12 17:39:10.491 17697-17697/com.android.stalkerradar D/BluetoothGatt: discoverServices() - device: 00:A0:50:30:C9:6B
2020-08-12 17:39:10.925 17697-17744/com.android.stalkerradar D/BluetoothGatt: onConnectionUpdated() - Device=00:A0:50:30:C9:6B interval=6 latency=0 timeout=500 status=0
2020-08-12 17:39:40.582 17697-17744/com.android.stalkerradar D/BluetoothGatt: onSearchComplete() = Device=00:A0:50:30:C9:6B Status=0
2020-08-12 17:39:40.584 17697-17744/com.android.stalkerradar I/BluetoothPeripheral: discovered 0 services for 'Stalker Pro IIs 1019'
2020-08-12 17:39:40.587 17697-17744/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=22 clientIf=7 device=00:A0:50:30:C9:6B
2020-08-12 17:39:40.587 17697-17697/com.android.stalkerradar I/BluetoothHandler$bluetoothCentralCallback: connected to 'Stalker Pro IIs 1019'
2020-08-12 17:39:40.587 17697-17697/com.android.stalkerradar I/BluetoothHandler$peripheralCallback: discovered services
2020-08-12 17:39:40.588 17697-17697/com.android.stalkerradar E/BluetoothPeripheral: cannot request MTU, peripheral not connected
2020-08-12 17:39:40.589 17697-17744/com.android.stalkerradar I/BluetoothPeripheral: peripheral 'Stalker Pro IIs 1019' disconnected with status 'GATT CONN TERMINATE LOCAL HOST' before completing service discovery
2020-08-12 17:39:40.589 17697-17744/com.android.stalkerradar D/BluetoothGatt: close()
2020-08-12 17:39:40.589 17697-17744/com.android.stalkerradar D/BluetoothGatt: unregisterApp() - mClientIf=7
2020-08-12 17:39:40.592 17697-17744/com.android.stalkerradar I/BluetoothCentral: retrying connection to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-12 17:39:40.694 17697-17697/com.android.stalkerradar I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-12 17:39:40.705 17697-17697/com.android.stalkerradar D/BluetoothGatt: connect() - device: 00:A0:50:30:C9:6B, auto: false
2020-08-12 17:39:40.705 17697-17697/com.android.stalkerradar D/BluetoothGatt: registerApp()
2020-08-12 17:39:40.705 17697-17697/com.android.stalkerradar D/BluetoothGatt: registerApp() - UUID=5e6c029f-deec-4834-ae54-b2c1edf07842
2020-08-12 17:39:40.707 17697-17744/com.android.stalkerradar D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2020-08-12 17:40:10.711 17697-17744/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=7 device=00:A0:50:30:C9:6B
2020-08-12 17:40:10.712 17697-17744/com.android.stalkerradar I/BluetoothPeripheral: connection failed with status 'GATT ERROR' (TIMEOUT)
2020-08-12 17:40:10.712 17697-17744/com.android.stalkerradar D/BluetoothGatt: close()
2020-08-12 17:40:10.712 17697-17744/com.android.stalkerradar D/BluetoothGatt: unregisterApp() - mClientIf=7
2020-08-12 17:40:10.717 17697-17744/com.android.stalkerradar I/BluetoothCentral: connection to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) failed
2020-08-12 17:40:10.719 17697-17697/com.android.stalkerradar E/BluetoothHandler$bluetoothCentralCallback: connection 'Stalker Pro IIs 1019' failed with status 8

Is there anything I could be doing to cause this? Some implementation detail I missed perhaps?

Again though, it works perfectly if I go into setting, forget the device, fire up the app and let it pair. It just fails if I have already paired.

darrinps, 2020-08-12:
Here is the log from what a successful connection from a run where I did the initial pair.

2020-08-12 17:18:14.551 13005-13005/com.android.stalkerradar I/PeripheralListAdapter: Firing up RadarEnabled to connect to Radar gun: Stalker Pro IIs 1019
2020-08-12 17:18:14.560 13005-13005/com.android.stalkerradar I/BluetoothCentral: no scan to stop because no scan is running
2020-08-12 17:18:14.650 13005-13005/com.android.stalkerradar I/RadarEnabledActivity: Checking permissions
2020-08-12 17:18:14.650 13005-13005/com.android.stalkerradar I/RadarEnabledActivity: Has location permission granted
2020-08-12 17:18:14.652 13005-13005/com.android.stalkerradar D/BluetoothHandler: Creating Central
2020-08-12 17:18:14.653 13005-13005/com.android.stalkerradar D/BluetoothHandler: Initiating popup hack
2020-08-12 17:18:14.653 13005-13005/com.android.stalkerradar D/BluetoothCentral: Not on a Samung device...hack needed
2020-08-12 17:18:14.665 13005-13005/com.android.stalkerradar I/BluetoothHandler: Alive and about to scan for Stalker Pro IIs 1019
2020-08-12 17:18:14.665 13005-13005/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-12 17:18:14.667 13005-13059/com.android.stalkerradar D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2020-08-12 17:18:14.671 13005-13005/com.android.stalkerradar I/BluetoothCentral: scan started
2020-08-12 17:18:14.976 13005-13005/com.android.stalkerradar I/BluetoothHandler$bluetoothCentralCallback: Found peripheral Stalker Pro IIs 1019
2020-08-12 17:18:14.976 13005-13005/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-12 17:18:14.977 13005-13005/com.android.stalkerradar I/BluetoothCentral: scan stopped
2020-08-12 17:18:15.081 13005-13005/com.android.stalkerradar I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-12 17:18:15.083 13005-13005/com.android.stalkerradar D/BluetoothGatt: connect() - device: 00:A0:50:30:C9:6B, auto: false
2020-08-12 17:18:15.083 13005-13005/com.android.stalkerradar D/BluetoothGatt: registerApp()
2020-08-12 17:18:15.083 13005-13005/com.android.stalkerradar D/BluetoothGatt: registerApp() - UUID=4e3d88b3-4b57-496a-8c75-134d92c0d0c5
2020-08-12 17:18:15.084 13005-13059/com.android.stalkerradar D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2020-08-12 17:18:15.429 13005-13005/com.android.stalkerradar I/PeripheralListAdapter: DeviceList size is 1
2020-08-12 17:18:15.429 13005-13005/com.android.stalkerradar I/chatty: uid=10455(com.android.stalkerradar) identical 1 line
2020-08-12 17:18:15.431 13005-13005/com.android.stalkerradar D/SelectRadarGunActivity: BLE: ScanDeviceActivity onStop()
2020-08-12 17:18:15.432 13005-13005/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-12 17:18:15.432 13005-13005/com.android.stalkerradar D/BluetoothLeScanner: could not find callback wrapper
2020-08-12 17:18:15.665 13005-13005/com.android.stalkerradar D/BluetoothCentral: popup hack completed
2020-08-12 17:18:15.734 13005-13061/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=00:A0:50:30:C9:6B
2020-08-12 17:18:15.736 13005-13005/com.android.stalkerradar D/BluetoothPeripheral: starting bonding with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-12 17:18:15.737 13005-13061/com.android.stalkerradar I/BluetoothPeripheral: connected to 'Stalker Pro IIs 1019' (BOND_BONDING) in 0.7s
2020-08-12 17:18:15.737 13005-13061/com.android.stalkerradar I/BluetoothPeripheral: waiting for bonding to complete
2020-08-12 17:18:15.743 13005-13005/com.android.stalkerradar D/BluetoothPeripheral: pairing request received , pairing variant: PAIRING_VARIANT_CONSENT (3)
2020-08-12 17:18:15.819 13005-13219/com.android.stalkerradar D/RadarEnabledActivity$deviceStateCallback: Camera device closed
2020-08-12 17:18:15.832 13005-13115/com.android.stalkerradar E/FA: Missing google_app_id. Firebase Analytics disabled. See https://goo.gl/NAOOOI
2020-08-12 17:18:16.215 13005-13061/com.android.stalkerradar D/BluetoothGatt: onConnectionUpdated() - Device=00:A0:50:30:C9:6B interval=6 latency=0 timeout=500 status=0
2020-08-12 17:18:16.529 13005-13061/com.android.stalkerradar D/BluetoothGatt: onConnectionUpdated() - Device=00:A0:50:30:C9:6B interval=36 latency=0 timeout=500 status=0
2020-08-12 17:18:17.667 13005-13005/com.android.stalkerradar D/BluetoothPeripheral: pairing request received , pairing variant: PAIRING_VARIANT_PIN (0)
2020-08-12 17:18:27.111 13005-13005/com.android.stalkerradar D/BluetoothPeripheral: bonded with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-12 17:18:27.113 13005-13005/com.android.stalkerradar D/BluetoothPeripheral: discovering services of 'Stalker Pro IIs 1019' with delay of 0 ms
2020-08-12 17:18:27.113 13005-13005/com.android.stalkerradar D/BluetoothGatt: discoverServices() - device: 00:A0:50:30:C9:6B
2020-08-12 17:18:27.125 13005-13061/com.android.stalkerradar D/BluetoothGatt: onSearchComplete() = Device=00:A0:50:30:C9:6B Status=0
2020-08-12 17:18:27.126 13005-13061/com.android.stalkerradar I/BluetoothPeripheral: discovered 4 services for 'Stalker Pro IIs 1019'
2020-08-12 17:18:27.127 13005-13061/com.android.stalkerradar I/BluetoothPeripheral: 00001800-0000-1000-8000-00805f9b34fb
2020-08-12 17:18:27.127 13005-13061/com.android.stalkerradar I/BluetoothPeripheral: 00001801-0000-1000-8000-00805f9b34fb
[... cut ...]
2020-08-12 17:18:27.646 13005-13005/com.android.stalkerradar D/BluetoothGatt: setCharacteristicNotification() - uuid: fec26ec4-6d71-4442-9f81-55bc21d658d6 enable: true
2020-08-12 17:18:27.735 13005-13061/com.android.stalkerradar D/BluetoothGatt: onConnectionUpdated() - Device=00:A0:50:30:C9:6B interval=12 latency=0 timeout=500 status=0
2020-08-12 17:18:27.766 13005-13005/com.android.stalkerradar I/BluetoothHandler$peripheralCallback: SUCCESS: Notify set to 'on' for fec26ec4-6d71-4442-9f81-55bc21d658d6

weliem, 2020-08-13:
Will, when reconnecting the service discovery seems to fail and it discovers 0 services. After that the connection is dropped and the retry also fails. It seems related to the bonding process because that starts right in the middle of the service discovery. Which is odd, because you are already bonded in the first place! In your initial connection the bonding happens before the service discovery and then it works fine.

What phone are you using? And which Android version?

Just for my understanding, when connecting to this device you need to enter a PIN code, right?

The other issue that is weird is this line, although that has nothing to do with the bonding issue.
BluetoothLeScanner: could not find callback wrapper

That indicates that have started scanning more than once...

darrinps, 2020-08-13:
Using a Google Pixel 3XL
Android version is 11 (latest update)
Yes, I am required to enter a 6 character PIN code

Yes, there are TWO scans. I first use your library to scan for all devices and offer the user a list to choose from. After they select the one they want, I shut down the scanning and send the device name over to another Activity which does various things including initiating another scan/connect and connect to it in the same way your sample does. In short, I get an instance of BluetoothHandler using the device name and do a scan. That looks like this:

`BluetoothHandler.getInstance(applicationContext, this, deviceName)`

Then inside the BluetoothHandler code I use scanForPeripheralsWithNames like this:

`   init {
        // Create BluetoothCentral
        // Callback for central
        // Reconnect to this device when it becomes available again
        // Bluetooth is on now, start scanning again
        // Scan for peripherals with a certain service UUIDs
        val bluetoothCentralCallback: BluetoothCentralCallback =
            object : BluetoothCentralCallback() {
                override fun onConnectedPeripheral(peripheral: BluetoothPeripheral) {
                    Timber.i("connected to '%s'", peripheral.name)
                }

                override fun onConnectionFailed(
                    peripheral: BluetoothPeripheral,
                    status: Int
                ) {
                    Timber.e("connection '%s' failed with status %d", peripheral.name, status)
                }

                override fun onDisconnectedPeripheral(
                    peripheral: BluetoothPeripheral,
                    status: Int
                ) {
                    Timber.i("disconnected '%s' with status %d", peripheral.name, status)

                    // Reconnect to this device when it becomes available again
                    handler.postDelayed({
                        central?.autoConnectPeripheral(
                            peripheral,
                            peripheralCallback
                        )
                    }, 5000)
                }

                override fun onDiscoveredPeripheral(
                    peripheral: BluetoothPeripheral,
                    scanResult: ScanResult
                ) {
                    Timber.i("Found peripheral ${peripheral.name}")
                    central?.stopScan()
                    central?.connectPeripheral(peripheral, peripheralCallback)
                }

                override fun onBluetoothAdapterStateChanged(state: Int) {
                    Timber.i("bluetooth adapter changed state to %d", state)
                    if (state == BluetoothAdapter.STATE_ON) {
                        // Bluetooth is on now, start scanning again
                        // Scan for peripherals with a certain service UUIDs
                        central?.startPairingPopupHack()
                        central?.scanForPeripheralsWithServices(
                            arrayOf(
                                STALKER_SERVICE_UUID
                            )
                        )
                    }
                }
            }

        Timber.d("Creating Central")
        central = BluetoothCentral(context, bluetoothCentralCallback, Handler())

        Timber.d("Initiating popup hack")
        central?.startPairingPopupHack()

        // Scan for peripherals with a certain service UUIDs
        //        central.scanForPeripheralsWithServices(new UUID[]{STALKER_SERVICE_UUID});
        Timber.i("Alive and about to scan for $deviceName")
        central?.scanForPeripheralsWithNames(arrayOf(deviceName))
    }
`

weliem, 2020-08-14:
Ok so you are running on a decent device with stock Android...good. If you can get hold of a Samsung phone, I'd be interested to know if the same behavior can be observed. Samsung phones are known to deal with bonding differently than Google does.

So there are 2 issues that a causing this trouble:

1. Bonding events are coming in while service discovery is in progress. I have never seen this happen so it is quite unusual. This should not happen because you are already bonded with the device. So something funky is going here with this device.
2. Because my library isn't prepared to deal with #1, it starts another service discovery when the bonding completes. So this will probably mess up the other service discovery that was already running.

So issue 2 I can try to fix over the weekend. And then let's see what happens. No idea if Android can handle it issue 1...we'll find out

darrinps, 2020-08-14:
Yes I will get a fairly late model Samsung and send you the logs no later than tomorrow afternoon. Just let me know of anything else I can do to help.

darrinps, 2020-08-15:
Looks like it works on the Samsung just fine.

2020-08-14 21:10:53.617 31737-31737/com.android.stalkerradar I/PeripheralListAdapter: BLE: Turning off scanning.
2020-08-14 21:10:53.618 31737-31737/com.android.stalkerradar I/SelectRadarGunActivity$BluetoothCentralCallbackWithScanningAdapter: Stopping scan for peripherals...
2020-08-14 21:10:53.618 31737-31737/com.android.stalkerradar D/BluetoothAdapter: STATE_ON
2020-08-14 21:10:53.618 31737-31737/com.android.stalkerradar D/BluetoothLeScanner: Stop Scan with callback
2020-08-14 21:10:53.620 31737-31737/com.android.stalkerradar I/BluetoothCentral: scan stopped
2020-08-14 21:10:53.621 31737-31737/com.android.stalkerradar I/PeripheralListAdapter: Firing up RadarEnabled to connect to Radar gun: Stalker Pro IIs 1019
2020-08-14 21:10:53.633 31737-31737/com.android.stalkerradar I/BluetoothCentral: no scan to stop because no scan is running
2020-08-14 21:10:53.785 31737-31737/com.android.stalkerradar I/RadarEnabledActivity: Registering stalkerDataReceiver
2020-08-14 21:10:53.786 31737-31737/com.android.stalkerradar I/RadarEnabledActivity: Checking permissions
2020-08-14 21:10:53.787 31737-31737/com.android.stalkerradar I/RadarEnabledActivity: Has location permission granted
2020-08-14 21:10:53.787 31737-31737/com.android.stalkerradar I/RadarEnabledActivity: Getting instance of BluetoothHandler for device: Stalker Pro IIs 1019
2020-08-14 21:10:53.792 31737-31737/com.android.stalkerradar D/BluetoothHandler: Creating Central
2020-08-14 21:10:53.793 31737-31737/com.android.stalkerradar D/BluetoothHandler: Initiating popup hack
2020-08-14 21:10:53.794 31737-31737/com.android.stalkerradar I/BluetoothHandler: Alive and about to scan for Stalker Pro IIs 1019
2020-08-14 21:10:53.796 31737-31737/com.android.stalkerradar D/BluetoothAdapter: STATE_ON
2020-08-14 21:10:53.797 31737-31737/com.android.stalkerradar D/BluetoothAdapter: STATE_ON
2020-08-14 21:10:53.797 31737-31737/com.android.stalkerradar D/BluetoothLeScanner: Start Scan with callback
2020-08-14 21:10:53.798 31737-31758/com.android.stalkerradar D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=11 mScannerId=0
2020-08-14 21:10:53.801 31737-31737/com.android.stalkerradar I/BluetoothCentral: scan started
2020-08-14 21:10:53.802 31737-31737/com.android.stalkerradar I/RadarEnabledActivity: Added a speed data point. Peak: 0.0 at time 1597457453801
2020-08-14 21:10:53.905 31737-31737/com.android.stalkerradar D/RadarEnabledActivity$surfaceListener: Surface has changed.
2020-08-14 21:10:53.917 31737-31737/com.android.stalkerradar D/RadarEnabledActivity: Camera deviceId: 0
2020-08-14 21:10:53.945 31737-31813/com.android.stalkerradar D/RadarEnabledActivity$deviceStateCallback: Camera device opened
2020-08-14 21:10:53.945 31737-31813/com.android.stalkerradar D/RadarEnabledActivity$deviceStateCallback: Camera device ready for preview session
2020-08-14 21:10:53.973 31737-31737/com.android.stalkerradar D/SelectRadarGunActivity: BLE: ScanDeviceActivity onStop()
2020-08-14 21:10:53.975 31737-31737/com.android.stalkerradar D/BluetoothAdapter: STATE_ON
2020-08-14 21:10:53.975 31737-31737/com.android.stalkerradar D/BluetoothAdapter: STATE_ON
2020-08-14 21:10:53.975 31737-31737/com.android.stalkerradar D/BluetoothLeScanner: could not find callback wrapper
2020-08-14 21:10:54.020 31737-31737/com.android.stalkerradar I/BluetoothHandler$bluetoothCentralCallback: Found peripheral Stalker Pro IIs 1019
2020-08-14 21:10:54.020 31737-31737/com.android.stalkerradar D/BluetoothAdapter: STATE_ON
2020-08-14 21:10:54.020 31737-31737/com.android.stalkerradar D/BluetoothLeScanner: Stop Scan with callback
2020-08-14 21:10:54.021 31737-31737/com.android.stalkerradar I/BluetoothCentral: scan stopped
2020-08-14 21:10:54.123 31737-31737/com.android.stalkerradar I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-14 21:10:54.124 31737-31737/com.android.stalkerradar D/BluetoothAdapter: STATE_ON
2020-08-14 21:10:54.125 31737-31737/com.android.stalkerradar D/BluetoothGatt: connect() - device: 00:A0:50:30:C9:6B, auto: false
2020-08-14 21:10:54.126 31737-31737/com.android.stalkerradar D/BluetoothAdapter: isSecureModeEnabled
2020-08-14 21:10:54.126 31737-31737/com.android.stalkerradar D/BluetoothGatt: registerApp()
2020-08-14 21:10:54.126 31737-31737/com.android.stalkerradar D/BluetoothGatt: registerApp() - UUID=043cb8b9-aeca-41c8-9b0e-11041088e0c7
2020-08-14 21:10:54.128 31737-31757/com.android.stalkerradar D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
2020-08-14 21:10:54.247 31737-31757/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=00:A0:50:30:C9:6B
2020-08-14 21:10:54.250 31737-31757/com.android.stalkerradar I/BluetoothPeripheral: connected to 'Stalker Pro IIs 1019' (BOND_BONDED) in 0.1s
2020-08-14 21:10:54.251 31737-31737/com.android.stalkerradar D/BluetoothPeripheral: discovering services of 'Stalker Pro IIs 1019' with delay of 0 ms
2020-08-14 21:10:54.251 31737-31737/com.android.stalkerradar D/BluetoothGatt: discoverServices() - device: 00:A0:50:30:C9:6B
2020-08-14 21:10:54.254 31737-31757/com.android.stalkerradar D/BluetoothGatt: onSearchComplete() = Device=00:A0:50:30:C9:6B Status=0
2020-08-14 21:10:54.257 31737-31757/com.android.stalkerradar I/BluetoothPeripheral: discovered 4 services for 'Stalker Pro IIs 1019'
[... cut ...]
2020-08-14 21:10:55.001 31737-31737/com.android.stalkerradar I/BluetoothHandler$peripheralCallback: onCharacteristicUpdate received. Value is: []
2020-08-14 21:10:55.002 31737-31737/com.android.stalkerradar I/BluetoothHandler$peripheralCallback: Received modelnumber: 
2020-08-14 21:10:55.002 31737-31737/com.android.stalkerradar D/BluetoothGatt: setCharacteristicNotification() - uuid: fec26ec4-6d71-4442-9f81-55bc21d658d6 enable: true
2020-08-14 21:10:55.063 31737-31737/com.android.stalkerradar I/BluetoothHandler$peripheralCallback: SUCCESS: Notify set to 'on' for fec26ec4-6d71-4442-9f81-55bc21d658d6

weliem, 2020-08-15:
Ok, that is sort of 'good news'. So it is just the Google Bluetooth stack not dealing with it properly.

Anyway, I pushed a fix (064cdff) for not starting the second discovery....can you try it?

darrinps, 2020-08-15:
Absolutely. I'll give it a try later today/tomorrow and report back here.

darrinps, 2020-08-15:
That doesn't seem to allow it to complete connection and sign up for notifications.

2020-08-15 10:57:42.438 18387-18387/com.android.stalkerradar I/SelectRadarGunActivity$BluetoothCentralCallbackWithScanningAdapter: Stopping scan for peripherals...
2020-08-15 10:57:42.438 18387-18387/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-15 10:57:42.441 18387-18387/com.android.stalkerradar I/BluetoothCentral: scan stopped
2020-08-15 10:57:42.441 18387-18387/com.android.stalkerradar I/PeripheralListAdapter: Firing up RadarEnabled to connect to Radar gun: Stalker Pro IIs 1019
2020-08-15 10:57:42.451 18387-18387/com.android.stalkerradar I/BluetoothCentral: no scan to stop because no scan is running
2020-08-15 10:57:42.553 18387-18387/com.android.stalkerradar I/RadarEnabledActivity: Registering stalkerDataReceiver
2020-08-15 10:57:42.555 18387-18387/com.android.stalkerradar I/RadarEnabledActivity: Checking permissions
2020-08-15 10:57:42.556 18387-18387/com.android.stalkerradar I/RadarEnabledActivity: Has location permission granted
2020-08-15 10:57:42.556 18387-18387/com.android.stalkerradar I/RadarEnabledActivity: Getting instance of BluetoothHandler for device: Stalker Pro IIs 1019
2020-08-15 10:57:42.557 18387-18387/com.android.stalkerradar D/BluetoothHandler: Creating Central
2020-08-15 10:57:42.558 18387-18387/com.android.stalkerradar D/BluetoothHandler: Initiating popup hack
2020-08-15 10:57:42.567 18387-18387/com.android.stalkerradar I/BluetoothHandler: Alive and about to scan for Stalker Pro IIs 1019
2020-08-15 10:57:42.567 18387-18387/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-15 10:57:42.569 18387-18424/com.android.stalkerradar D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2020-08-15 10:57:42.573 18387-18387/com.android.stalkerradar I/BluetoothCentral: scan started
2020-08-15 10:57:42.575 18387-18387/com.android.stalkerradar I/RadarEnabledActivity: Added a speed data point. Peak: 0.0 at time 1597507062573
2020-08-15 10:57:42.830 18387-18387/com.android.stalkerradar I/BluetoothHandler$bluetoothCentralCallback: Found peripheral Stalker Pro IIs 1019
2020-08-15 10:57:42.830 18387-18387/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-15 10:57:42.832 18387-18387/com.android.stalkerradar I/BluetoothCentral: scan stopped
2020-08-15 10:57:42.934 18387-18387/com.android.stalkerradar I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-15 10:57:42.935 18387-18387/com.android.stalkerradar D/BluetoothGatt: connect() - device: 00:A0:50:30:C9:6B, auto: false
2020-08-15 10:57:42.936 18387-18387/com.android.stalkerradar D/BluetoothGatt: registerApp()
2020-08-15 10:57:42.936 18387-18387/com.android.stalkerradar D/BluetoothGatt: registerApp() - UUID=a24c8059-446b-4efb-82af-4f153753ff08
2020-08-15 10:57:42.937 18387-18422/com.android.stalkerradar D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2020-08-15 10:57:43.292 18387-18387/com.android.stalkerradar D/SelectRadarGunActivity: BLE: ScanDeviceActivity onStop()
2020-08-15 10:57:43.293 18387-18387/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-15 10:57:43.293 18387-18387/com.android.stalkerradar D/BluetoothLeScanner: could not find callback wrapper
2020-08-15 10:57:43.566 18387-18387/com.android.stalkerradar D/BluetoothCentral: popup hack completed
2020-08-15 10:57:43.622 18387-18424/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=00:A0:50:30:C9:6B
2020-08-15 10:57:43.623 18387-18424/com.android.stalkerradar I/BluetoothPeripheral: connected to 'Stalker Pro IIs 1019' (BOND_BONDED) in 0.7s
2020-08-15 10:57:43.624 18387-18387/com.android.stalkerradar D/BluetoothPeripheral: discovering services of 'Stalker Pro IIs 1019' with delay of 0 ms
2020-08-15 10:57:43.625 18387-18387/com.android.stalkerradar D/BluetoothGatt: discoverServices() - device: 00:A0:50:30:C9:6B
2020-08-15 10:57:43.946 18387-18387/com.android.stalkerradar D/BluetoothPeripheral: starting bonding with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-15 10:57:43.948 18387-18387/com.android.stalkerradar D/BluetoothPeripheral: bonded with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-15 10:57:44.382 18387-18424/com.android.stalkerradar D/BluetoothGatt: onConnectionUpdated() - Device=00:A0:50:30:C9:6B interval=6 latency=0 timeout=500 status=0
2020-08-15 10:58:14.057 18387-18424/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=22 clientIf=7 device=00:A0:50:30:C9:6B
2020-08-15 10:58:14.060 18387-18424/com.android.stalkerradar I/BluetoothPeripheral: peripheral 'Stalker Pro IIs 1019' disconnected with status 'GATT CONN TERMINATE LOCAL HOST' before completing service discovery
2020-08-15 10:58:14.060 18387-18424/com.android.stalkerradar D/BluetoothGatt: close()
2020-08-15 10:58:14.060 18387-18424/com.android.stalkerradar D/BluetoothGatt: unregisterApp() - mClientIf=7
2020-08-15 10:58:14.064 18387-18424/com.android.stalkerradar I/BluetoothCentral: retrying connection to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-15 10:58:14.167 18387-18387/com.android.stalkerradar I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-15 10:58:14.170 18387-18387/com.android.stalkerradar D/BluetoothGatt: connect() - device: 00:A0:50:30:C9:6B, auto: false
2020-08-15 10:58:14.170 18387-18387/com.android.stalkerradar D/BluetoothGatt: registerApp()
2020-08-15 10:58:14.170 18387-18387/com.android.stalkerradar D/BluetoothGatt: registerApp() - UUID=32cebcae-bd74-4203-b7de-03af6a369991
2020-08-15 10:58:14.173 18387-18424/com.android.stalkerradar D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7

It still works fine when you have to pair though.

weliem, 2020-08-15:
Ok, well at least my fix did what it was supposed to do. But it didn't solve the connection problem.

The only other thing I can think of is to delay the service discovery until the bonding part has completed.

To quickly test this hypothesis, you could comment out line 527:
`delayedDiscoverServices(getServiceDiscoveryDelay(bondstate));`

This is not a real fix of course, but it would disable the service discovery directly after connecting....

Can you try that?

darrinps, 2020-08-15:
Here you go. I added a log statement where you had me comment out the line.

2020-08-15 15:25:08.829 3462-3462/com.android.stalkerradar I/BluetoothCentral: no scan to stop because no scan is running
2020-08-15 15:25:09.017 3462-3462/com.android.stalkerradar I/RadarEnabledActivity: Registering stalkerDataReceiver
2020-08-15 15:25:09.019 3462-3462/com.android.stalkerradar I/RadarEnabledActivity: Checking permissions
2020-08-15 15:25:09.019 3462-3462/com.android.stalkerradar I/RadarEnabledActivity: Has location permission granted
2020-08-15 15:25:09.019 3462-3462/com.android.stalkerradar I/RadarEnabledActivity: Getting instance of BluetoothHandler for device: Stalker Pro IIs 1019
2020-08-15 15:25:09.021 3462-3462/com.android.stalkerradar D/BluetoothHandler: Creating Central
2020-08-15 15:25:09.022 3462-3462/com.android.stalkerradar D/BluetoothHandler: Initiating popup hack
2020-08-15 15:25:09.038 3462-3462/com.android.stalkerradar I/BluetoothHandler: Alive and about to scan for Stalker Pro IIs 1019
2020-08-15 15:25:09.039 3462-3462/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-15 15:25:10.229 3462-3488/com.android.stalkerradar D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=6 mScannerId=0
2020-08-15 15:25:10.236 3462-3462/com.android.stalkerradar I/BluetoothCentral: scan started
2020-08-15 15:25:10.328 3462-3462/com.android.stalkerradar D/BluetoothCentral: popup hack completed
2020-08-15 15:25:11.411 3462-3462/com.android.stalkerradar D/SelectRadarGunActivity: BLE: ScanDeviceActivity onStop()
2020-08-15 15:25:11.411 3462-3462/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-15 15:25:11.411 3462-3462/com.android.stalkerradar D/BluetoothLeScanner: could not find callback wrapper
2020-08-15 15:25:12.278 3462-3462/com.android.stalkerradar I/BluetoothHandler$bluetoothCentralCallback: Found peripheral Stalker Pro IIs 1019
2020-08-15 15:25:12.278 3462-3462/com.android.stalkerradar D/BluetoothAdapter: isLeEnabled(): ON
2020-08-15 15:25:12.279 3462-3462/com.android.stalkerradar I/BluetoothCentral: scan stopped
2020-08-15 15:25:12.381 3462-3462/com.android.stalkerradar I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-15 15:25:12.383 3462-3462/com.android.stalkerradar D/BluetoothGatt: connect() - device: 00:A0:50:30:C9:6B, auto: false
2020-08-15 15:25:12.383 3462-3462/com.android.stalkerradar D/BluetoothGatt: registerApp()
2020-08-15 15:25:12.383 3462-3462/com.android.stalkerradar D/BluetoothGatt: registerApp() - UUID=b2113605-23f6-479d-925e-c4cc3599cc87
2020-08-15 15:25:12.392 3462-3543/com.android.stalkerradar D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
2020-08-15 15:25:14.800 3462-3543/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=00:A0:50:30:C9:6B
2020-08-15 15:25:14.802 3462-3543/com.android.stalkerradar I/BluetoothPeripheral: connected to 'Stalker Pro IIs 1019' (BOND_BONDED) in 2.4s
2020-08-15 15:25:14.802 3462-3543/com.android.stalkerradar I/BluetoothPeripheral: Commented out to test theory
2020-08-15 15:25:15.107 3462-3462/com.android.stalkerradar D/BluetoothPeripheral: starting bonding with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-15 15:25:15.114 3462-3462/com.android.stalkerradar D/BluetoothPeripheral: bonded with 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-15 15:25:15.115 3462-3462/com.android.stalkerradar D/BluetoothPeripheral: discovering services of 'Stalker Pro IIs 1019' with delay of 0 ms
2020-08-15 15:25:15.115 3462-3462/com.android.stalkerradar D/BluetoothGatt: discoverServices() - device: 00:A0:50:30:C9:6B
2020-08-15 15:25:15.547 3462-3543/com.android.stalkerradar D/BluetoothGatt: onConnectionUpdated() - Device=00:A0:50:30:C9:6B interval=6 latency=0 timeout=500 status=0
2020-08-15 15:25:45.192 3462-3543/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=22 clientIf=6 device=00:A0:50:30:C9:6B
2020-08-15 15:25:45.195 3462-3543/com.android.stalkerradar I/BluetoothPeripheral: peripheral 'Stalker Pro IIs 1019' disconnected with status 'GATT CONN TERMINATE LOCAL HOST' before completing service discovery
2020-08-15 15:25:45.195 3462-3543/com.android.stalkerradar D/BluetoothGatt: close()
2020-08-15 15:25:45.195 3462-3543/com.android.stalkerradar D/BluetoothGatt: unregisterApp() - mClientIf=6
2020-08-15 15:25:45.199 3462-3543/com.android.stalkerradar I/BluetoothCentral: retrying connection to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B)
2020-08-15 15:25:45.302 3462-3462/com.android.stalkerradar I/BluetoothPeripheral: connect to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) using TRANSPORT_LE
2020-08-15 15:25:45.306 3462-3462/com.android.stalkerradar D/BluetoothGatt: connect() - device: 00:A0:50:30:C9:6B, auto: false
2020-08-15 15:25:45.306 3462-3462/com.android.stalkerradar D/BluetoothGatt: registerApp()
2020-08-15 15:25:45.306 3462-3462/com.android.stalkerradar D/BluetoothGatt: registerApp() - UUID=f36d5921-6d8e-4c75-a95d-43aba12c5ce8
2020-08-15 15:25:45.307 3462-3543/com.android.stalkerradar D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
2020-08-15 15:26:15.312 3462-3543/com.android.stalkerradar D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=6 device=00:A0:50:30:C9:6B
2020-08-15 15:26:15.314 3462-3543/com.android.stalkerradar I/BluetoothPeripheral: connection failed with status 'GATT ERROR' (TIMEOUT)
2020-08-15 15:26:15.314 3462-3543/com.android.stalkerradar D/BluetoothGatt: close()
2020-08-15 15:26:15.314 3462-3543/com.android.stalkerradar D/BluetoothGatt: unregisterApp() - mClientIf=6
2020-08-15 15:26:15.319 3462-3543/com.android.stalkerradar I/BluetoothCentral: connection to 'Stalker Pro IIs 1019' (00:A0:50:30:C9:6B) failed
2020-08-15 15:26:15.322 3462-3462/com.android.stalkerradar E/BluetoothHandler$bluetoothCentralCallback: connection 'Stalker Pro IIs 1019' failed with status 8

weliem, 2020-08-16:
Hmmm, it is not looking good. Bonding issue is still there. Effectively, the code is doing absolutely nothing else than connecting and calling discoverServices()....but the 'stalker' is doing something 'wrong' that the Google Bluetooth stack doesn't like because the service discovery does not succeed and it disconnects.

Last try, can you change line 664, from:
`delayedDiscoverServices(0);`
to
`delayedDiscoverServices(500);`

That adds a bit more delay. Probably wont help but it is the last thing you could try... 

Make sure you:
- Reboot your phone, this will 'reset' the bluetooth stack
- Pair first...should still work
- Reconnect

darrinps, 2020-08-16:
Tried both with leaving line 527 commented out, and not commented out...same thing...no good. :(

This is really sad when you think about it as a Google Pixel with the latest OS should, one would think, be the "reference" implementation of the stack. I'm worried now though that whatever they do will get incorporated into other vendor's stacks.

Do you have any insight into what specific bug I should file with Google that might help correct this?

weliem, 2020-08-16:
Well, there is always two sides to the story. In this case, I think the peripheral is doing something funky that is definitely non-standard. I have never seen bonding events during service discovery and I worked with several devices that required PIN-based bonding.

Apparently, iOS and Samsung phones can handle this scenario but not the Google Pixel phones. You'd have to do a low-level bluetooth packet analysis to determine which side is 'wrong'.

I also checked the Stalker's manufacturer's website and they only offer an iOS app and no Android app. I can guess why....

There is only one last thing I can suggest to you:
1. Use the setPinForPeripheral() method so that Blessed can programmatically enter the pin code for you.
2. Then do a first connect. This should succeed as we have seen.
3. In onDisconnectedPeripheral, call removeBond(). That will remove the bonding on the phone
4. Reconnecting now becomes the same as the initial connect. Bonding will happen again and should succeed again.

It is not pretty, but it might work as a 'workaround'....

darrinps, 2020-08-16:
I'm in the process of taking that log now (was thinking they'd need that too).

Yes, I thought about doing the removeBond() but was wondering how that would work in some instances. There used to be a way to detect if you were bonded or not so I was thinking of trying to detect that, and if on a Google Phone, remove the bond, then pair it automatically as you mention.

Not sure if your library supports checking for bond information or not, but I could always wire something up myself if I had to.

In the mean time, I'll report this back to Stalker as well as Google.

darrinps, 2020-08-16:
Here's the snoop log for completeness sake. I've filed a bug with Google referencing this issue. 

Many thanks for the valiant effort put forth.

[btsnoop_hci.log](https://github.com/weliem/blessed-android/files/5080964/btsnoop_hci.log)

darrinps, 2020-08-17:
FWIW, I've found a Note 10 running Android 10 won't even pair with this device through any means (my app, third party apps, or even Android's built in way).

I'll speak with the vendor and update here what we find out.

weliem, 2020-08-17:
I think the stalker is probably at fault in this case.

I hope you can convince the manufacturer to improve the firmware so that it also works on Android!

Good luck!

closed 2020-08-17 by weliem