weliem/blessed-android #49: How to connect to 2 Devices parallel

Hpuhome, 2020-08-19:
I want to build a bikecomputer with heartrate function.
After I implemented the blessed app successfully and expanded the app to get the 
batterylevel plus heartrate on the Screen. I enhanced the BluetoothHandler.java
to receive the signals from a combodevice which sends cycling and cadence values.
With your old version I could not receive signals, but after download the newer blessed app
last weekend, I could receive the signals from the combodevice. The problem is,
I could only receive HR- or CSC-signals, not both together.
I think it is necessary to change something in the "blessed App" especially in 
the peripheral.java. But I don’t understand the "blessed App" so far that I can
do that.
I hope to get help from you.
Thank You.

weliem, 2020-08-19:
I don't really understand your question. It sounds like you want to connect to 1 peripheral that has both the HR service and the CSC service. Right?

The HR service is already enabled in the example app. You just need to add the CSC service and turn on the notify for the relevant characteristics. That should be all.

gocamiot, 2020-08-20:
In BlueToothHandler class Change the following class to this:
override fun onDiscoveredPeripheral(peripheral: BluetoothPeripheral, scanResult: ScanResult) {
            Timber.i("Found peripheral '%s'", peripheral.name)
            //central.stopScan()
            //central.connectPeripheral(peripheral, peripheralCallback)
            central.autoConnectPeripheral(peripheral, peripheralCallback)
        }

Let us know if that answers your question.

weliem, 2020-10-10:
Closing due to no activity....

closed 2020-10-10 by weliem