weliem/blessed-android #51: writeCharacteristic fails

Koxx3, 2020-09-08:
Hello !

Glad I found this library. Great job !

I added this simple function in BluetoothHandler.java... but the write operation fails.
I checked the UUID, and it's good.
I can write it with some BluetoothScanner app.
What be wrong ?

`
    public void setValue(byte value) {

        BluetoothGattCharacteristic characteristic = new BluetoothGattCharacteristic(UUID.fromString("beb5483e-36e1-4688-b7f5-ea07361b26a8"), PROPERTY_WRITE, 0);

        central.getConnectedPeripherals().get(0).writeCharacteristic(characteristic, new byte[]{0}, WRITE_TYPE_DEFAULT);

    }`

Here is the trace :
`
2020-09-08 19:03:48.936 4600-4600/com.welie.blessedexample D/MainActivity: onClickMode
2020-09-08 19:03:48.936 4600-4600/com.welie.blessedexample D/MainActivity: lcdMode sent : 0
2020-09-08 19:03:48.937 4600-4600/com.welie.blessedexample E/BluetoothPeripheral: writeCharacteristic failed for characteristic: beb5483e-36e1-4688-b7f5-ea07361b26a8`

Koxx3, 2020-09-08:
of course, my peripheral is connected (and only one), and I can read some other characteristics.

Koxx3, 2020-09-08:
ah yes ... I have notifications arriving at 10Hz on the app on a different characteristic.

weliem, 2020-09-09:
You should not create your own characteristics objects. Use peripheral.getCharacteristic()

Typically you would do something like this:

```
BluetoothGattCharacteristic currentTimeCharacteristic = peripheral.getCharacteristic(CTS_SERVICE_UUID, CURRENT_TIME_CHARACTERISTIC_UUID);
if (currentTimeCharacteristic != null) {
        peripheral.writeCharacteristic(currentTimeCharacteristic, <your_byte_array>, WRITE_TYPE_DEFAULT);
}
```

Koxx3, 2020-09-19:
works fine ! Thanks !

closed 2020-09-19 by Koxx3