weliem/blessed-android #52: using custom ScanFilters not possible right now

mwkaufmann, 2020-09-16:
First of all: great project. Thank you very much. Getting Android BLE right is not an easy task!

I evaluated this library for a project and it would be perfect, if there was an option to provide custom ScanFilters.

Right now there is no public method that takes filters as a parameter. Could be easily added. If you agree that this would be useful and have not time to implement it, I can prepare a PR.

weliem, 2020-09-16:
Actually there is already such a method: 
`public void scanForPeripheralsUsingFilters(List<ScanFilter> filters)`

It was recently added....perhaps you need to update your version?

mwkaufmann, 2020-09-16:
You are right! I should have checked that. I certainly underestimated the dynamic of this project. :)

Well, all I have to say, thanks again and have a nice day!

closed 2020-09-16 by mwkaufmann