weliem/blessed-android #53: Error writing charactestic with more than 20/23 bytes (MTU changed)

Koxx3, 2020-09-19:
Hello,
I am trying to write a chunk of datas (~40 bytes).
I changed the MTU but it's still refused. any idea why ?

The MTU request change seems to be accepted (to 64 bytes).

`
2020-09-19 13:44:00.769 6789-8440/com.welie.blessedexample I/BluetoothPeripheral: discovered 3 services for 'SmartLCD'
2020-09-19 13:44:00.772 6789-6789/com.welie.blessedexample I/BluetoothHandler: connected to 'SmartLCD'
2020-09-19 13:44:00.772 6789-6789/com.welie.blessedexample I/BluetoothHandler: discovered services
2020-09-19 13:44:00.774 6789-6789/com.welie.blessedexample D/BluetoothGatt: configureMTU() - device: 30:AE:A4:99:85:CA mtu: 185
2020-09-19 13:44:00.788 6789-8440/com.welie.blessedexample D/BluetoothGatt: onConfigureMTU() - Device=30:AE:A4:99:85:CA mtu=64 status=0
2020-09-19 13:44:00.789 6789-6789/com.welie.blessedexample I/BluetoothHandler: new MTU set: 64 / status: 0
2020-09-19 13:44:00.789 6789-6789/com.welie.blessedexample D/BluetoothGatt: requestConnectionPriority() - params: 1
2020-09-19 13:44:00.791 6789-6789/com.welie.blessedexample D/BluetoothPeripheral: requesting connection priority 1
2020-09-19 13:44:00.792 6789-6789/com.welie.blessedexample D/BluetoothGatt: setCharacteristicNotification() - uuid: beb5483e-36e1-4688-b7f5-ea07361b26a0 enable: true
2020-09-19 13:44:00.828 6789-6789/com.welie.blessedexample I/BluetoothHandler: SUCCESS: Notify set to 'on' for beb5483e-36e1-4688-b7f5-ea07361b26a0

...
2020-09-19 13:44:01.875 6789-6789/com.welie.blessedexample D/BluetoothPeripheral: writing <03010203000600000430303030001161613a62623a63633a64643a65653a6666b4640164020501fb0150001601> to characteristic <beb5483e-36e1-4688-b7f5-ea07361b26a9>
2020-09-19 13:44:06.903 6789-8440/com.welie.blessedexample E/BluetoothPeripheral: writing <03010203000600000430303030001161613a62623a63633a64643a65653a6666b4640164020501fb0150001601> to characteristic <beb5483e-36e1-4688-b7f5-ea07361b26a9> failed, status GATT ERROR
2020-09-19 13:44:06.909 6789-6789/com.welie.blessedexample I/BluetoothHandler: ERROR: Failed writing <03010203000600000430303030001161613a62623a63633a64643a65653a6666b4640164020501fb0150001601> to <beb5483e-36e1-4688-b7f5-ea07361b26a9>
2020-09-19 13:44:06.913 6789-8440/com.welie.blessedexample D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=4 device=30:AE:A4:99:85:CA
2020-09-19 13:44:06.920 6789-8440/com.welie.blessedexample I/BluetoothPeripheral: peripheral 'SmartLCD' disconnected with status 'GATT CONN TIMEOUT'
2020-09-19 13:44:06.921 6789-8440/com.welie.blessedexample D/BluetoothGatt: close()
2020-09-19 13:44:06.926 6789-8440/com.welie.blessedexample D/BluetoothGatt: unregisterApp() - mClientIf=4
2020-09-19 13:44:06.950 6789-6789/com.welie.blessedexample I/BluetoothHandler: disconnected 'SmartLCD' with status 8`

weliem, 2020-09-22:
I have no idea. Are you sure your device supports a larger MTU and that you are writing the correct bytes?

Koxx3, 2020-09-30:
Yes, it's an ESP32 and in the other direction, it works (ESP -> Samsung S10e).
How can I help to debug ?
My limit is clearly at 20 bytes. one more and it fails.

Koxx3, 2020-09-30:
to be clear, in the other direction, I manage to send 128 bytes without issue.

weliem, 2020-10-01:
I found this:

https://forum.developer.samsung.com/t/samsung-android-10-ble-connectivity-regression/509

Can you try your code on another phone?

weliem, 2020-10-31:
Any update on this issue?

weliem, 2020-11-12:
Closing issue due to no feedback

closed 2020-11-12 by weliem