weliem/blessed-android #54: ble device crashes on streaming data

sandeepcmsm, 2020-10-02:
Hi,
Thanks for your library.

I'm trying to streaming ECG data from the movesense HR+ device. Basically, we have implemented firmware on movesense platform and we are utilizing the service to issue commands and responses. 
one of the features is ECG data streaming, where it approximately streams 45bytes of data every 20ms. ble device is running at 512hz to stream the data. I will use this data to plot in a graph.

I'm building an app using the flutter platform and on the native side, I'm utilizing this library.  the device streaming starts it will start emitting the data from  "onCharacteristicUpdate" to the flutter side to plot it. But once the streaming start it will run for a certain time (10 to 60seconds) and ble device suddenly shutdowns. 

I have also tried plotting graph in the native app without using the flutter platform, there it works totally fine without crashing the device.  I can make the device crash by issue 1 to 2 seconds of thread sleep on "onCharacteristicUpdate". 

any thoughts why it is causing the ble device to crash? 

Thanks you.

weliem, 2020-10-02:
The onCharacteristicUpdate runs on the main thread. You cannot do a thread sleep on the main thread.

When you create BluetoothCentral, you can provide your own Handler object that runs on a different thread.

sandeepcmsm, 2020-10-02:
Thanks for the reply.

I'm not doing the thread sleep on the main thread instead I did in binder thread (for around 50ms) to reproduce the ble device crash. 
I'm expecting the android Bluetooth stack to take care of all the data pushed by the BLE device and hand it over to the app. 

in my scenario with the flutter app, I'm copying all the properties of BluetoothGattCharacteristic and serialize it into protobuf and transfer it to the flutter side. I have to do it in binder thread because as you mentioned https://medium.com/@martijn.van.welie/making-android-ble-work-part-3-117d3a8aee23, android reuses the object. due to this little processing its making the BLE device shut down in the middle of streaming. 

What might be the reasons for the BLE device to shut down,  is it causing due to buffer overload on the BLE device ??

Thanks.

weliem, 2020-10-02:
Hmmm, now I am confused. If you use Blessed, then the callback that Blessed does will not be on the Binder thread anymore. Blessed takes care of this issue...But you say you are doing it on the Binder thread, so does that mean you are not using Blessed?

The proper way would be to implement onCharacteristicUpdate (which will not be on Binder thread anymore, but now it will be on main thread) and do the protobuf serialization there. To make sure it is not on the main thread easy you need to create your own thread and create a Handler object with it. Then pass it when initializing BluetoothCentral.

Anyway, you also cannot do thread sleep on a Binder thread.....will indeed also cause a crash

sandeepcmsm, 2020-10-03:
Thanks weliem. 
Initially, I went with utilizing the android API I was faced with this problem. but after started using your library problem is solved. but I was curious to know what all might be the reasons which are causing the device to shut down. thanks for your articles.

closed 2020-10-03 by sandeepcmsm