weliem/blessed-android #55: Connection issue on SDK 22

rauliordache, 2020-10-08:
Hello,

Due to some problems in 7.0, I have downgraded my tablet to the stock version of Android 5.1 (Lollipop) - SDK 22. I cannot connect to my medical device from SDK-22:
```
01-05 20:27:08.597 749-779/? D/BluetoothManagerService: Message: 20
01-05 20:27:08.597 749-779/? D/BluetoothManagerService: Added callback: android.bluetooth.IBluetoothManagerCallback$Stub$Proxy@1deabccf:true
01-05 20:27:08.601 6158-6158/com.myapp D/BluetoothAdapter: isEnabled
01-05 20:27:08.602 5922-5960/? D/BluetoothAdapterService(563897221): getState() - mAdapterProperties: com.android.bluetooth.btservice.AdapterProperties@b9637f5
01-05 20:27:08.602 6158-6158/com.myapp D/BluetoothAdapter: 574927617: getState(). Returning 12
01-05 20:27:08.603 5922-5950/? D/BluetoothAdapterService(563897221): getState() - mAdapterProperties: com.android.bluetooth.btservice.AdapterProperties@b9637f5
01-05 20:27:08.603 6158-6158/com.myapp D/BluetoothAdapter: 574927617: getState(). Returning 12
01-05 20:27:08.605 5922-5942/? D/BluetoothAdapterService(563897221): getState() - mAdapterProperties: com.android.bluetooth.btservice.AdapterProperties@b9637f5
01-05 20:27:08.605 6158-6158/com.myapp D/BluetoothAdapter: 574927617: getState(). Returning 12
01-05 20:27:08.615 6158-6179/com.myapp D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=1
01-05 20:27:08.619 6158-6158/com.myapp I/BluetoothLeScanner: startRegisteration: mLeScanClients={com.welie.blessed.BluetoothCentral$2@3990b5a6=android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper@11c87e3d}
01-05 20:27:08.620 5922-5957/? D/BluetoothAdapterService(563897221): getState() - mAdapterProperties: com.android.bluetooth.btservice.AdapterProperties@b9637f5
01-05 20:27:08.620 5922-5957/? D/BluetoothAdapter: 644363702: getState(). Returning 12
01-05 20:27:08.621 5922-5957/? D/BluetoothAdapterService(563897221): getState() - mAdapterProperties: com.android.bluetooth.btservice.AdapterProperties@b9637f5
01-05 20:27:08.621 5922-5957/? D/BluetoothAdapter: 644363702: getState(). Returning 12
01-05 20:27:08.622 5922-5957/? D/BluetoothAdapterService: getAdapterService() - returning com.android.bluetooth.btservice.AdapterService@219c6385
01-05 20:27:08.622 5922-5957/? D/BluetoothAdapterService(563897221): getState() - mAdapterProperties: com.android.bluetooth.btservice.AdapterProperties@b9637f5
01-05 20:27:08.622 5922-5957/? D/BluetoothAdapter: 644363702: getState(). Returning 12
01-05 20:27:08.623 5922-5957/? I/BtGatt.ScanManager: configureScanFilters: client=com.android.bluetooth.gatt.ScanClient@20
01-05 20:27:08.623 5922-5957/? I/BtGatt.ScanManager: gattClientScanFilterEnableNative(com.android.bluetooth.gatt.ScanClient@20,true);
01-05 20:27:09.040 6158-6158/com.myapp D/BluetoothAdapter: isEnabled
01-05 20:27:09.090 5922-5948/? D/BluetoothRemoteDevices: addDeviceProperties: Remote Address = 34:81:F4:4B:E7:48
01-05 20:27:09.091 5922-5948/? D/BluetoothDevice: mAddress: 34:81:F4:4B:E7:48
01-05 20:27:09.091 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: bdDevice = 34:81:F4:4B:E7:48
01-05 20:27:09.091 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Acquire Lock - RemoteDevices.mObject
01-05 20:27:09.091 5922-5948/? D/BluetoothRemoteDevices: Send Broadcast: Device Name
01-05 20:27:09.091 749-889/? V/ActivityManager: Broadcast: Intent { act=android.bluetooth.device.action.NAME_CHANGED flg=0x4000010 (has extras) } ordered=false userid=0 callerApp=ProcessRecord{78cc718 5922:com.android.bluetooth/1002}
01-05 20:27:09.092 5922-5948/? D/BluetoothRemoteDevices: Remote Device name is: BerryMed
01-05 20:27:09.092 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Release Lock - RemoteDevices.mObject
01-05 20:27:09.092 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Acquire Lock - RemoteDevices.mObject
01-05 20:27:09.092 5922-5948/? D/BluetoothRemoteDevices: Remote Address is:34:81:F4:4B:E7:48
01-05 20:27:09.093 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Release Lock - RemoteDevices.mObject
01-05 20:27:09.093 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Acquire Lock - RemoteDevices.mObject
01-05 20:27:09.093 5922-5948/? D/BluetoothRemoteDevices: Send Broadcast: Device Class
01-05 20:27:09.093 5505-5505/? V/BluetoothEventManager: Received android.bluetooth.device.action.NAME_CHANGED
01-05 20:27:09.094 749-1450/? V/ActivityManager: Broadcast: Intent { act=android.bluetooth.device.action.CLASS_CHANGED flg=0x4000010 (has extras) } ordered=false userid=0 callerApp=ProcessRecord{78cc718 5922:com.android.bluetooth/1002}
01-05 20:27:09.095 5922-5948/? D/BluetoothRemoteDevices: Remote class is:0
01-05 20:27:09.095 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Release Lock - RemoteDevices.mObject
01-05 20:27:09.095 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Acquire Lock - RemoteDevices.mObject
01-05 20:27:09.095 5922-5948/? D/BluetoothRemoteDevices: Remote type is:3
01-05 20:27:09.095 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Release Lock - RemoteDevices.mObject
01-05 20:27:09.095 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Acquire Lock - RemoteDevices.mObject
01-05 20:27:09.095 5922-5948/? D/BluetoothRemoteDevices: Remote RSSI is:-56
01-05 20:27:09.095 5922-5948/? D/BluetoothRemoteDevices: devicePropertyChangedCallback: Release Lock - RemoteDevices.mObject
01-05 20:27:09.095 5922-5948/? D/BluetoothDevice: mAddress: 34:81:F4:4B:E7:48
01-05 20:27:09.095 5922-5948/? D/BluetoothRemoteDevices: deviceFoundCallback: Remote Address is:34:81:F4:4B:E7:48
01-05 20:27:09.096 5505-5505/? V/BluetoothEventManager: Received android.bluetooth.device.action.CLASS_CHANGED
01-05 20:27:09.096 749-1448/? V/ActivityManager: Broadcast: Intent { act=android.bluetooth.device.action.FOUND flg=0x10 (has extras) } ordered=false userid=0 callerApp=ProcessRecord{78cc718 5922:com.android.bluetooth/1002}
01-05 20:27:09.098 5505-5505/? V/BluetoothEventManager: Received android.bluetooth.device.action.FOUND
01-05 20:27:09.099 5922-5948/? D/BluetoothDevice: mAddress: 34:81:F4:4B:E7:48
01-05 20:27:09.099 5505-5505/? D/BluetoothEventManager: Device BerryMed ,Class: 0
01-05 20:27:09.102 5922-5942/? D/BluetoothDevice: mAddress: 34:81:F4:4B:E7:48
01-05 20:27:09.102 5922-5942/? D/BluetoothRemoteDevices: getAlias: mAlias = null
01-05 20:27:09.102 5505-5505/? D/BluetoothDevice: getAlias: alias = null
01-05 20:27:09.104 5922-5941/? D/BluetoothDevice: mAddress: 34:81:F4:4B:E7:48
01-05 20:27:09.104 5505-5505/? D/BluetoothDevice: getName: name = BerryMed
01-05 20:27:09.104 5505-5505/? D/BluetoothDevice: getAliasName: name = BerryMed
01-05 20:27:09.104 5505-5505/? D/CachedBluetoothDevice: fetchName, AlaisName is BerryMed
01-05 20:27:09.104 5505-5505/? D/CachedBluetoothDevice: fetchName, Return Name BerryMed
01-05 20:27:09.106 5922-5958/? D/BluetoothDevice: mAddress: 34:81:F4:4B:E7:48
01-05 20:27:09.106 5505-5505/? D/BluetoothDevice: getBluetoothClass: classInt = 0
01-05 20:27:09.106 5505-5505/? D/CachedBluetoothDevice: fetchClass, mBtClass is 0
01-05 20:27:09.107 5922-5960/? D/BluetoothDevice: mAddress: 34:81:F4:4B:E7:48
01-05 20:27:09.108 5505-5505/? D/CachedBluetoothDevice: Bluetooth device get uuid is null
01-05 20:27:09.108 5505-5505/? D/BluetoothDevice: mAddress: 34:81:F4:4B:E7:48
01-05 20:27:09.108 5505-5505/? D/BluetoothDevice: mAddress: 34:81:F4:4B:E7:48
[... 112 lines cut ...]
D/BluetoothGatt: connect() - device: 34:81:F4:4B:E7:48, auto: true
    registerApp()
D/BluetoothGatt: registerApp() - UUID=7382b766-e028-4eaa-8999-627a8014eb2c
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=34:81:F4:4B:E7:48
D/BluetoothDevice: getBondState: state = 12
D/BluetoothDevice: getName: name = BerryMed
D/BluetoothDevice: getName: name = BerryMed
D/BluetoothGatt: discoverServices() - device: 34:81::4B:E7:48
D/BluetoothGatt: onSearchComplete() = Device=34:81:F4:4B:E7:48 Status=0
D/BluetoothDevice: getName: name = BerryMed
D/BluetoothGatt: requestConnectionPriority() - params: 1
D/BluetoothGatt: setCharacteristicNotification() - uuid: 49535343-1e4d-4bd9-ba61-23c647249616 enable: true
D/BluetoothGatt: writeDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb
D/BluetoothGatt: onDescriptorWrite() - Device=34:81:F4:4B:E7:48 handle=56
D/BluetoothGatt: onNotify() - Device=34:81:F4:4B:E7:48 handle=55
D/BluetoothGatt: onNotify() - Device=34:81:F4:4B:E7:48 handle=55
D/BluetoothGatt: onNotify() - Device=34:81:F4:4B:E7:48 handle=55
D/BluetoothGatt: onNotify() - Device=34:81:F4:4B:E7:48 handle=55
D/BluetoothGatt: onNotify() - Device=34:81:F4:4B:E7:48 handle=55
D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=7 device=34:81:F4:4B:E7:48
```

Can you help me, at least, with the 5.1(SDK22) version?
Thanks.

weliem, 2020-10-08:
And what does this have to do with the Blessed library? You don't seem to be using it.....

rauliordache, 2020-10-08:
I am using it ...
 ```
/**
     * Start the module. It starts the parsing thread and also scans & connects to a BerryMed device
     */
    private void start(){
   
        // Create BluetoothCentral
        central = new BluetoothCentral(context, bluetoothCentralCallback, new Handler());

        // Scan for peripherals with a certain service SSID or MAC
        central.startPairingPopupHack();
        central.scanForPeripheralsWithNames(new String[]{"BerryMed"});
        //central.scanForPeripheralsWithAddresses(new String[]{"34:81:F4:4B:E7:48"});
    }
```

weliem, 2020-10-08:
That's odd, I don't see any logs that Blessed would normally create. Did you include Timber as a dependency in your project?

The second log you post looks normal but in the first one, discoverServices() isn't even called. It seems the connection breaks directly after being connected...

weliem, 2020-10-10:
Hi, ok now we can see better what is happening.

The basically problem I see is that in the first case, the device seems to connect fairly quickly but the bluetooth stack never calls onClientConnectionState(). So you see:

01-06 05:26:16.242 29556-29908/? D/BluetoothRemoteDevices: aclStateChangeCallback: State:Connected to Device:34:81:F4:4B:E7:48

But there is no call to onClientConnectionState(). So Blessed doesn't get to know the device actually connected. Hence after 35 seconds the timeout goes off and a disconnect is issued.

In the second case the onClientConnectionState is called normally:

rauliordache, 2020-10-10:
Doing from the first time an autoConnect() (instead of connect()) in BluetoothCentral - connectPeripheral(...), works (with a little delay of 4-5 s)

https://github.com/weliem/blessed-android/blob/master/blessed/src/main/java/com/welie/blessed/BluetoothCentral.java#L595

// Now connect
peripheral.autoConnect(); // instead of peripheral.connect();

I will try to find the issue in peripheral.connect();

weliem, 2020-10-31:
Any update on this issue?

rauliordache, 2020-10-31:
I am using peripheral.autoConnect() instead of peripheral.connect() (even from the first attempt). 
I will let you know if I find the issue in .connect().

closed 2020-12-15 by weliem