weliem/blessed-android #58: connect ble success, but peripheral.getName()  = 'null'

luning2016, 2020-11-05:
Hello， I use blessed-android ， phone  connect ble success, but peripheral.getName()  = 'null' 

W/BluetoothCentral: peripheral with address '85:AA:CA:04:04:99' is not in the Bluetooth cache, hence connection may fail
I/BluetoothPeripheral: connect to 'null' (85:AA:CA:04:04:99) using TRANSPORT_LE
D/BluetoothGatt: connect() - device: 85:AA:CA:04:04:99, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=9941b794-2de8-4673-9653-b2da35d19bc4
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=85:AA:CA:04:04:99
I/BluetoothPeripheral: connected to 'null' (BOND_NONE) in 3.5s
D/BluetoothPeripheral: discovering services of 'null' with delay of 0 ms
D/BluetoothGatt: discoverServices() - device: 85:AA:CA:04:04:99
D/BluetoothGatt: onConnectionUpdated() - Device=85:AA:CA:04:04:99 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onSearchComplete() = Device=85:AA:CA:04:04:99 Status=0
I/BluetoothPeripheral: discovered 6 services for 'null'
I/BluetoothHandler: connected to 'null'
I/BluetoothHandler: discovered services bleNodes=1
D/BluetoothGatt: configureMTU() - device: 85:AA:CA:04:04:99 mtu: 185
D/BluetoothGatt: onConnectionUpdated() - Device=85:AA:CA:04:04:99 interval=39 latency=0 timeout=500 status=0
D/BluetoothGatt: onConfigureMTU() - Device=85:AA:CA:04:04:99 mtu=185 status=0
I/BluetoothHandler: new MTU set: 185
D/BluetoothGatt: requestConnectionPriority() - params: 1
D/BluetoothPeripheral: requesting connection priority 1
D/BluetoothGatt: setCharacteristicNotification() - uuid: 0783b03e-8535-b5a0-7140-a304d2495cb1 enable: true
I/BluetoothHandler: SUCCESS: Notify set to 'on' for 0783b03e-8535-b5a0-7140-a304d2495cb1
D/BluetoothGatt: onConnectionUpdated() - Device=85:AA:CA:04:04:99 interval=12 latency=0 timeout=2000 status=0

weliem, 2020-11-05:
Well, it look like your peripheral is not advertising a name.

You can doublecheck with other apps like nRF Connect

closed 2020-11-06 by weliem

luning2016, 2020-11-30:
Thank you for your reply. I use nRF Connect can get the correct name and mac address 。I use BLESSED scanning methods can get the same name and mac address as  nRF Connect  result ，then use  BluetoothPeripheral peripheral = central.getPeripheral(macAddr);
        central.connectPeripheral(peripheral, peripheralCallback); can connect Bluetooth device  correctly ，peripheral.getName()  is 
 no 'null' .     Here is a strange phenomenon, when it happens, I cloose my app, use nRF Connect to  scan and connect the device, After the normal connection ,I closse nRF Connect  ,then reopen my app, Connect to the Bluetooth device, this time can get the right name