weliem/blessed-android #59: Handling spurious connection events

niladrieko, 2020-11-06:
Hi,

We have seen that the Android BLE stack sometimes sends spurious connection events. Specifically with the Samsung s9 when we disconnect the peripheral by switching it off we first get a `STATE_DISCONNECTED` callback immediately followed by a `STATE_CONNECTED`. This causes blessed to think that it is connected when it is not. So when you actually try to connect to the peripheral it ignores the connection request. The whole flow  can be seen here https://www.dropbox.com/s/eupbyt6xut80c8q/Blessed%20behavior.mp4?dl=0 Apparently other have also run into this issue https://sweetblue.io/docs/Android-BLE-Issues :

"You can get a connection success callback but device isn’t actually connected, and vice versa for disconnect."

What are your thoughts on handling this by ignoring `STATE_CONNECTED` if it was immediately preceded by a `STATE_DISCONNECTED` ?

weliem, 2020-11-06:
Hi, hmmmm I have never seen that...

Does it only happen on the Samsung S9? Or does it also happen on other (non Samsung) phones?

Can you try to reproduce it again, but this time don't put any breakpoints and please share the log? I would like to see that...because putting breakpoints on Binder threads is already problematic by itself...

niladrieko, 2020-11-06:
We think we have seen this on the Nexus 5X in the past but at the moment we can only reproduce this on the S9. Here is the log:

2020-11-16 21:36:11.758 W/BluetoothCentral: already issued autoconnect for '88:D2:11:DC:FA:CB' 
2020-11-16 21:36:11.761 I/BluetoothPeripheral: autoConnect to 'Eko DUO v3.6' (88:D2:11:DC:FA:CB) using TRANSPORT_LE
2020-11-16 21:36:11.761 D/BluetoothAdapter: STATE_ON
2020-11-16 21:36:11.763 D/BluetoothGatt: connect() - device: 88:D2:11:DC:FA:CB, auto: true
2020-11-16 21:36:11.763 D/BluetoothAdapter: isSecureModeEnabled
2020-11-16 21:36:11.763 D/BluetoothGatt: registerApp()
2020-11-16 21:36:11.763 D/BluetoothGatt: registerApp() - UUID=388506f0-7a2d-4506-a47a-07a5b6dd62ac
2020-11-16 21:36:11.765 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
2020-11-16 21:36:12.391 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=88:D2:11:DC:FA:CB
2020-11-16 21:36:12.416 I/BluetoothPeripheral: connected to 'Eko DUO v3.6' (BOND_NONE) in 0.6s
2020-11-16 21:36:12.423 D/BluetoothPeripheral: discovering services of 'Eko DUO v3.6' with delay of 0 ms
2020-11-16 21:36:12.424 D/BluetoothGatt: discoverServices() - device: 88:D2:11:DC:FA:CB
2020-11-16 21:36:13.033 D/BluetoothGatt: onConnectionUpdated() - Device=88:D2:11:DC:FA:CB interval=15 latency=0 timeout=500 status=0
2020-11-16 21:36:13.207 D/BluetoothGatt: onConnectionUpdated() - Device=88:D2:11:DC:FA:CB interval=6 latency=0 timeout=500 status=0
2020-11-16 21:36:13.411 D/BluetoothGatt: onSearchComplete() = Device=88:D2:11:DC:FA:CB Status=0
2020-11-16 21:36:13.413 I/BluetoothPeripheral: discovered 4 services for 'Eko DUO v3.6'
2020-11-16 21:36:13.423 D/BluetoothGatt: requestConnectionPriority() - params: 1
2020-11-16 21:36:13.424 D/BluetoothPeripheral: requesting connection priority 1
2020-11-16 21:36:13.424 D/BluetoothGatt: setCharacteristicNotification() - uuid: ba9c5360-9999-11e3-966f-0002a5d5c51b enable: true
2020-11-16 21:36:13.477 D/BluetoothGatt: onConnectionUpdated() - Device=88:D2:11:DC:FA:CB interval=15 latency=0 timeout=500 status=0
2020-11-16 21:36:13.610 D/BluetoothGatt: onConnectionUpdated() - Device=88:D2:11:DC:FA:CB interval=12 latency=0 timeout=2000 status=0
2020-11-16 21:36:13.779 D/BluetoothPeripheral: writing <80000000> to characteristic <ba9c5360-9999-11e3-966f-0002a5d5c51b>
2020-11-16 21:36:14.013 I/BluetoothCentral: no scan to stop because no scan is running
2020-11-16 21:36:25.149 I/BluetoothPeripheral: force disconnect 'Eko DUO v3.6' (88:D2:11:DC:FA:CB)
2020-11-16 21:36:25.149 D/BluetoothGatt: cancelOpen() - device: 88:D2:11:DC:FA:CB
2020-11-16 21:36:25.172 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=88:D2:11:DC:FA:CB
2020-11-16 21:36:25.189 I/BluetoothPeripheral: disconnected 'Eko DUO v3.6' on request
2020-11-16 21:36:25.189 D/BluetoothGatt: close()
2020-11-16 21:36:25.200 D/BluetoothGatt: unregisterApp() - mClientIf=5
2020-11-16 21:36:25.285 I/BluetoothPeripheral: autoConnect to 'Eko DUO v3.6' (88:D2:11:DC:FA:CB) using TRANSPORT_LE
2020-11-16 21:36:25.286 D/BluetoothAdapter: STATE_ON
2020-11-16 21:36:25.287 D/BluetoothGatt: connect() - device: 88:D2:11:DC:FA:CB, auto: true
2020-11-16 21:36:25.287 D/BluetoothAdapter: isSecureModeEnabled
2020-11-16 21:36:25.288 D/BluetoothGatt: registerApp()
2020-11-16 21:36:25.288 D/BluetoothGatt: registerApp() - UUID=173b9477-1967-45d9-84c8-d663ead172f1
2020-11-16 21:36:25.292 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
2020-11-16 21:36:25.311 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=88:D2:11:DC:FA:CB
2020-11-16 21:36:25.324 I/BluetoothPeripheral: connected to 'Eko DUO v3.6' (BOND_NONE) in 0.0s
2020-11-16 21:36:25.328 D/BluetoothPeripheral: discovering services of 'Eko DUO v3.6' with delay of 0 ms
2020-11-16 21:36:25.328 D/BluetoothGatt: discoverServices() - device: 88:D2:11:DC:FA:CB
2020-11-16 21:36:27.297 D/BluetoothAdapter: STATE_ON
2020-11-16 21:36:27.303 D/BluetoothAdapter: STATE_ON
2020-11-16 21:36:27.305 D/BluetoothAdapter: BLE support array set: 010011
2020-11-16 21:36:27.305 D/BluetoothLeScanner: Start Scan with callback
2020-11-16 21:36:27.313 D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=9 mScannerId=0
2020-11-16 21:36:27.326 I/BluetoothCentral: scan started
2020-11-16 21:36:27.333 I/BluetoothPeripheral: force disconnect 'Eko DUO v3.6' (88:D2:11:DC:FA:CB)
2020-11-16 21:36:27.333 D/BluetoothGatt: cancelOpen() - device: 88:D2:11:DC:FA:CB
2020-11-16 21:36:27.350 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=88:D2:11:DC:FA:CB
2020-11-16 21:36:27.361 I/BluetoothPeripheral: disconnected 'Eko DUO v3.6' on request
2020-11-16 21:36:27.361 D/BluetoothGatt: close()
2020-11-16 21:36:27.367 D/BluetoothGatt: unregisterApp() - mClientIf=5
2020-11-16 21:36:28.348 D/BluetoothAdapter: STATE_ON
2020-11-16 21:36:28.348 D/BluetoothLeScanner: Stop Scan with callback
2020-11-16 21:36:28.351 I/BluetoothCentral: scan stopped
2020-11-16 21:36:28.440 I/BluetoothPeripheral: connect to 'Eko DUO v3.6' (88:D2:11:DC:FA:CB) using TRANSPORT_LE
2020-11-16 21:36:28.441 D/BluetoothAdapter: STATE_ON
2020-11-16 21:36:28.445 D/BluetoothGatt: connect() - device: 88:D2:11:DC:FA:CB, auto: false
2020-11-16 21:36:28.445 D/BluetoothAdapter: isSecureModeEnabled
2020-11-16 21:36:28.445 D/BluetoothGatt: registerApp()
2020-11-16 21:36:28.446 D/BluetoothGatt: registerApp() - UUID=50262cd3-691c-4f06-887c-ea5f00add0df
[... cut ...]
2020-11-16 21:37:55.328 D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=5 device=88:D2:11:DC:FA:CB
2020-11-16 21:37:55.369 I/BluetoothPeripheral: peripheral 'Eko DUO v3.6' disconnected with status 'GATT CONN TIMEOUT' before completing service discovery
2020-11-16 21:37:55.369 D/BluetoothGatt: close()
2020-11-16 21:37:55.383 D/BluetoothGatt: unregisterApp() - mClientIf=5
2020-11-16 21:37:55.409 I/BluetoothCentral: connection to 'Eko DUO v3.6' (88:D2:11:DC:FA:CB) failed
2020-11-16 21:38:01.939 I/BluetoothCentral: no scan to stop because no scan is running

weliem, 2020-11-06:
Well this is giving a different perspective....

- You connect and you get connected
- Then your application calls cancelConnection() at (2020-11-16 21:36:25.149 I/BluetoothPeripheral: force disconnect 'Eko DUO v3.6' (88:D2:11:DC:FA:CB)
- It disconnects
- You immediately call autoConnectPeripheral() again at (2020-11-16 21:36:25.285 I/BluetoothPeripheral: autoConnect to 'Eko DUO v3.6' (88:D2:11:DC:FA:CB) using TRANSPORT_LE
- It very quickly gets connected again...within 1 sec...
- Then the service discovery doesn't complete because disconnect is called again...

So I think you need to review you application logic....
- Why is the first cancelConnection() happening? Looks like it shouldn't happen.
- Try removing the autoConnectPeripheral after disconnection, or at least delay it a bit...

jmzorko, 2020-11-06:
The application calls cancelConnection() bc it didn't receive any data from our device (which sends ADPCM audio at 4khz) for 2 seconds (bc the user turned it off). When it "very quickly gets connected again", the device is _off_. Android is sending the STATE_CONNECTED event when the device is _off_.

weliem, 2020-11-06:
Ok, then I would recommend adding a delay of +/- 5 seconds before calling autoConnectPeripheral again....that will probably fix the issue

weliem, 2020-11-09:
I think that the following is happening....

- You issue a disconnect while the device is not connected anymore. The Samsung stack tries to send a disconnect command but never gets a reply...
- But although it never get's a reply, it tells Blessed that we are disconnected. But internally it doesn't seem to update the state in the Samsung BLE stack correctly.
- When you issue an autoConnect, it seems to think we are actually still connected to this device and send Blessed a 'connected' event. 
- But since we are not connected, the service discovery starts failing and then everything fails...

So indeed a bug in the Samsung BLE stack. I am going to try to reproduce it because it seems unrelated to your 'Eko' device.

The reason I think adding a delay will fix it, is because according to the connection parameters you received (in the log), the stack should disconnect anyway after 2000ms....So if you add a couple of more seconds delay, the Samsung stack will hopefully figure out by itself that the device has disconnected. You might have to experiment a bit with how much delay is needed...

weliem, 2020-11-09:
The other thing I can think of is that you make the 'turn off' a bit more graceful. So instead of simply powering off, you send a disconnect from the 'Eko' side....and then power off. That would let the Samsung phone know you have disconnected....

But I am not sure you have control over the firmware to do that.

jmzorko, 2020-11-09:
Yes, if we disconnect from the app side, these issues are not as prevalent. However, this is not something we can expect from our users. We do create and regularly update the firmware for our devices (and have had to make adjustments to said fw to deal with weird Samsung behavior twice before), but we still have to deal with the users running the old fw. Also, apparently adding the delay before auto connect doesn't fix the ghost connection issue. Plus, our devices do not require the phone, so we can't require that the phone respond to any command sent by the device before the device will turn off.

weliem, 2020-11-09:
So adding delays had no effect? Not even a 1 minute delay? At some point the stack should figure out that there is no connection anymore...

Alternatively, what happens if you don't issue a cancelConnection at all? Also in that case the stack should send a disconnected event after max 20 seconds....

weliem, 2020-11-27:
Any update on this?

niladrieko, 2020-12-01:
Sorry for the late reply. It does seem to work if we delay the auto connection by 30 seconds.

weliem, 2020-12-01:
Ok good to hear! On most Android phones the connection timeout is 20s. So with 30s you should be safe indeed...

closed 2020-12-05 by weliem