weliem/blessed-android #60: leaked IntentReceiver com.welie.blessed.BluetoothCentral

debduttaVxplore, 2020-11-12:
Activity has **leaked** IntentReceiver com.welie.blessed.BluetoothCentral$10@f57c7cd that was originally registered here. Are you missing a call to unregisterReceiver()?
    android.app.IntentReceiverLeaked: Activity com.nextergo.activities.StandalonePresenceDeviceIdInputActivity has leaked IntentReceiver com.welie.blessed.BluetoothCentral$10@f57c7cd that was originally registered here. Are you missing a call to unregisterReceiver()?

debduttaVxplore, 2020-11-12:
2020-11-12 18:30:47.806 9869-9869/com.nextergo E/DecorView: mWindow.mActivityCurrentConfig is null
2020-11-12 18:30:47.813 9869-9869/com.nextergo E/DecorView: mWindow.mActivityCurrentConfig is null
2020-11-12 18:30:49.858 1268-4231/? E/InputDispatcher: Window handle Window{70cd9fc u0 com.nextergo/com.nextergo.activities.PresenceCalibrationActivity} has no registered input channel
2020-11-12 18:30:49.876 1268-1313/? E/InputDispatcher: Window handle Window{70cd9fc u0 com.nextergo/com.nextergo.activities.PresenceCalibrationActivity} has no registered input channel
2020-11-12 18:30:49.939 9869-9869/com.nextergo E/ViewRootImpl: sendUserActionEvent() mView returned.
2020-11-12 18:30:49.966 664-17335/? E/TrafficController: FREECESS
2020-11-12 18:30:50.026 1268-1429/? E/WindowManager: win=Window{70cd9fc u0 com.nextergo/com.nextergo.activities.PresenceCalibrationActivity EXITING} destroySurfaces: appStopped=false win.mWindowRemovalAllowed=true win.mRemoveOnExit=true win.mViewVisibility=0 caller=com.android.server.wm.AppWindowToken.destroySurfaces:1234 com.android.server.wm.AppWindowToken.destroySurfaces:1215 com.android.server.wm.WindowState.onExitAnimationDone:5100 com.android.server.wm.WindowStateAnimator.onAnimationFinished:308 com.android.server.wm.WindowState.onAnimationFinished:5541 com.android.server.wm.-$$Lambda$yVRF8YoeNdTa8GR1wDStVsHu8xM.run:2 com.android.server.wm.SurfaceAnimator.lambda$getFinishedCallback$0$SurfaceAnimator:100 
2020-11-12 18:30:51.543 9869-9869/com.nextergo E/DecorView: mWindow.mActivityCurrentConfig is null
2020-11-12 18:30:51.552 9869-9869/com.nextergo E/DecorView: mWindow.mActivityCurrentConfig is null
2020-11-12 18:30:53.581 1268-4231/? E/InputDispatcher: Window handle Window{2b2fae1 u0 com.nextergo/com.nextergo.activities.PresenceCalibrationActivity} has no registered input channel
2020-11-12 18:30:53.597 1268-2320/? E/InputDispatcher: Window handle Window{2b2fae1 u0 com.nextergo/com.nextergo.activities.PresenceCalibrationActivity} has no registered input channel
2020-11-12 18:30:53.611 9869-9869/com.nextergo E/ViewRootImpl: sendUserActionEvent() mView returned.
2020-11-12 18:30:53.749 1268-1429/? E/WindowManager: win=Window{2b2fae1 u0 com.nextergo/com.nextergo.activities.PresenceCalibrationActivity EXITING} destroySurfaces: appStopped=false win.mWindowRemovalAllowed=true win.mRemoveOnExit=true win.mViewVisibility=0 caller=com.android.server.wm.AppWindowToken.destroySurfaces:1234 com.android.server.wm.AppWindowToken.destroySurfaces:1215 com.android.server.wm.WindowState.onExitAnimationDone:5100 com.android.server.wm.WindowStateAnimator.onAnimationFinished:308 com.android.server.wm.WindowState.onAnimationFinished:5541 com.android.server.wm.-$$Lambda$yVRF8YoeNdTa8GR1wDStVsHu8xM.run:2 com.android.server.wm.SurfaceAnimator.lambda$getFinishedCallback$0$SurfaceAnimator:100 
2020-11-12 18:30:54.353 1996-1996/? E/wpa_sslib: Unknown driver command received: SET_FCC_CHANNEL -1
2020-11-12 18:30:54.788 2010-2897/? E/bt_stack: [ERROR:bta_gattc_utils.cc(434)] bta_gattc_mark_bg_conn unable to find the bg connection mask for bd_addr=c9:87:6b:4a:20:8e
2020-11-12 18:30:54.794 2010-19055/? E/BtGatt.ContextMap: remove() - removed: 14
2020-11-12 18:30:54.803 736-764/? E/ANDR-PERF-UTIL: Failed to update node /proc/9869/sched_boost with value 3
2020-11-12 18:30:54.803 736-764/? E/ANDR-PERF-OPTSHANDLER: can't boost task 9869
2020-11-12 18:30:54.803 736-764/? E/ANDR-PERF-RESOURCEQS: Failed to apply optimization [3, 32]
2020-11-12 18:30:54.931 2010-2648/? E/BtGatt.GattService: [GSIM LOG]: gsimLogHandler, msg: MESSAGE_SCAN_START, appName: com.nextergo, scannerId: 13, reportDelayMillis=0
2020-11-12 18:30:54.953 14533-14533/? E/Zygote: isWhitelistProcess - Process is Whitelisted
2020-11-12 18:30:54.954 14533-14533/? E/Zygote: accessInfo : 1
2020-11-12 18:30:54.964 14533-14533/? E/com.samsung.SM: Not starting debugger since process cannot load the jdwp agent.
2020-11-12 18:30:54.986 9869-9869/com.nextergo E/DecorView: mWindow.mActivityCurrentConfig is null
2020-11-12 18:30:55.000 2010-2177/? E/BluetoothRemoteDevices: Remote class is UNCATEGORIZED
2020-11-12 18:30:55.018 664-1114/? E/Netd: getNetworkForDns: getNetId from enterpriseCtrl is netid 0
2020-11-12 18:30:55.020 1268-1905/? E/NetdEventListenerService: handleMessage: { when=0 what=10001 obj=com.android.server.connectivity.NetdEventListenerService$DnsResultParams@6905b7f target=com.android.server.connectivity.NetdEventListenerService$DnsEventHandler }
2020-11-12 18:30:55.088 664-1114/? E/Netd: getNetworkForDns: getNetId from enterpriseCtrl is netid 0
2020-11-12 18:30:55.089 1268-1905/? E/NetdEventListenerService: handleMessage: { when=0 what=10001 obj=com.android.server.connectivity.NetdEventListenerService$DnsResultParams@7ab1c38 target=com.android.server.connectivity.NetdEventListenerService$DnsEventHandler }
2020-11-12 18:30:55.120 664-1114/? E/Netd: getNetworkForDns: getNetId from enterpriseCtrl is netid 0
2020-11-12 18:30:55.122 1268-1905/? E/NetdEventListenerService: handleMessage: { when=-1ms what=10001 obj=com.android.server.connectivity.NetdEventListenerService$DnsResultParams@c193a50 target=com.android.server.connectivity.NetdEventListenerService$DnsEventHandler }
2020-11-12 18:30:55.182 664-1114/? E/Netd: getNetworkForDns: getNetId from enterpriseCtrl is netid 0
[... cut ...]
2020-11-12 18:35:04.464 2440-2828/? E/GlobalSettingsRepoBase: No matched key : sprint_ir92_support
2020-11-12 18:35:06.206 1996-1996/? E/wpa_sslib: Unknown driver command received: SET_FCC_CHANNEL -1
2020-11-12 18:35:10.030 664-17335/? E/TrafficController: FREECESS
2020-11-12 18:35:12.053 1996-1996/? E/wpa_sslib: Unknown driver command received: SET_FCC_CHANNEL -1
2020-11-12 18:35:12.112 2440-2828/? E/GlobalSettingsRepoBase: No matched key : sprint_ir92_support
2020-11-12 18:35:14.611 1996-1996/? E/wpa_sslib: Unknown driver command received: SET_FCC_CHANNEL -1

weliem, 2020-11-12:
It looks like you didn't call close() on BluetoothCentral. The close() method will do the cleanup...

debduttaVxplore, 2020-11-12:
Thanks I will try to fix

debduttaVxplore, 2020-11-12:
Yea, solved.
I forgot to central.close() somewhere

debduttaVxplore, 2020-11-12:
Actually the problem was that missed a central.close() somewhere. after putting the central.close() the problem fixed

closed 2020-11-12 by debduttaVxplore

weliem, 2020-11-12:
ok great