weliem/blessed-android #61: Blessed thinks peripheral is connected when it is not

arielm, 2020-11-19:
Hi,

First, thanks for sharing this great piece of code.

I'm using blessed on a Samsung S10e to connect to a peripheral running on Raspberry Pi.
When I exit the peripheral app on Raspberry Pi, and run it again, I would expect blessed to allow to reconnect to the peripheral but instead I get an "already connected to..." message.
It's like blessed is not aware of the disconnection that took place on the peripheral side.

How should I handle such a situation?

weliem, 2020-11-19:
It sounds like Blessed is not receiving a disconnected event.

Can you share a log dump so I can see what is happening?

arielm, 2020-11-19:
This is the log dump:

```
2020-11-19 22:08:32.357 D/BluetoothAdapter: STATE_ON
2020-11-19 22:08:32.357 D/BluetoothAdapter: STATE_ON
2020-11-19 22:08:32.358 D/BluetoothAdapter: STATE_ON
2020-11-19 22:08:32.358 D/BluetoothAdapter: BLE support array set: 111111
2020-11-19 22:08:32.359 D/BluetoothLeScanner: Start Scan with callback
2020-11-19 22:08:32.360 D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2020-11-19 22:08:32.362 I/BluetoothCentral: scan started
2020-11-19 22:08:35.069 I/BluetoothPeripheral: connect to 'Klydo' (DC:A6:32:79:C3:42) using TRANSPORT_LE
2020-11-19 22:08:35.079 D/BluetoothAdapter: STATE_ON
2020-11-19 22:08:35.082 D/BluetoothGatt: connect() - device: DC:A6:32:79:C3:42, auto: false
2020-11-19 22:08:35.082 D/BluetoothAdapter: isSecureModeEnabled
2020-11-19 22:08:35.083 D/BluetoothGatt: registerApp()
2020-11-19 22:08:35.084 D/BluetoothGatt: registerApp() - UUID=8618b353-eb78-471e-b090-6d9a2fdd175e
2020-11-19 22:08:35.092 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
2020-11-19 22:08:37.873 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=DC:A6:32:79:C3:42
2020-11-19 22:08:37.905 I/BluetoothPeripheral: connected to 'Klydo' (BOND_NONE) in 2.8s
2020-11-19 22:08:37.912 D/BluetoothPeripheral: discovering services of 'Klydo' with delay of 0 ms
2020-11-19 22:08:37.912 D/BluetoothGatt: discoverServices() - device: DC:A6:32:79:C3:42
2020-11-19 22:08:38.734 D/BluetoothGatt: onConnectionUpdated() - Device=DC:A6:32:79:C3:42 interval=6 latency=0 timeout=500 status=0
2020-11-19 22:08:38.958 D/BluetoothGatt: onSearchComplete() = Device=DC:A6:32:79:C3:42 Status=0
2020-11-19 22:08:38.964 I/BluetoothPeripheral: discovered 3 services for 'Klydo'
2020-11-19 22:08:39.001 D/BluetoothAdapter: STATE_ON
2020-11-19 22:08:39.001 D/BluetoothLeScanner: Stop Scan with callback
2020-11-19 22:08:39.010 I/BluetoothCentral: scan stopped
2020-11-19 22:08:39.039 D/BluetoothGatt: onConnectionUpdated() - Device=DC:A6:32:79:C3:42 interval=39 latency=0 timeout=500 status=0
```
Then I exit the peripheral app and run it again but blessed is not logging anything.
And then I'm re-scanning and try to reconnect to the peripheral:
```
2020-11-19 22:10:52.164 I/BluetoothCentral: no scan to stop because no scan is running
2020-11-19 22:10:52.236 D/BluetoothAdapter: STATE_ON
2020-11-19 22:10:52.236 D/BluetoothAdapter: STATE_ON
2020-11-19 22:10:52.236 D/BluetoothLeScanner: Start Scan with callback
2020-11-19 22:10:52.237 D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2020-11-19 22:10:52.238 I/BluetoothCentral: scan started
2020-11-19 22:10:56.291 W/BluetoothCentral: already connected to DC:A6:32:79:C3:42'
```

weliem, 2020-11-21:
So are you saying you see nothing in the log when you exit the peripheral app? Not even after 5 min?

arielm, 2020-11-21:
Nothing in the log, not even after 5 minutes.

weliem, 2020-11-21:
Then we are simply not receiving a diconnection event. Hence Blessed thinks you are still connected.

I dont really know how to deal with this at it seems to be a phone bug. I dont have an S10 but I will try to reproduce with my S20...

weliem, 2020-11-21:
I just tried with m S20 and some device I got laying around, but I can't reproduce the issue.

Did you double check if killing your peripheral app really disconnected the device? Bluez may keep the connection open although your app has been stopped. You can check this with bluetoothctl or just turn bluetooth off on your raspberry. If you get a disconnected event after turning bluetooth off, you were still connected...

arielm, 2020-11-22:
In the meantime, I've been using this horrible hack in order to reconnect to the same peripheral:

```
public void startScanningPeripherals()
{
    if (connectedPeripheral != null)
    {
        connectedPeripheral.clearServicesCache();

        try
        {
            Thread.sleep(250);
        }
        catch (InterruptedException e)
        {}

        connectedPeripheral = null;
        central = new BluetoothCentral(context, bluetoothCentralCallback, new Handler(Looper.getMainLooper()));
    }

    central.scanForPeripherals();
}
```
The most horrible part being re-instantiating the BluetoothCentral object...

arielm, 2020-11-22:
Sorry, I've missed your previous comment...

Indeed, when I turn bluetooth off manually on the Raspberry Pi, I get the following on the Android side:
```
2020-11-22 11:19:22.542 D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=10 device=DC:A6:32:79:C3:42
2020-11-22 11:19:22.584 I/BluetoothPeripheral: peripheral 'Klydo' disconnected with status 'GATT CONN TIMEOUT'
2020-11-22 11:19:22.584 D/BluetoothGatt: close()
2020-11-22 11:19:22.592 D/BluetoothGatt: unregisterApp() - mClientIf=10
```
It seems that the [BLE peripheral software](https://github.com/Douglas6/cputemp) I'm using is not doing anything to disconnect from bluetooth upon exit. I guess it corresponds to a real-world situation where the user is turning off the device instead of properly quitting. I thought this kind of scenario would be handled somehow...

weliem, 2020-11-22:
Yeah, Bluez is a bit peculiar....

So I guess there is no issue with the Blessed library?

arielm, 2020-11-22:
I wouldn't call it an issue, more like a design decision.

The question is: do want to allow reconnection to a presumably already connected peripheral?
I have tested such a scenario with CoreBluetooth on iOS and it works as intended (i.e. a connection is always established regardless of the previous connection state...)

I think Blessed should behave similarly. After all, proper disconnection from the peripheral side is rarely happening in the real-world.

weliem, 2020-11-22:
I struggle to understand why you would want to connect again to an already connected device....

In your case, you were really still connected, although you didnt expect that. You could also simply have called cancelConnection () to disconnect before connecting again.

Normally if a peripheral is 'turned off', the connection timeout automatically kicks in. On Android this is usually 20 seconds. So then the peripheral becomes disconnected....

arielm, 2020-11-22:
> You could also simply have called cancelConnection () to disconnect before connecting again.

This is the first idea that came to my mind actually, but it's not working as intended.

> Normally if a peripheral is 'turned off', the connection timeout automatically kicks in.

It's not the case with the peripheral I'm using. In any case, I think that some peripherals won't have the opportunity to properly close the bluetooth connection before being turned-off.

weliem, 2020-11-23:
The connection timeout seems to be working fine in your case. Once you turned off bluetooth on the Pi, it immediately disconnected. I have never seen a case where this didn't work.

The Pi is apparently keeping the connection alive when you close the peripheral app you are using. So what happened when you tried cancelConnection()? You seem to suggest it didn't disconnect....

arielm, 2020-11-23:
When I made a cancelConnection() and directly connected to the peripheral, I got an "already connected to..." message.

From that I understood that I need to give the system some time to effectively disconnect, so I used the onDisconnectedPeripheral() callback (which I received with status = 0), and there I made an attempt to reconnect.

But then, I received new onDisconnectedPeripheral() callback (with status = 22) so from there I had to make a new attempt to reconnect.

So it eventually works, but in term of user experience it's not ok because I show a toast upon every connection/disconnection event, and it looks like a mess (I could not find any heuristic to avoid showing 'CONNECTED TO DEVICE' twice).

arielm, 2020-11-23:
Adding some log dump to complete my previous comment:

Stage 1: initial connection to device
```
2020-11-23 16:41:05.229 D/BluetoothAdapter: STATE_ON
2020-11-23 16:41:05.230 D/BluetoothAdapter: STATE_ON
2020-11-23 16:41:05.230 D/BluetoothAdapter: BLE support array set: 111111
2020-11-23 16:41:05.230 D/BluetoothLeScanner: Start Scan with callback
2020-11-23 16:41:05.231 D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2020-11-23 16:41:05.232 I/BluetoothCentral: scan started
2020-11-23 16:41:06.655 I/BluetoothPeripheral: connect to 'Klydo' (DC:A6:32:79:C3:42) using TRANSPORT_LE
2020-11-23 16:41:06.661 D/BluetoothAdapter: STATE_ON
2020-11-23 16:41:06.666 D/BluetoothGatt: connect() - device: DC:A6:32:79:C3:42, auto: false
2020-11-23 16:41:06.666 D/BluetoothAdapter: isSecureModeEnabled
2020-11-23 16:41:06.667 D/BluetoothGatt: registerApp()
2020-11-23 16:41:06.668 D/BluetoothGatt: registerApp() - UUID=c18b0696-c68b-453a-b290-e28cb6f7003f
2020-11-23 16:41:06.673 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
2020-11-23 16:41:08.102 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=DC:A6:32:79:C3:42
2020-11-23 16:41:08.122 I/BluetoothPeripheral: connected to 'Klydo' (BOND_NONE) in 1.4s
2020-11-23 16:41:08.127 D/BluetoothPeripheral: discovering services of 'Klydo' with delay of 0 ms
2020-11-23 16:41:08.127 D/BluetoothGatt: discoverServices() - device: DC:A6:32:79:C3:42
2020-11-23 16:41:08.771 D/BluetoothGatt: onConnectionUpdated() - Device=DC:A6:32:79:C3:42 interval=6 latency=0 timeout=500 status=0
2020-11-23 16:41:08.934 D/BluetoothGatt: onSearchComplete() = Device=DC:A6:32:79:C3:42 Status=0
2020-11-23 16:41:08.941 I/BluetoothPeripheral: discovered 3 services for 'Klydo'
2020-11-23 16:41:08.984 D/BluetoothAdapter: STATE_ON
2020-11-23 16:41:08.984 D/BluetoothLeScanner: Stop Scan with callback
2020-11-23 16:41:08.990 I/BluetoothCentral: scan stopped
2020-11-23 16:41:09.026 D/BluetoothGatt: onConnectionUpdated() - Device=DC:A6:32:79:C3:42 interval=39 latency=0 timeout=500 status=0
```

Stage 2: attempt to reconnect to device by cancelling existing connection first
```
2020-11-23 16:41:14.242 I/BluetoothCentral: no scan to stop because no scan is running
2020-11-23 16:41:14.308 D/BluetoothAdapter: STATE_ON
2020-11-23 16:41:14.309 D/BluetoothAdapter: STATE_ON
2020-11-23 16:41:14.309 D/BluetoothLeScanner: Start Scan with callback
2020-11-23 16:41:14.309 D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2020-11-23 16:41:14.311 I/BluetoothCentral: scan started
2020-11-23 16:41:17.118 I/BluetoothPeripheral: force disconnect 'Klydo' (DC:A6:32:79:C3:42)
2020-11-23 16:41:17.119 D/BluetoothGatt: cancelOpen() - device: DC:A6:32:79:C3:42
2020-11-23 16:41:17.121 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=DC:A6:32:79:C3:42
2020-11-23 16:41:17.125 I/BluetoothPeripheral: disconnected 'Klydo' on request
2020-11-23 16:41:17.125 D/BluetoothGatt: close()
2020-11-23 16:41:17.128 D/BluetoothGatt: unregisterApp() - mClientIf=10
2020-11-23 16:41:17.242 I/BluetoothPeripheral: connect to 'Klydo' (DC:A6:32:79:C3:42) using TRANSPORT_LE
2020-11-23 16:41:17.246 D/BluetoothAdapter: STATE_ON
2020-11-23 16:41:17.249 D/BluetoothGatt: connect() - device: DC:A6:32:79:C3:42, auto: false
2020-11-23 16:41:17.250 D/BluetoothAdapter: isSecureModeEnabled
2020-11-23 16:41:17.250 D/BluetoothGatt: registerApp()
2020-11-23 16:41:17.250 D/BluetoothGatt: registerApp() - UUID=cd7b3ac8-acb5-4dc9-ac2a-dc2db8652702
2020-11-23 16:41:17.254 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10
2020-11-23 16:41:17.261 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=DC:A6:32:79:C3:42
2020-11-23 16:41:17.265 I/BluetoothPeripheral: connected to 'Klydo' (BOND_NONE) in 0.0s
2020-11-23 16:41:17.268 D/BluetoothPeripheral: discovering services of 'Klydo' with delay of 0 ms
2020-11-23 16:41:17.269 D/BluetoothGatt: discoverServices() - device: DC:A6:32:79:C3:42
2020-11-23 16:41:17.759 D/BluetoothGatt: onConnectionUpdated() - Device=DC:A6:32:79:C3:42 interval=6 latency=0 timeout=500 status=0
2020-11-23 16:41:17.924 D/BluetoothGatt: onSearchComplete() = Device=DC:A6:32:79:C3:42 Status=0
2020-11-23 16:41:17.927 I/BluetoothPeripheral: discovered 3 services for 'Klydo'
2020-11-23 16:41:18.044 D/BluetoothGatt: onConnectionUpdated() - Device=DC:A6:32:79:C3:42 interval=39 latency=0 timeout=500 status=0
2020-11-23 16:41:18.432 D/BluetoothAdapter: STATE_ON
2020-11-23 16:41:18.432 D/BluetoothLeScanner: Stop Scan with callback
2020-11-23 16:41:18.438 I/BluetoothCentral: scan stopped
2020-11-23 16:41:18.926 D/BluetoothGatt: onClientConnectionState() - status=22 clientIf=10 device=DC:A6:32:79:C3:42
2020-11-23 16:41:18.937 I/BluetoothPeripheral: peripheral 'Klydo' disconnected with status 'GATT CONN TERMINATE LOCAL HOST'
2020-11-23 16:41:18.937 D/BluetoothGatt: close()
2020-11-23 16:41:18.946 D/BluetoothGatt: unregisterApp() - mClientIf=10
2020-11-23 16:41:19.078 I/BluetoothPeripheral: connect to 'Klydo' (DC:A6:32:79:C3:42) using TRANSPORT_LE
2020-11-23 16:41:19.085 D/BluetoothAdapter: STATE_ON
2020-11-23 16:41:19.088 D/BluetoothGatt: connect() - device: DC:A6:32:79:C3:42, auto: false
2020-11-23 16:41:19.088 D/BluetoothAdapter: isSecureModeEnabled
2020-11-23 16:41:19.089 D/BluetoothGatt: registerApp()
2020-11-23 16:41:19.090 D/BluetoothGatt: registerApp() - UUID=4a50c476-489e-4ad1-9d5e-12a01566d51b
[... cut ...]
2020-11-23 16:41:20.802 D/BluetoothGatt: discoverServices() - device: DC:A6:32:79:C3:42
2020-11-23 16:41:21.446 D/BluetoothGatt: onConnectionUpdated() - Device=DC:A6:32:79:C3:42 interval=6 latency=0 timeout=500 status=0
2020-11-23 16:41:21.617 D/BluetoothGatt: onSearchComplete() = Device=DC:A6:32:79:C3:42 Status=0
2020-11-23 16:41:21.620 I/BluetoothPeripheral: discovered 3 services for 'Klydo'
2020-11-23 16:41:21.703 D/BluetoothGatt: onConnectionUpdated() - Device=DC:A6:32:79:C3:42 interval=39 latency=0 timeout=500 status=0
```

weliem, 2020-11-27:
To be honest I see nothing weird...that is, knowing that you stopped your peripheral app.

The Pi is refusing your connection....nothing Blessed can do about that.

closed 2020-12-05 by weliem