weliem/blessed-android #62: System PIN dialog not showing

DaxeshV, 2020-11-23:
BLE PIN checking; the system PIN dialog when the pairing request is happen missing.

weliem, 2020-11-23:
Showing the system PIN dialog is being done by Android, and not by the Blessed library.

DaxeshV, 2020-11-23:
Yes. But when i using library and make request for connect it connect without PIN.

DaxeshV, 2020-11-23:
Also is it support auto connect BLE when device back in range?

weliem, 2020-11-23:
The bluetooth device needs to send a message that it requires a PIN code. Otherwise the PIN dialog will not appear and apparently your device does not do that....so it is a problem with your device

DaxeshV, 2020-11-23:
So its automatic process right. for showing system dialog i need to add any code?

weliem, 2020-11-30:
No nothing needed from the app side...

closed 2020-11-30 by weliem