weliem/blessed-android #63: autoConnectPeripheral() gets stuck

Maartinsh, 2020-11-27:
So, for peripheral that has not been bonded yet I call `autoConnectPeripheral`, but it seems that for some reason it gets stuck here..

DEVICE MODEL:SM-G960F
ANDROID:29

```
11 27 11:43:23:001 BluetoothCentral: peripheral with address 'E6:F6:05:7E:6A:5D' not in Bluetooth cache, autoconnecting by scanning
11 27 11:43:23:005 BluetoothCentral: Bluetooth disabled
...
11 27 11:45:03:475 BluetoothCentral: bluetooth turned on
...
11 27 11:45:23:445 BluetoothCentral: peripheral with address 'E6:F6:05:7E:6A:5D' not in Bluetooth cache, autoconnecting by scanning
11 27 11:45:23:446 BluetoothCentral: peripheral already on list for reconnection
...
11 27 11:45:26:474 SBeaconManager: Auto connection
11 27 11:45:26:476 BluetoothCentral: peripheral with address 'E6:F6:05:7E:6A:5D' not in Bluetooth cache, autoconnecting by scanning
11 27 11:45:26:477 BluetoothCentral: peripheral already on list for reconnection
...
11 27 11:45:29:530 BluetoothCentral: peripheral with address 'E6:F6:05:7E:6A:5D' not in Bluetooth cache, autoconnecting by scanning
11 27 11:45:29:531 BluetoothCentral: peripheral already on list for reconnection
```

Proof that beacon is nearby and was scanned successfully:

`
11 27 11:45:29:115 BeaconManager: Beacon detected: E6:F6:05:7E:6A:5D majorId: 0 minorId: 38
`

Any idea what could cause this?

weliem, 2020-11-27:
Why are you calling autoConnectPeripheral() every 3 seconds?

Maartinsh, 2020-12-04:
I changed connection time period to 30 seconds and it seems to be working now.
One small extra question: what's the safest way to check which devices are connected? 
Can I use:   `central.connectedPeripherals.any { it.address == peripheralAddress }` or it's better to use OS supported methods?

weliem, 2020-12-05:
`connectedPeripherals` will give you the list of peripherals that connected through Blessed. If you also want other peripherals you will need to use the OS supported methods.

closed 2020-12-05 by weliem