weliem/blessed-android #65: connect ble success, but peripheral.getName() = 'null'  reopen #58

luning2016, 2020-12-09:
[https://github.com/weliem/blessed-android/issues/58#issue-736561560](url)
Hello， I use blessed-android ， phone connect ble success, but peripheral.getName() = 'null'

W/BluetoothCentral: peripheral with address '85:AA:CA:04:04:99' is not in the Bluetooth cache, hence connection may fail
I/BluetoothPeripheral: connect to 'null' (85:AA:CA:04:04:99) using TRANSPORT_LE
D/BluetoothGatt: connect() - device: 85:AA:CA:04:04:99, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=9941b794-2de8-4673-9653-b2da35d19bc4
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=85:AA:CA:04:04:99
I/BluetoothPeripheral: connected to 'null' (BOND_NONE) in 3.5s
D/BluetoothPeripheral: discovering services of 'null' with delay of 0 ms
D/BluetoothGatt: discoverServices() - device: 85:AA:CA:04:04:99
D/BluetoothGatt: onConnectionUpdated() - Device=85:AA:CA:04:04:99 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onSearchComplete() = Device=85:AA:CA:04:04:99 Status=0
I/BluetoothPeripheral: discovered 6 services for 'null'
I/BluetoothHandler: connected to 'null'
I/BluetoothHandler: discovered services bleNodes=1
D/BluetoothGatt: configureMTU() - device: 85:AA:CA:04:04:99 mtu: 185
D/BluetoothGatt: onConnectionUpdated() - Device=85:AA:CA:04:04:99 interval=39 latency=0 timeout=500 status=0
D/BluetoothGatt: onConfigureMTU() - Device=85:AA:CA:04:04:99 mtu=185 status=0
I/BluetoothHandler: new MTU set: 185
D/BluetoothGatt: requestConnectionPriority() - params: 1
D/BluetoothPeripheral: requesting connection priority 1
D/BluetoothGatt: setCharacteristicNotification() - uuid: 0783b03e-8535-b5a0-7140-a304d2495cb1 enable: true
I/BluetoothHandler: SUCCESS: Notify set to 'on' for 0783b03e-8535-b5a0-7140-a304d2495cb1
D/BluetoothGatt: onConnectionUpdated() - Device=85:AA:CA:04:04:99 interval=12 latency=0 timeout=2000 status=0

   I use nRF Connect can get the correct name and mac address .I use BLESSED scanning methods can get the same name and mac address as nRF Connect result ，then use BluetoothPeripheral peripheral = central.getPeripheral(macAddr);
central.connectPeripheral(peripheral, peripheralCallback); can connect Bluetooth device correctly ，peripheral.getName() is
no 'null' . Here is a strange phenomenon, when it happens, I cloose my app, use nRF Connect to scan and connect the device, After the normal connection ,I closse nRF Connect ,then reopen my app, Connect to the Bluetooth device, this time can get the right name
    Every time this happens, I log out of the app, use NRF Connect to search and connect the device once, then log out of NFR Connect, reopen our app, and get the name right

luning2016, 2020-12-10:
I can workaround it in this way, when the app start, I use BLESSED scanning methods start scan 3s,then stop scan, after this operation，connect ble by mac address success，and  peripheral.getName()  get the name right

weliem, 2020-12-10:
Hi,

As you can see in the first line of the log, your peripheral isn't cached by Android yet:
W/BluetoothCentral: peripheral with address '85:AA:CA:04:04:99' is not in the Bluetooth cache, hence connection may fail

Which suggests you are connecting to a device without scanning first. So if you scan using NRF connect, the cache would be updated indeed and then you see the name again.

Alternatively, you could try to use autoConnectPeripheral, which will first scan for the peripheral if it is uncached.

luning2016, 2020-12-11:
Thank you very much for your reply. The problem appears to have been solved. We're doing some more tests to make sure it's fixed

closed 2020-12-15 by luning2016