weliem/blessed-android #66: How to know if connection was not successful

ericbouchard, 2020-12-14:
Reading the code, if you call connect and it fails early (out of cache, wrong state etc...) it "return" and won't notify the caller of an error.

I would suggest to add a boolean return type and return false when the operation is not completed and true when it is completed so the caller can know to wait for the listener callback to complete.

Did not look everywhere, but if other methods have the same behavior i think it would be preferable to do the same.

weliem, 2020-12-14:
Yes, I think you are right. It would be more consistent to return a boolean when connecting. 

I will put it on my todo-list

weliem, 2020-12-14:
However, note that connecting is asynchronous. So you need to listen to the callbacks 'onConnectedPeripheral' or 'onConnectionFailed'....

closed 2020-12-19 by weliem