weliem/blessed-android #67: Samsung OS 10 devices pops up the paring dialog. No callback/delegate about paring allowed/Denied.

abhi-coditas, 2020-12-14:
First of all thanks to all the contributors to this special library for BLE.

Recently for Samsung OS 10, BluetoothCenteral is requesting the paring pop up, And before requesting the paring pop up I got device connected callback. So I'm unable to Identify, whether it's really connected or it's still in pairing mode.

How I can differentiate these two different states, whether the connection is actually done or it's still in progress to create a bond(Paring)?

weliem, 2020-12-14:
Connecting and bonding are two separate things. When you get the 'connected' callback, you are connected but bonding might still happen. When bonding starts you get the callback 'onBondingStarted' and when it completes you get the callback 'onBondingSucceeded'

abhi-coditas, 2020-12-14:
@weliem Thanks a lot. I got it :-)

closed 2020-12-14 by abhi-coditas