weliem/blessed-android #68: need option to disable Automatic Bonding on connectPeripheral

joshbohmer, 2021-01-04:
I have a use case where a successful authentication needs to be made on a public r/w characteristic before attempting to Bond. Without a successful auth challenge any bond attempt causes the remote to end the connection immediately.

I need to connect to the peripheral, subscribe to notifications on the Auth characteristic so I can read/write the auth challenge & only after a successful auth ack can I initiate the Bond.

I've attempted to look through the library but can't see when this auto bond is initiated, tho I'm sure I've missed something.

Thanks in advance for your input.

weliem, 2021-01-04:
Hi, the library doesn't initiate bonding when trying to read/write encrypted characteristic. Android does it by itself, so that is why you can't find it in the code...so you can't prevent that from happening.

Perhaps you could just connect and call `createBond()` before subscribing to notifications?

joshbohmer, 2021-01-04:
interesting. Before I started implementing blessed I was able to connect & subscribe without a bond attempt being made. I just got so tired of working around Android's quirks I wanted an easier way :)

thanks for the response, I appreciate its very much a fringe case

joshbohmer, 2021-01-04:
So it seems the Android initiated bonding is a result of attempting to read a characteristic that we don't yet have access to.  As yet my handler isn't attempting to read anything so I assume this is done elsewhere in the library as part of the discovery process. I'll keep looking but my guess is a check for permission before the read operation will solve it.

closed 2021-01-04 by joshbohmer