weliem/blessed-android #69: Sending payloads bigger than 512 bytes

trancee, 2021-01-15:
Hi @weliem 

I am following your project for some time and I am using it to develop an app to communicate with other devices (phones) via BLE.

I am glad you added support for long reads/writes but I was wondering how would it be possible to send payloads that are bigger than 512 bytes like transferring photo data?

I know that there are other projects that are able to do so, but not sure if you have any plans to add support in your library as well?

Thanks,
Phil.

weliem, 2021-01-16:
Hi Phil,

I am not sure if there is anything meaningful that can be done. The standard 'long writes' are part of the Bluetooth standard, which exactly defines how packets are to be chopped-up, how the peripheral knows when the last packet has been sent and how to reassemble the chunks to the full packet again.

As far as I am aware, other libraries provide a function to chop up the packet in smaller packets and then send them. However, there is **no guarantee at all** that the receiving peripheral will know what to do with them. So only if you also can control the peripheral, you can make this work. Unlike with long writes, there is no 'established' way of doing this.

So I think it is the best if people come up with their own 'protocol' for sending large pakets. Chopping up packets in MTU -3 sized packets is easy and make sure you send them with WRITE_WITHOUT_RESPONSE for maximum speed. And try to get the largest MTU your peripheral supports. But then you still need to figure out how to mark the start/end of your long write and the peripheral needs to implement the corresponding logic.

So I have  nothing planned for now, but if somebody comes with a good suggestion I am happy to add it...

trancee, 2021-01-16:
Thank you for your reply.

I understand that I probably have to implement my own way of doing large packet transfers. In my case at least, both parties are known (smartphones) and so I am not communicating with a beacon or smart device. I should be in control on how to define this protocol.

I will study some other libraries that have already implemented a certain way of transferring larger files, like AirShare. But I will keep using your library as it abstracts all the other things I do not want to waste my time on.

Thanks for doing this library, it has been very helpful so far and keep up with the great project!

weliem, 2021-01-16:
Ok, good luck with your protocol!

...by the way I am working on a BluetoothPeripheralManager class, so that you can also make your own peripherals with Blessed. Hope to add it soon...

weliem, 2021-01-16:
Tip: have a look at Nordic's DFU (firmware update) protocol....
https://developer.nordicsemi.com/nRF51_SDK/nRF51_SDK_v8.x.x/doc/8.0.0/s110/html/a00103.html

closed 2021-02-16 by weliem