weliem/blessed-android #70: Sending big payload to BLE device

Maartinsh, 2021-01-26:
Hi,
I went through the new updates on library and noticed, that you added exception whenever payload size exceeds 512 bytes. 
I do request maximum MTU whenever services are discovered, but that of course does not help fully in my case as there can be a payload of ~900 bytes in my use case.

Conclusion for me is that maximum is 512 and I should probably manually split these payloads and send to the device, but the weird thing is that if I comment out exception of "512" as maximum, I can see that I write ~800 bytes to the characteristic and callback responds with "successful writing". If there's the limit of 512, then why it succeeds in writing big chunks of bytes in my case?

weliem, 2021-01-26:
Hi Maartin,

According to the BLE standard, a characteristic can only hold up to 512 bytes. Internally, Android will allow up 600 bytes as you can see in the Android source code. But Android also never enforces the 512 limit. In your case, the receiving peripheral is apparently willing to deal with ~800 bytes and Android is letting it happen.

So what you see happening shouldn't be allowed according to the BLE standard. But if it works, it works...

Maartinsh, 2021-01-26:
Thanks for clarification.

closed 2021-01-26 by Maartinsh