weliem/blessed-android #71: Stop scanning when onPause()

mjm987, 2021-01-28:
Congratulation, very nice library!

I would like to stop scanning if the main activity is paused, so do I tried adding:

```
@Override
protected void onStop() {
      super.onStop();
      BluetoothCentralManager central = BluetoothHandler.getInstance(getApplicationContext()).central;
      central.stopScan();
}
```

But then when the activity is visible again, it does not start scanning.
Possibly onBluetoothAdapterStateChanged() might be called in onResume() but how could I prevent then a conflict with the app start?

mjm987, 2021-01-28:
Btw: 
1.) without stopping scan in onStop(), the blessed example scans forever in background - even when leaving the app by the BACK button. This drains battery.
2.) to get better scan results I use SCAN_MODE_LOW_LATENCY because SCAN_MODE_LOW_POWER scans only 500ms on each 5000ms interval.

weliem, 2021-01-28:
Hi,

Can you share a bit more? Where do you restart the scanning? In onResume?

And indeed, scanForPeripherals does not time out. So you have to stop it explicitly if you don't want to scan in the background. This is intentional as in most use case you actually want to scan even when the app is in the background..

mjm987, 2021-01-28:
yes, it should be done in onResume() but I haven't implemented yet, I guess modification of the BluetoothHandler or  BluetoothCentralManager is needed (?)

mjm987, 2021-02-01:
Above solution works for me and saves battery because I need to use SCAN_MODE_LOW_LATENCY which would otherwisedrain battery when the App is in background (CPU-Load ~10%).

closed 2021-02-01 by mjm987