weliem/blessed-android #74: BluetoothAdapter.getDefaultAdapter() crash when using emulator

moodysache, 2021-02-12:
Hi @weliem 

I found a crash when trying to instantiate BluetoothCentral. This happens when calling BluetoothAdapter.getDefaultAdapter() on an emulated device with Android 7.1 and Android 10. Is there something that can be done regarding this? 

      Stack trace: 
      Instance creation error : could not create instance for [Single:'com.welie.blessed.BluetoothCentral',qualifier:BluetoothCentral]: java.lang.NullPointerException: no bluetooth adapter found
    	java.util.Objects.requireNonNull(Objects.java:228)
    	com.welie.blessed.BluetoothCentral.<init>(BluetoothCentral.java:329)
    	org.koin.core.instance.InstanceFactory.create(InstanceFactory.kt:50)
    	org.koin.core.instance.SingleInstanceFactory.create(SingleInstanceFactory.kt:40)
    	org.koin.core.instance.SingleInstanceFactory.get(SingleInstanceFactory.kt:48)
    	org.koin.core.registry.InstanceRegistry.resolveInstance$koin_core(InstanceRegistry.kt:87)
    	org.koin.core.scope.Scope.resolveInstance(Scope.kt:214)
    	org.koin.core.scope.Scope.get(Scope.kt:181)

Thanks,
Stefan

weliem, 2021-02-12:
Well, the emulator doesn't have bluetooth. So it makes no sense to instantiate a BluetoothCentral object.....

moodysache, 2021-02-13:
Hello and thank you for your answer! The problem is that this will crash the whole app. Since the emulator doesn't have bluetooth I would say there should be some kind of Failure result without preventing the user to use other application features. As I said, this seems to work properly on some versions of Android. 
    I wonder if you think that there is something to do in order to prevent crashing, or this should be handled in the application.

weliem, 2021-02-13:
In order to use Bluetooth, your application must first do a whole bunch of checks.

- Is there hardware support for Bluetooth?
- Is it enabled?
- Are the relevant permission granted?
- Are location services on?

Only when you have checked all of those, you can instantiate a BluetoothCentralManager class.

Have a look at the example application to see how you do all the checks.

closed 2021-02-20 by weliem