weliem/blessed-android #75: scanForPeripheralsWithServices returns nothing. OnScanResult not get called.

vinuraj-ign, 2021-02-22:
I am new to BLE and trying to connect to a custom Service UUID using **scanForPeripheralsWithServices**. 

`//Custom UUID
    private static final UUID CUSTOM_MEASUREMENT_CHARACTERISTIC_UUID = UUID.fromString("340a1b80-cf4b-11e1-ac36-0002a5d5c51b");
    private static final UUID CUSTOM_SERVICE_UUID = UUID.fromString("02366e80-cf3a-11e1-9ab4-0002a5d5c51b");`

And gets the called when BluetoothHandler is created. 

`handler.postDelayed(new Runnable() {
            @Override
            public void run() {
                Timber.i("bluetooth handler called a");
                central.scanForPeripheralsWithServices(new UUID[]{CUSTOM_SERVICE_UUID});
            }
        },1000);`

As per the logs, I can see that the scan is started, but nothing after that. It is neither triggering OnScanResult or OnScanFailed. 

`2021-02-22 15:05:33.949 23101-23101/com.welie.blessedexample I/BluetoothCentralManager: scan started
2021-02-22 15:06:12.766 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 102 capacity
2021-02-22 15:06:12.770 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x7fd0b7a090: freeing with 120 capacity
2021-02-22 15:06:12.817 23101-23101/com.welie.blessedexample I/Choreographer: Skipped 2 frames!  The application may be doing too much work on its main thread.
2021-02-22 15:06:12.840 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 138 capacity
2021-02-22 15:06:12.845 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 120 capacity
2021-02-22 15:06:12.864 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 102 capacity
2021-02-22 15:06:12.867 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 156 capacity
2021-02-22 15:06:13.161 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 414 capacity
2021-02-22 15:08:33.944 23101-23101/com.welie.blessedexample D/BluetoothCentralManager: scanning timeout, restarting scan
2021-02-22 15:08:33.948 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 102 capacity
2021-02-22 15:08:33.949 23101-23101/com.welie.blessedexample D/BluetoothAdapter: isLeEnabled(): ON
2021-02-22 15:08:33.962 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 114 capacity
2021-02-22 15:08:33.969 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 114 capacity
2021-02-22 15:08:33.971 23101-23101/com.welie.blessedexample I/BluetoothCentralManager: scan stopped
2021-02-22 15:08:34.980 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 216 capacity
2021-02-22 15:08:34.984 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 102 capacity
2021-02-22 15:08:34.987 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 234 capacity
2021-02-22 15:08:34.991 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 102 capacity
2021-02-22 15:08:34.992 23101-23101/com.welie.blessedexample D/BluetoothAdapter: isLeEnabled(): ON
2021-02-22 15:08:34.993 23101-23101/com.welie.blessedexample E/CheckPermission: _bluetooth code = 10 
2021-02-22 15:08:34.996 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 120 capacity
2021-02-22 15:08:34.999 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 102 capacity
2021-02-22 15:08:35.003 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 102 capacity
2021-02-22 15:08:35.013 23101-23101/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0f50: freeing with 114 capacity
2021-02-22 15:08:35.016 23101-23147/com.welie.blessedexample D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=6 mScannerId=0
2021-02-22 15:08:35.037 23101-23147/com.welie.blessedexample D/Parcel: Parcel 0x729a5c0ee0: freeing with 282 capacity
2021-02-22 15:08:35.039 23101-23101/com.welie.blessedexample I/BluetoothCentralManager: scan started`

I have not changed anything apart from the UUID from your code. 

Additionally, When I remove the filter, I can see the devices are getting detected, trying to connect to one of the devices nearby ( Not sure if it is my device ) and fails with the error as below. 

_connection 'null' failed with status CONNECTION_FAILED_ESTABLISHMENT_

Adding more, when I tried this before where there was less BLE devices around me, connection was successful!! 

Any help would be deeply appreciated. 

PS : This is my first Issue raise, let me know if there is any additional information required.

weliem, 2021-02-24:
If you scan for a Service UUID, you will only get results if there are devices that advertise that UUID in their advertising packet. Since you don’t get any results, it seems that the device you are looking for is not advertising the uuid you are supplying.

you can use an app like nRF Connect to see what your device is advertising.

vinuraj-ign, 2021-02-24:
I am able connect the device and read the data with nRF Connect. Forgot to mention that part. Tried to connect with **scanForPeripheralsWithNames** which went pretty well without any issues. I am good to go with this, but still curious to know what went wrong with **scanForPeripheralsWithServices**.

weliem, 2021-02-24:
It often happens that a device has a certain service, but it doesn't list it in the advertisement package. Then you cannot find it by scanForPeripheralWithService

I suspect that is the case for your device. And hence you have to find it by its name or address.

vinuraj-ign, 2021-02-24:
Got it. Closing it then.

closed 2021-02-24 by vinuraj-ign

apo-d, 2021-07-01:
Hi, I have this issue too with a Samsung A40 (Android 11). The service is definitely advertised, since I use the similar scan-call of iOS and it works there.

weliem, 2021-07-01:
Please share a log. Otherwise I cannot help

apo-d, 2021-07-02:
Sure
`I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
I/BluetoothAdapter: STATE_ON
D/BluetoothLeScanner: Start Scan with callback
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=10 mScannerId=0`